stnolting/neorv32 #485: Bootloader UART application upload not working

danirizziero, 2023-02-02:
Hello!
I've been implementing the NEORV32 (extensions: C, M, Zicsr, Zicntr) on a Xilinx Artix-7 FPGA device, embedded on a [TUL PYNQ-Z2](https://www.tulembedded.com/FPGA/ProductsPYNQ-Z2.html) development board. I've been able to map the proper interfaces and interact with the bootloader, but I have serious problems uploading a binary image through the bootloader, via UART communication.
During the transmission these are the two main symptoms:
1) sometimes the bootloader just freezes, seems like waiting for more data to arrive before verifying the checksum and giving back the command prompt. 
2) sometimes seems like the processor simply stops running during the upload, and the heartbeat led stops blinking.

The real issue is that it **DOES** work **SOMETIMES**, for small binary files <5KB, and in these cases I can properly run the uploaded application by further interacting with the bootloader. 

What I tried to do:
- as suggested in the [guide](https://umarcor.github.io/neorv32/ug/#_uploading_and_starting_of_a_binary_executable_image_via_uart) I've been trying TeraTerm as first choice (sending in "binary" mode with no protocol overhead), but also other terminal programs (minicom, mobaXterm, others) obtaining same issue
- I tried to modify the memory DMEM/IMEM dimensions to a bigger value (>> that the needed size), no luck.
- I also tried to modify the UART FIFO width from 1 to 8, but nothing changed.
- thinking about some synchronization problem, I made a python script that manually sends the binary file "word-by-word", but the transmission **un**deterministically fails at some point, never completing for binary files >4-5KB
- to further investigate the problem I modified the bootloader to tell me an increasing counting of received words, and that number always coincides with what I'm sending (e.g., through the python helper script)
- another modification I made to the bootloader was to print back the "signature", "size" and "checksum" that it was receiving as first binary file words, all are received correctly.

**Do you have any other suggestions on what to try or what could be the cause of the problem?** 

Obviously one could compile the application in VHD image file and run it without passing through the BL, but this requires the implementation of the whole FPGA design for each "run", resulting in an enormous time usage when developing an application for this processor.

Thanks a lot in advance, hope that someone can help!

Daniele

GideonZ, 2023-02-02:
This sounds very much like an issue that we encountered in another project.
Turned out to be the cheap-a$$ chinese junk as a USB 2 UART converter. My
initial suggestion would be: try to use a different USB 2 UART converter.

stnolting, 2023-02-02:
Hey @danirizziero!

Sorry to hear that you're encountering problems with the bootloader...

We had a similar problem a while ago, see https://github.com/stnolting/neorv32/pull/215. Some terminal programs (like minicom) have "issues" when transferring larger executables. I am using TeraTerm, which works quite nice. Since you have tried that as well there might be another problem...

* Did you set the top's `CLOCK_FREQUENCY` generic according to your source clock? The bootloader uses this value to compute the actual baud rate for the UART. Maybe there is a small deviation that causes the program to loose "synchronization"?! What clock speed are you using?
* What kind of processor setup are you using? The default one(s) with internal IMEM and DMEM?

> sometimes the bootloader just freezes, seems like waiting for more data to arrive before verifying the checksum and giving back the command prompt.

* The executable for upload via the bootloader contains a small header priot to the actual executable (see https://stnolting.github.io/neorv32/#_executable_image_generator section "Executable Header"). This seems to be trasmitted correctly - otherwise the bootloader would issue an error message. The bootloader then waits until N bytes ares received (N is also encoded in the header) - this might be what you refer as "freezin". After your upload is "completed" you could try holding down the space bar to transmit further (some ~1000s) byte to see if the bootloader ever leaves its receive loop again.

> sometimes seems like the processor simply stops running during the upload, and the heartbeat led stops blinking.

* This is very strange. When the LED stops blinking, does it remain on or off? If it permanently lights up then the core encountered an error it cannot resume from.
* General question: were there any critical warnings during synthesis?

-----------------------------

Hey @GideonZ!

> Turned out to be the cheap-a$$ chinese junk as a USB 2 UART converter. My initial suggestion would be: try to use a different USB 2 UART converter.

I would not generalize here. I have a quite cheap USB2UART adapter from "an Asian purchase website" that works just fine. However, I once bought another adapter from a local electronic store that was really crap 😅

@danirizziero I see that your board also provides an on-board FTDI UART-2-USB adapter. Are you using that one? Maybe you could try using an external one!?

GideonZ, 2023-02-02:
> Turned out to be the cheap-a$$ chinese junk as a USB 2 UART converter. My
>
> I would not generalize here. I have a quite cheap USB2UART adapter from
> "an Asian purchase website" that works just fine. [...]
Oh, don't get me wrong... we just took another one from the same stack,
same brand, same supply... Worked flawlessly!  You just have to be aware
that the stuff you rely on *MIGHT* be broken or unreliable. Just as a
measure to eliminate the source of the problem.

> Message ID: ***@***.***>
>

stnolting, 2023-02-02:
> we just took another one from the same stack, same brand, same supply... Worked flawlessly! You just have to be aware that the stuff you rely on *MIGHT* be broken or unreliable.

Annoying but definitively true! 😅 :+1:

danirizziero, 2023-02-02:
I found the problem!

Turns out the TUL PYNQ-Z2 uses the physical on-board Ethernet-RJ45 socket to provide the on-board 125MHz clock I'm using for the processor. 
[image]
Connecting the board to the PC Ethernet solved any issue <del>...and that's how I wasted the last two days of my life!<del>

Thanks again for the help @GideonZ @stnolting !

stnolting, 2023-02-02:
> I found the problem!

Great to hear! :+1:

I'm curios. What was the actual problem? Did the PHY require some enable to generate the 125MHz signal?

danirizziero, 2023-02-03:
I stumbled upon [this](https://github.com/ThalesGroup/cva6-softcore-contest/issues/6#issuecomment-989186092) saying:

> ...the Pynq-Z2 needs its ethernet cable attached to have a proper clock signal.

So I looked at the board user manual and found the scheme I reported above, then I actually tried connecting the ethernet and found it was the root of the problem. I have no actual insight on the 125MHz clock derivation from the Ethernet PHY, but I suppose that the clock quality and synchronization is mantained by the physical connection to another Ethernet PHY chip/device.