stnolting/neorv32 #544: A few minor things

vhdlnerd, 2023-03-10:
Suggestion: Change the type of the `IO_SPI_FIFO` generic from `natural` to `positive`, since it cannot be 0. (I already made that dumb mistake!)

I was reading the user guide and found these typos.
In the 4.4.3. C Standard Library section:

_These stubs are available in `sw/source/syscalls.c` and were adapted for the NEORV32 processor._
The path should be `sw/lib/source/syscalls.c`.

_See `sw/example/hellp_cpp` for a minimal example._
Should be: `sw/example/hello_cpp`

stnolting, 2023-03-10:
Thanks for the hints! Since these are not really critical, I'll update/fix those within the next commits.

stnolting, 2023-03-16:
I fixed the documentary typo within the docs rework in https://github.com/stnolting/neorv32/pull/549.

I am not sure about the `natural`/`positive` thing... I would like to keep the amount of different generic types to a minimum (just `boolean`, `natural` and `std_ulogic_vector`) as this makes conversion to Verilog slightly more complicated (I know that the CFS config generics are still using `positive`...).

The VHDL sanity checks (`assert ... report ...`) should inform the user if there is an invalid generic configuration. However, at least Xilinx Vivado seems not to care about those assertions... 😅

So I am not sure what to do with those generic types. What do you think?

vhdlnerd, 2023-03-16:
I don't do Verilog but I thought Verilog only had an `integer` type not even a `natural` type. So, wouldn't you avoid 'natural' types, also?

I'm not all that set on the use of `positive` or `natural` here -- `natural` is fine.  The tools throw an error fairly quickly if `IO_SPI_FIFO` is set to 0.

The reason I really like your RISC-V core is that its written in VHDL (I cannot read Verilog) and how well documented it is. Now, I know why you used 'std_ulogic' --  for Verilog conversion. However, the use of the 'std_ulogic' drives me nuts -- after 25 years of VHDL coding (using `std_logic`/`std_logic_vector`), I cannot get my fingers to type that `u`!! 😂

stnolting, 2023-03-28:
> I don't do Verilog but I thought Verilog only had an integer type not even a natural type. So, wouldn't you avoid 'natural' types, also?

Good point, so maybe we should move all generics to type `integer` :thinking:

> The reason I really like your RISC-V core is that its written in VHDL (I cannot read Verilog) and how well documented it is.

Thank you very much!

> Now, I know why you used 'std_ulogic' -- for Verilog conversion.

Actually, I have been using the unresolved VHDL types year before making friends with Verilog. Just my opinion, but I think it makes debugging a bit easier (like finding multi-driver situations).

tmeissner, 2023-03-28:
@stnolting  I do not like it so much to restrict to the less sophisticated type-system of Verilog. What makes you believe that using VHDL's constrained integer types 
> makes conversion to Verilog slightly more complicated

?

stnolting, 2023-03-28:
> I do not like it so much to restrict to the less sophisticated type-system of Verilog

Me neither - VHDL is just so much ~cleaner~ **_better_**! 😅

> What makes you believe that using VHDL's constrained integer types [...]

It is not really about this specific situation. Using `integer` and bit vectors as generics seems to be closer to Verilog - as this is what I have seen so far being used as Verilog parameters.

vhdlnerd, 2023-03-29:
I think it would be fine to use `integer`, `natural`, or `positive` for generics -- they all map to an integer in Verilog. In Verilog, you just would not have the range checking on the constrained types (but that is the way it is in Verilog). (Again, I'm not very knowledgeable in Verilog, so I may be missing something.)

> Actually, I have been using the unresolved VHDL types year before making friends with Verilog. Just my opinion, but I think it makes debugging a bit easier (like finding multi-driver situations).

But, you give up so much more: modeling pull/down resistors, "don't cares" (e.g., with `std_match()`), spotting undriven signals, ...

stnolting, 2023-04-07:
> I think it would be fine to use integer, natural, or positive for generics -- they all map to an integer in Verilog. In Verilog, you just would not have the range checking on the constrained types [...]

:+1:

> But, you give up so much more: modeling pull/down resistors, "don't cares" (e.g., with std_match()), spotting undriven signals, ..

That's right, but modelling pull-down/pull-up resistors should not be a thing "inside" a digital design. Don't cares are not very well supported by the tools (just my opinion) - seems like they are treated like a `'0'`. Undriven signals are not pretty, but they do not hurt I think ;)

stnolting, 2023-04-15:
I think this can be closed, right? 😉

closed 2023-04-15 by stnolting