stnolting/neorv32 #547: Common Makefile Enhancement

vhdlnerd, 2023-03-11:
If the user includes the `common.mk`, there is no way to specify user libraries.  I have a few static libraries I want to link with but I need to modify the `common.mk` to make it work.

I suggest modifying the `common.mk` file with something like:

```makefile
# Compiler & linker flags
CC_OPTS  = -march=$(MARCH) -mabi=$(MABI) $(EFFORT) -Wall -ffunction-sections -fdata-sections -nostartfiles -mno-fdiv
CC_OPTS += -g
CC_OPTS += $(USER_FLAGS)
LD_LIBS =  -lm -lc -lgcc
LD_LIBS += $(USER_LD_LIBS)
LD_OPTS =  -Wl,--gc-sections
LD_OPTS += $(USER_LD_FLAGS)
```
And later for the `elf` target:

```makefile
# Link object files and show memory utilization
$(APP_ELF): $(OBJ)
	@$(CC) $(CC_OPTS) $(LD_OPTS) -T $(LD_SCRIPT) $(OBJ) $(LD_LIBS) -o $@
	@echo "Memory utilization:"
	@$(SIZE) $(APP_ELF)
```

I'm not a `makefile` expert so there may be a better/cleaner way to do this.

vhdlnerd, 2023-03-14:
Also, can I request a `common.mk` change to allow the user to override the linker script?
I.e., change:
```makefile
# Linker script
LD_SCRIPT = $(NEORV32_COM_PATH)/neorv32.ld
```

To:
```makefile
# Linker script
LD_SCRIPT ?= $(NEORV32_COM_PATH)/neorv32.ld
```
Obviously, this would be for advanced users.  (I'll want to do something like that in the near future.)

Thanks.

stnolting, 2023-03-14:
Providing an option to add custom libraries to the compile flow sounds like good idea. :+1: Are we talking about pre-compiled object files (`*.o`) or entire ELF files?

I am not Makefile/GCC expert, but wouldn't it be possible to use the already existent `USER_FLAGS` variable to pass those? Maybe using a specific flag that theses additional parameter should only be processed by the linker (e.g. `-Wl,YOUR_LINKER_ADDONS`)?

> Also, can I request a common.mk change to allow the user to override the linker script?

Sure! This looks like a nice feature add-on!
Would do a PR for this? :wink:

vhdlnerd, 2023-03-14:
I need to link a custom static library (*.a).  I tried adding the lib in via `USER_FLAGS` but its expand before `$(OBJ)` on the linking command line.  This causes the linker not to find references in my library -- the libraries need to be after `$(OBJ)` on the linker command line. (But, other flags in `USER_FLAGS` need to be before)

Technically, adding `LD_OPTS` and `USER_LD_FLAGS` is not needed (`USER_FLAGS` would work).  I thought its cleaner (easier to understand?) to keep the compile flags separate from the linker flags.

Although the current linker script is quite customizable, I want to define one large RAM (16MB of DDR2) and put both `text` and `data` sections into it (a design with no internal IMEM or DMEM).  (Like a more traditional linking scheme.)

vhdlnerd, 2023-03-14:
I'll take a shot at creating a PR with these changes.

stnolting, 2023-03-16:
Sounds good!

> Although the current linker script is quite customizable, I want to define one large RAM (16MB of DDR2) and put both text and data sections into it (a design with no internal IMEM or DMEM). (Like a more traditional linking scheme.

I think that should be no problem with the current linker script as you can override the sections base addresses: https://stnolting.github.io/neorv32/#_linker_script

closed 2023-03-16 by stnolting