stnolting/neorv32 #563: sw/lib/source/neorv32_rte.c

Arun789788, 2023-03-30:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment:**
 - OS
 - GCC Version (RISC-V and native)
 - Libraries used (clib, newlib, ...)

**Hardware:**
 - Hardware version (x.x.x.x)
 - Implemented CPU extensions, peripherals, ...
 - Hardware modifications

**Additional context**
Add any other context about the problem here.

Arun789788, 2023-03-30:
My address country India jila Sultanpur post kamtaganj gram basti paharpur Narsinghpur my name is Arun bihari
Google bug sw/lib/source/neorv32_rte.c

closed 2023-03-30 by Arun789788

stnolting, 2023-03-31:
Was this just a mistake?