stnolting/neorv32 #564: dCache Docs

vhdlnerd, 2023-03-30:
Its great the dCache was added! A few things:
1. A couple typos in the dCache docs (section 2.7.5).  `likt` should be `like` and `257 MB` should be `256 MB`
2. It would be helpful in the docs, section 2.6 Address Space, if the memory map figure showed which region is data cacheable.

vhdlnerd, 2023-03-31:
I fixed the two typos in PR: #565

stnolting, 2023-03-31:
> Its great the dCache was added! A few things:

Haha thanks! But it is still quite basic and not very performance-boosting 😉

> A couple typos in the dCache docs (section 2.7.5). likt should be like and 257 MB should be 256 MB

Thanks for finding (and fixing)!

> It would be helpful in the docs, section 2.6 Address Space, if the memory map figure showed which region is data cacheable.

I agree. I haven't done this yet because I am still thinking about a more generic mapping option. Maybe some programmable boundaries so you can define which addresses are cached and which are not?! 🤔

stnolting, 2023-04-15:
> It would be helpful in the docs, section 2.6 Address Space, if the memory map figure showed which region is data cacheable.

What do you think about this?

[image]

vhdlnerd, 2023-04-16:
Looks great to me!

closed 2023-04-16 by stnolting