stnolting/neorv32 #580: LiberoSoC Microsemi FPGAs: unable to use DMEM lower than 4KB

thegabriele97, 2023-04-14:
Hi!
I am trying to implement the neorv32 core using LiberoSoC for Microsemi FPGAs. Due to the extreme low amount of physical resources available, I need to lower the DMEM and IMEM sizes (I am trying with 1 KB for each one) but I have a strange behaviour: in the pre-synthesis simulation, everything works, while in the post-synthesis simulation it stops working. I am using a simple software that writes 1 to the first bit of the GPIO.

If I use an IMEM of 1 KB and a DMEM of 4KB, it works. Obviously, sizes are changed accordingly in the linker script before generating the app image.

Instatiation:
```vhdl
library ieee;
use IEEE.std_logic_1164.all;

library neorv32;
use neorv32.neorv32_package.all;

entity toptop is
port (
    signal clk: in std_logic;
	signal rstn: in std_logic;
	signal gpio_o_i:  out std_ulogic_vector(63 downto 0);
	signal uart0_rxd: in std_ulogic;
	signal uart0_txd: out std_ulogic

);
end toptop;
architecture architecture_toptop of toptop is

    signal gpio: std_ulogic_vector(63 downto 0);

begin

    gpio_o_i <= gpio;

    DUT: neorv32_top generic map(
		CLOCK_FREQUENCY => 1000000,
		INT_BOOTLOADER_EN => false,
		IO_GPIO_NUM => 64,
        MEM_INT_IMEM_EN => true,
        MEM_INT_IMEM_SIZE => 1*1024,
        MEM_INT_DMEM_EN => true,
        MEM_INT_DMEM_SIZE => 1*1024
	) port map(
		clk_i => clk,
		rstn_i => rstn,
		gpio_o => gpio,
		gpio_i => gpio,
		uart0_txd_o => uart0_txd,
		uart0_rxd_i => uart0_rxd
	);

end architecture_toptop;
```

PRE-SYNTHESIS:
[image]

POST-SYNTHESIS:
[image]

Thank you!

thegabriele97, 2023-04-14:
After more investigations, I connected to the outside the curr_pc signal coming from the cpu instance and this is the result

[image]

stnolting, 2023-04-14:
Hey there!

The configuration looks good, so I think there might be a problem with the internal memories. Which memory modules from [rtl/core/mem](https://github.com/stnolting/neorv32/tree/main/rtl/core/mem) do you use?

The `.default.vhd` files should successfully infer block RAM on most FPGAs. But maybe the Microsemi synthesizer fails here. I would suggest to replace the by the `.legacy.vhd` files.

thegabriele97, 2023-04-14:
> Hey there!
> 
> The configuration looks good, so I think there might be a problem with the internal memories. Which memory modules from [rtl/core/mem](https://github.com/stnolting/neorv32/tree/main/rtl/core/mem) do you use? The `.default.vhd` files should successfully infer block RAM on most FPGAs. But maybe the Microsemi synthesizer fails here. I would suggest to replace them by the `.legacy.vhd` files.

Hi! Thanks for the response. I am using the default one because with the legacy one it doesn't work neither if I set 4 KB.

stnolting, 2023-04-14:
Hmm... that's strange...

So then let's compare the content of the IMEM memory array before synthesis and afterwards. I think that something is going wrong there...

Do you get any synthesis warnings? Can you see if the synthesizer actually infers blockRAM?

thegabriele97, 2023-04-14:
[image]

From this log here I think it infers bRAM correctly. At least for what concerns the DMEM. I don't know about the IMEM but we should consider the fact that there are very few BRAMs available.

Moreover, during the simulation , I see those kind of warnings:

[image]

stnolting, 2023-04-14:
I just had quick view at your synthesis log. Seems like the actual memory files (IMEM and DMEM) are missing - at least I could not find them there. So you'll need to add:
* `rtl/core/mem/neorv32_dmem.default.vhd`
* `rtl/core/mem/neorv32_imem.default.vhd`

(Try the "default" versions first 😉)

stnolting, 2024-02-09:
I think this can be closed. Feel free to open a new issue at any time :wink:

closed 2024-02-09 by stnolting