stnolting/neorv32 #597: synthesis tool error when IO_DMA_EN => true

biosbob, 2023-04-24:
i have my own top-level `.vhd` file, which internally instantiates `neorv32_top.vhd`....  wanting to try the latest DMA peripheral, i added `IO_DMA_EN => true` in the appropriate place....

when i synthesize my design for my iCEBreaker FPGA, however, i receive the following message:

`ERROR: cell type 'neorv32_dma' is unsupported (instantiated as 'neorv32_inst.neorv32_dma_complex_true.neorv32_dma_inst')`

from what i can tell, the dma **IS** instantiated in `neorv32_top.vhd` based on `IO_DMA_EN` being set....

help -- as this sort of problem is (currently) way above my job-level 😉

biosbob, 2023-04-24:
i'm using same yosys/ghdl build flow as supported within the `neorv32-setups/osflow` folder....

my setup uses the `neorv32/rtl/core/neorv32_top.vhd` file, which itself has been updated in the latest commits to this repo....

as a further experiment, i first set `IO_DMA_EN =>` in one of the `neorv32/rtl/processor_template` files and then tried building one of the configurations in `neorv32-setups/osflow`....  same problem....

stnolting, 2023-04-24:
Oh, I think I have not updated the file list yet (-> https://github.com/stnolting/neorv32-setups/blob/main/osflow/filesets.mk) 🙈 I'll fix that ASAP!

biosbob, 2023-04-24:
that explains it....  i can certainly make a quick change on my end, so i can proceed....

interesting, but i did *NOT* have a problem when using vivado to synthesize for the arty FPGA....  why???  because the `create_project.tcl` script "globs" all the `neorv32/rtl/*.vhd` files....

stnolting, 2023-04-24:
`neorv32-setup` should be up-to-date now.

> interesting, but i did NOT have a problem when using vivado to synthesize for the arty FPGA.... why??? because the create_project.tcl script "globs" all the neorv32/rtl/*.vhd files....

Yeah, that might be reason. :wink:

biosbob, 2023-05-15:
yes

closed 2023-05-15 by stnolting