stnolting/neorv32 #602: OrangeCrab - Bootloader + OCD won't work

leo-mrmd, 2023-04-25:
Hi, 

First of all, thank you for your high-quality work on this project. This is a great way to learn VHDL, and CPU architecture.

I am trying to implement the NeoRV32 Core into the OrangeCrab r2.0.1 board (featuring the Lattice ECP5U FPGA).
I've managed to make the CPU and the Bootloader work (second file in **rtl/test_setups**). 

Now I'm using the third file to implement the OCD alongside the CPU, using the **rtl/test_setups/neorv32_test_setup_on_chip_debugger.vhd** file (with some minor modifications to drive some LEDs down) : 

```vhdl
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library neorv32;
use neorv32.neorv32_package.all;

entity neorv32_test_on_chip_debugger is
  generic (
    -- adapt these for your setup --
    CLOCK_FREQUENCY   : natural := 48*1000000; -- clock frequency of clk_i in Hz
    MEM_INT_IMEM_SIZE : natural := 32*1024;   -- size of processor-internal instruction memory in bytes
    MEM_INT_DMEM_SIZE : natural := 16*1024     -- size of processor-internal data memory in bytes
  );
  port (
    -- Global control --
    clk_i       : in  std_ulogic; -- global clock, rising edge
    rstn_i      : in  std_ulogic; -- global reset, low-active, async
    -- JTAG on-chip debugger interface --
    jtag_trst_i : in  std_ulogic; -- low-active TAP reset (optional)
    jtag_tck_i  : in  std_ulogic; -- serial clock
    jtag_tdi_i  : in  std_ulogic; -- serial data input
    jtag_tdo_o  : out std_ulogic; -- serial data output
    jtag_tms_i  : in  std_ulogic; -- mode select
    -- GPIO --
    gpio_o      : out std_ulogic_vector(7 downto 0); -- parallel output
    -- UART0 --
    uart0_txd_o : out std_ulogic; -- UART0 send data
    uart0_rxd_i : in  std_ulogic;  -- UART0 receive data
	--Useless LEDs drive--
	led0_o : out std_ulogic;
	led1_o : out std_ulogic
  );
end entity;

architecture neorv32_test_on_chip_debugger_rtl of neorv32_test_on_chip_debugger is

  signal con_gpio_o : std_ulogic_vector(63 downto 0);

begin

  -- The Core Of The Problem ----------------------------------------------------------------
  -- -------------------------------------------------------------------------------------------
  neorv32_top_inst: neorv32_top
  generic map (
    -- General --
    CLOCK_FREQUENCY              => CLOCK_FREQUENCY,   -- clock frequency of clk_i in Hz
    INT_BOOTLOADER_EN            => true,              -- boot configuration: true = boot explicit bootloader; false = boot from int/ext (I)MEM
    -- On-Chip Debugger (OCD) --
    ON_CHIP_DEBUGGER_EN          => true,              -- implement on-chip debugger
    -- RISC-V CPU Extensions --
    CPU_EXTENSION_RISCV_C        => true,              -- implement compressed extension?
    CPU_EXTENSION_RISCV_M        => true,              -- implement mul/div extension?
    CPU_EXTENSION_RISCV_Zicntr   => true,              -- implement base counters?
    CPU_EXTENSION_RISCV_Zifencei => true,              -- implement instruction stream sync.? (required for the on-chip debugger)
    -- Internal Instruction memory --
    MEM_INT_IMEM_EN              => true,              -- implement processor-internal instruction memory
    MEM_INT_IMEM_SIZE            => MEM_INT_IMEM_SIZE, -- size of processor-internal instruction memory in bytes
    -- Internal Data memory --
    MEM_INT_DMEM_EN              => true,              -- implement processor-internal data memory
    MEM_INT_DMEM_SIZE            => MEM_INT_DMEM_SIZE, -- size of processor-internal data memory in bytes
    -- Processor peripherals --
    IO_GPIO_NUM                  => 8,                 -- number of GPIO input/output pairs (0..64)
    IO_MTIME_EN                  => true,              -- implement machine system timer (MTIME)?
    IO_UART0_EN                  => true               -- implement primary universal asynchronous receiver/transmitter (UART0)?
  )
  port map (
    -- Global control --
    clk_i       => clk_i,       -- global clock, rising edge
    rstn_i      => rstn_i,      -- global reset, low-active, async
    -- JTAG on-chip debugger interface (available if ON_CHIP_DEBUGGER_EN = true) --
    jtag_trst_i => jtag_trst_i, -- low-active TAP reset (optional)
    jtag_tck_i  => jtag_tck_i,  -- serial clock
    jtag_tdi_i  => jtag_tdi_i,  -- serial data input
    jtag_tdo_o  => jtag_tdo_o,  -- serial data output
    jtag_tms_i  => jtag_tms_i,  -- mode select
    -- GPIO (available if IO_GPIO_NUM > 0) --
    gpio_o      => con_gpio_o,  -- parallel output
    -- primary UART0 (available if IO_UART0_EN = true) --
    uart0_txd_o => uart0_txd_o, -- UART0 send data
    uart0_rxd_i => uart0_rxd_i  -- UART0 receive data

  );

  -- GPIO output --
  gpio_o <= con_gpio_o(7 downto 0);
  
  -- Useless LEDs output --
  led0_o <= '1';
  led1_o <= '1';

end architecture;
```
My pinout, using the Lattice Spreadsheet View, is as follow : 

```lpf

BLOCK RESETPATHS ;
BLOCK ASYNCPATHS ;
LOCATE COMP "clk_i" SITE "A9" ;
IOBUF PORT "clk_i" IO_TYPE=LVCMOS33 PULLMODE=NONE ;
IOBUF PORT "rstn_i" IO_TYPE=LVCMOS12 PULLMODE=NONE HYSTERESIS=NA ;
LOCATE COMP "rstn_i" SITE "J17" ;
IOBUF PORT "uart0_rxd_i" IO_TYPE=LVCMOS33 PULLMODE=NONE ;
IOBUF PORT "uart0_txd_o" IO_TYPE=LVCMOS33 OPENDRAIN=OFF ;
IOBUF PORT "gpio_o[6]" PULLMODE=NONE IO_TYPE=LVCMOS33 ;
LOCATE COMP "jtag_trst_i" SITE "B8" ;
IOBUF PORT "jtag_trst_i" IO_TYPE=LVCMOS33 PULLMODE=UP ;
LOCATE COMP "jtag_tms_i" SITE "A8" ;
IOBUF PORT "jtag_tms_i" IO_TYPE=LVCMOS33 PULLMODE=UP ;
LOCATE COMP "jtag_tdi_i" SITE "H2" ;
IOBUF PORT "jtag_tdi_i" IO_TYPE=LVCMOS33 PULLMODE=UP ;
LOCATE COMP "jtag_tck_i" SITE "J2" ;
IOBUF PORT "jtag_tck_i" IO_TYPE=LVCMOS33 PULLMODE=UP ;
LOCATE COMP "jtag_tdo_o" SITE "C8" ;
IOBUF PORT "jtag_tdo_o" IO_TYPE=LVCMOS33 ;
LOCATE COMP "gpio_o[0]" SITE "J3" ;
IOBUF PORT "gpio_o[0]" IO_TYPE=LVCMOS33 ;
LOCATE COMP "led0_o" SITE "M3" ;
IOBUF PORT "led0_o" IO_TYPE=LVCMOS33 ;
LOCATE COMP "led1_o" SITE "K4" ;
IOBUF PORT "led1_o" IO_TYPE=LVCMOS33 ;
LOCATE COMP "gpio_o[1]" SITE "A10" ;
LOCATE COMP "gpio_o[2]" SITE "A11" ;
LOCATE COMP "gpio_o[3]" SITE "C11" ;
LOCATE COMP "gpio_o[4]" SITE "B11" ;
LOCATE COMP "gpio_o[5]" SITE "B10" ;
LOCATE COMP "gpio_o[6]" SITE "C10" ;
LOCATE COMP "gpio_o[7]" SITE "N15" ;
BANK 8 VCCIO 3.3 V;
SYSCONFIG MCCLK_FREQ=2.4 CONFIG_IOVOLTAGE=3.3 ;
BANK 7 VCCIO 1.35 V;
BANK 6 VCCIO 1.35 V;
BANK 3 VCCIO 3.3 V;
BANK 2 VCCIO 1.35 V;
BANK 1 VCCIO 3.3 V;
BANK 0 VCCIO 3.3 V;
IOBUF PORT "gpio_o[1]" IO_TYPE=LVCMOS33 ;
IOBUF PORT "gpio_o[2]" IO_TYPE=LVCMOS33 ;
IOBUF PORT "gpio_o[3]" IO_TYPE=LVCMOS33 ;
IOBUF PORT "gpio_o[4]" IO_TYPE=LVCMOS33 ;
IOBUF PORT "gpio_o[5]" IO_TYPE=LVCMOS33 ;
IOBUF PORT "gpio_o[7]" IO_TYPE=LVCMOS33 ;
LOCATE COMP "uart0_txd_o" SITE "M18" ;
LOCATE COMP "uart0_rxd_i" SITE "N17" ;

```

I ran a testbench, which works fine : I can see the UART frames and the *heartbeat* on GPIO(0) : 

[image]

But when I load the bitstream onto the FPGA, the GPIO(0) is stuck to 0V (the LED is not blinking), and I can't connect to the bootloader via UART (I can't even see the frames with a scope...). I tried to lock the JTAG pins to various states, in case this was the reason (I don't know much about JTAG). 

I'm running out of ideas on what might be the problem. Am I missing something ?

Also, I'm using Lattice Diamond 3.12.1.454.

Let me know if I forgot to share some crucial informations.

Thanks for the help !

leo-mrmd, 2023-04-26:
Hello again !

> So you made them work _on the FPGA_?

Yes, I was able to send commands and even some of your examples via UART.

> What about the `rstn_i` signal? This is connected to the "user button", right? Is this button low-active or high-active? What happens if you keep pushing it?
> 
> The OrangeCrab also seems to have some kind of hardware bootloader (for uploading bitstreams?). I am not familiar with this board but it seems like there is an output signal that re-triggers this bitstream loader:

I already tried to tweak the user button. Wouldn't work. 
For the bootloader, it was already erased when I got the board. Either way, this was not a problem for the first implementation.

> However, I would recommend to disable the on-chip debugger for now (`ON_CHIP_DEBUGGER_EN => false,`) until the basic bootloader works (again?).

I followed this idea and found something interesting. I went back to the project that works, turned off the ON_CHIP_DEBUGGER generic and let the Zifencei on : wouldn't work. So i turned the OCD back on, I commented these two lines in neorv32_cpu.vhd : 

```vhdl
assert not ((CPU_EXTENSION_RISCV_Sdext = true) and (CPU_EXTENSION_RISCV_Zifencei = false)) report
"NEORV32 CPU CONFIG ERROR! Debug mode requires <CPU_EXTENSION_RISCV_Zifencei> extension to be enabled." severity error;
```
And I also deactivated the Zifencei extension. This time it worked ! So the problem seems to be the Zifencei extension. Does that ring a bell to you ? I'll be running some tests tomorrow. 

Thanks again for your time.

leo-mrmd, 2023-04-28:
Hey, 
Sorry for the late answer.

I'm using version 1.8.3.7. 
No relevant warnings, nor notes. The RTL view seems good too.

stnolting, 2023-04-30:
Hm, the version should not be a problem. However, maybe you can try to update to the most recent one :wink:

I am still trying to understand why enabling `Zifencei` would make the processor stall/crash/whatever...

stnolting, 2023-05-02:
> And I also deactivated the Zifencei extension. This time it worked ! So the problem seems to be the Zifencei extension. 

So in this setup the bootloader is responding as expected?

Which files from `rtl/core/mem` are you using for your setup?

leo-mrmd, 2023-05-03:
> So in this setup the bootloader is responding as expected?

Yes, I got the UART communication up and running.

> Which files from `rtl/core/mem` are you using for your setup?

The 'default' ones, as they synthesized properly.

> I am still trying to understand why enabling `Zifencei` would make the processor stall/crash/whatever...

I ran some other tests, and I think that the Lattice Diamond software is greatly responsible in the problems I encountered : I have to create another project from scratch if I want to make the 'MinimalBoot' setup work. Once the Zifencei generic has been synthesized with the 'true' value, turning it back to false or even erasing the line won't make it work again. I didn't notice this at first, because this is not a behavior I've seen anywhere... so I'm now not even sure that the problem comes from the processor.

(I also updated to the last NeoRV32 version)

stnolting, 2023-05-03:
> I ran some other tests, and I think that the Lattice Diamond software is greatly responsible for the problems I encountered

I know this kind of problems from Lattice Radiant. Which synthesis engine do you use? The default _Lattice Synthesis Engine (LSE)_ seems to have some issue with VHDL - or at least with the VHDL style of the processor. However, _Synplify Pro_ runs without any problems. So I would suggest to give that synthesis engine a try.

Just a side note: Synplify Pro seems to create much more efficient designs (higher f_max, less resource consumption).

leo-mrmd, 2023-05-05:
> So I would suggest to give that synthesis engine a try.

I quickly switched to that engine when I started this project. I agree that it is way more efficient than LSE, with more parameters to tweak and match the result I want.

> I know this kind of problems from Lattice Radiant

I did switch, for now, to a Gowin Dev Board, using the Gowin IDE. Even with its limited capabilities, I can make the processor work while having the _Zifencei_ extension activated, and I am confident that the Debug Module will work (If there is enough place on this small chip 😄). So I guess that was definitely the Lattice software causing trouble. 

Anyway, thanks for your time and your dedication to this project 🚀 !

stnolting, 2023-05-13:
Sorry for the late answer...

So you are still having trouble with Lattice? What version of Diamond are you using?

> Anyway, thanks for your time and your dedication to this project 🚀 !

Thank you! 😉

leo-mrmd, 2023-06-05:
Hi, 
sorry for the long time, I took a break from all of this and focused on other things.

I tried again last week (using Diamond 3.12) and still can't figure what the problem is. I'll probably try on a Linux VM (I currently am on Windows 10). I had no problem on the Gowin board and I'm waiting for a Xilinx dev kit to arrived.

stnolting, 2023-06-17:
Let me know if I can help you in any way 😉

stnolting, 2024-02-09:
I think this can be closed. Feel free to open a new issue if you need additional help/information.

closed 2024-02-09 by stnolting