stnolting/neorv32 #623: Random errors occuring with code that is not being executed (!!??)

agamez, 2023-05-26:
Hi!

I'm experiencing some strange bugs that have me completely flabbergasted. I am writing a relatively simple program that makes use of an external interrupt to reprogram some SPI devices. The software also presents an ASCII based user interface through the serial port. I don't think any of this is out of the ordinary, but typing it just presents the characteristics of the configuration of the neorv32.

The architecture of the software, however, is maybe a little bit complex because I'm using OOP concepts (structs with pointers to functions, vtables, etc), so it's a bit difficult sometimes which function is being called. Now, I'm trying to printf-debug this UART interface because I don't have an available JTAG connection (this is implemented on an iCEBreaker and I don't have external hardware to connect JTAG to the output pins). I have my software running flawlessly, reacting perfectly fine to the interrupts and all is fine, but I want to debug this little function, which is only called when I type a specific command through the serial port. If there's no interaction from my part, the CPU will continue to react to the external stimulii and processing interrupts.

```
static char *find_reg(const struct bank_manager *bm, int reg_number)
{
        neorv32_uart0_printf("Reg number %d max %d\n", reg_number, bm->bank_desc->n_regs);
        if (reg_number >= bm->bank_desc->n_regs)
                return NULL;

        /* Calculate register position by adding reg_size */
        char *reg = bm->regs;
        int increment = 0;
        for (int i = 0; i < reg_number; i++) {
                if (increment > 0) {
                        reg += increment;
                } else {
                        if (bm->bank_desc->regs_size[i] == 0 && i > 0)
                                increment = bm->bank_desc->regs_size[i - 1];
                        else
                                increment = bm->bank_desc->regs_size[i];
                        reg += increment;
                }
        }

        return reg;
}
```

So I added that little printf at the beginning of the function. And, when I do that and load that new neorv32_exe.bin file, the CPU spits this out through the serial port **without even having been requested find_reg function to be executed**:

```
Booting from 0x00000000...

> ruction access faultMEOUT_ERR]C=0x7494D4B6VAL=0x44444444TAL EXCEPTION] Halting CPU. </RTE>
```

So, changes in pieces of code that are not being executed produce some kind of error on the CPU. Even the message it spits out is somewhat mangled.

I've been experiencing effects like this for the last couple of days and I've tried almost everything I can think of. This is running on a ice40up5k, which synthesizes with an estimate of 24.81MHz and I've tried running the CPU at just 16MHz and exactly the same thing is happening, so it doesn't seem like a timing issue.

Not being able to run gdb on this is definitely hindering my progress, but my guess is this would probably be a heisenbug which wouldn't show itself when being debugged, as I don't believe this to be a software issue but a hardware (RTL) one.

Any and all kinds of help would be greatly appreciated. Thanks!

agamez, 2023-05-26:
Here's another example that shows the CPU behaving strangely. There's this function:

```
static size_t get_reg_size(const struct bank_manager *bm, int reg_number)
{
        if (reg_number >= bm->bank_desc->n_regs)
                return 0;

        for (int i = 0; i < reg_number; i++) {
                neorv32_uart0_printf("n = %d, %d, %d, %d\n", reg_number, i, bm->bank_desc->regs_size[i], bm->bank_desc->regs_size[i - 1]);
                if (bm->bank_desc->regs_size[i] == 0) {
                        if (i > 0)
                                return bm->bank_desc->regs_size[i - 1];
                        else
                                return 0;
                }
        }

        return bm->bank_desc->regs_size[reg_number];
}
```

I realized I had a off-by-one error and I just replaced the loop to be `for (int i = 0; i <= reg_number; i++)` instead of `for (int i = 0; i < reg_number; i++)`. Again, this function is not being executed, as the CPU fails before it can even reach UI code:

```
Booting from 0x00000000...

> ruction access faultMEOUT_ERR]C=0x7494D4B6VAL=0x44444444TAL EXCEPTION] Halting CPU. </RTE>

```

stnolting, 2023-05-31:
Hey @agamez!

I am not sure if this is really caused by a hardware bug... This damaged massage looks like some problem with the stack:

> ruction access faultMEOUT_ERR]C=0x7494D4B6VAL=0x44444444TAL EXCEPTION] Halting CPU. </RTE>

Maybe it is a problem on the software side. Maybe a crash of the stack and the bss segment or some rogue code that overrides parts of the program code.

Anyway, could you provide a minimal example program?

agamez, 2023-06-01:
Hi @stnolting !

I've tried to reduce the offending code as much as possible, and I've reached a state that is quite small. I'm observing right now a different error, which isn't triggering a TRAP, but I think you are right and it's got something to do with a stack overflow bug because it's corrupting serial output.

The current code should just print: `ART-NG-MCE-1.0 / art_reg_if-v1.0` and keep reading from uart0. However, it prints `NG-MCE-1.0rt_reg_if-v1.0` which really looks like data is being overwritten by someone (and it's definitely not me!).

I've tried to make the code even smaller, but right now it doesn't matter what I change, because removing a variable or struct members, or removing a loop, or reducing the number of arguments of the function or whatever always makes it so the proper string is seen on the UART. Even moving the function on the second file to the main one, makes the error disappear. 

The only relatively unusual thing there is I believe is the union, but it's not that rare, isn't it? Removing it makes the message turn OK, but also does it any other change, so I'm not sure if that's the culprit.

Anyway, since there are a couple of files I created a [repo](https://github.com/agamez/neorv32-debug) to store them, although I'm including the files here too for completeness.

[neorv32-debug.tar.gz](https://github.com/stnolting/neorv32/files/11624243/neorv32-debug.tar.gz)

Thanks a lot!

stnolting, 2023-06-01:
I just tested the code you have attached to your last comment. I just modified the NEORV32 home folder path in the makefile and executed a simple GHDL simulation using the default testbench (latest version of the core):

```
$ make USER_FLAGS+=-DUART0_SIM_MODE clean_all sim
art_reg_if.c: In function 'art_reg_if':
art_reg_if.c:12:21: warning: unused variable 'instruction' [-Wunused-variable]
   12 |                 int instruction = neorv32_uart0_getc();
      |                     ^~~~~~~~~~~
../../../sw/lib/source/neorv32_uart.c: In function 'neorv32_uart_setup':
../../../sw/lib/source/neorv32_uart.c:116:2: warning: #warning UART0_SIM_MODE (primary UART) enabled! Sending all UART0.TX data to text.io simulation output instead of real UART0 transmitter. Use this for simulations only! [-Wcpp]
  116 | #warning UART0_SIM_MODE (primary UART) enabled! Sending all UART0.TX data to text.io simulation output instead of real UART0 transmitter. Use this for simulations only!
      |  ^~~~~~~
Memory utilization:
   text    data     bss     dec     hex filename
   3296       0     120    3416     d58 main.elf
Compiling ../../../sw/image_gen/image_gen
Installing application image to ../../../rtl/core/neorv32_application_image.vhd
Simulating neorv32_application_image.vhd...
Tip: Compile application with USER_FLAGS+=-DUART[0/1]_SIM_MODE to auto-enable UART[0/1]'s simulation mode (redirect UART output to simulator console).
../../src/ieee/math_real-body.vhdl:830:14:warning: declaration of "real_vector" hides type "real_vector" [-Whide]
        type REAL_VECTOR is array (NATURAL range <>) of REAL;
             ^
Using simulation runtime args: --stop-time=10ms
../../rtl/core/neorv32_top.vhd:347:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIGURATION: IMEM DMEM I-CACHE D-CACHE WISHBONE OCD + GPIO MTIME UART0 UART1 SPI SDI TWI PWM WDT TRNG CFS NEOLED XIRQ GPTMR XIP ONEWIRE DMA
../../rtl/core/neorv32_top.vhd:380:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIG NOTE: Boot configuration = direct boot from memory (processor-internal IMEM).
../../rtl/core/neorv32_top.vhd:392:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIG NOTE: Implementing on-chip debugger (OCD).
../../rtl/core/neorv32_cpu.vhd:144:3:@0ms:(assertion note): The NEORV32 RISC-V Processor (Version 0x01080502) - github.com/stnolting/neorv32
../../rtl/core/neorv32_cpu.vhd:148:3:@0ms:(assertion note): NEORV32 CPU CONFIG NOTE: Core ISA ('MARCH') = RV32IMCBU_Zicsr_Zicntr_Zicond_Zifencei_Zfinx_Zihpm_Zxcfu_Sdext_Sdtrig
../../rtl/core/neorv32_cpu.vhd:169:3:@0ms:(assertion warning): NEORV32 CPU WARNING! Assuming this is a simulation.
../../rtl/core/neorv32_cpu.vhd:177:3:@0ms:(assertion note): NEORV32 CPU CONFIG NOTE: Boot from address 0x00000000.
../../rtl/core/neorv32_cpu.vhd:183:3:@0ms:(assertion warning): NEORV32 CPU CONFIG WARNING! Overriding <CPU_IPB_ENTRIES> configuration (setting =2) because C ISA extension is enabled.
../../rtl/core/neorv32_cpu_cp_bitmanip.vhd:173:3:@0ms:(assertion note): NEORV32 CPU: Implementing bit-manipulation (B) sub-extensions Zba Zbb Zbc Zbs
../../rtl/core/neorv32_cpu_cp_cond.vhd:65:3:@0ms:(assertion warning): NEORV32 PROCESSOR CONFIG WARNING: The RISC-V 'Zicond' ISA extension is not ratified yet.
../../rtl/core/neorv32_dcache.vhd:136:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIG NOTE: d-cache uncached memory space 0xf0000000..0xffffffff.
../../rtl/core/mem/neorv32_imem.legacy.vhd:91:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIG NOTE: Using legacy HDL style IMEM.
../../rtl/core/mem/neorv32_imem.legacy.vhd:94:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIG NOTE: Implementing processor-internal IMEM as ROM (16384 bytes), pre-initialized with application (3296 bytes).
../../rtl/core/mem/neorv32_dmem.legacy.vhd:74:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIG NOTE: Using legacy HDL style DMEM.
../../rtl/core/mem/neorv32_dmem.legacy.vhd:77:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIG NOTE: Implementing processor-internal DMEM (RAM, 8192 bytes).
../../rtl/core/neorv32_wishbone.vhd:138:3:@0ms:(assertion note): NEORV32 PROCESSOR CONFIG NOTE: Ext. Bus Interface - CLASSIC/STANDARD Wishbone protocol, auto-timeout (256 cycles), LITTLE-endian byte order, registered RX, registered TX
../../rtl/core/neorv32_trng.vhd:144:3:@0ms:(assertion warning): NEORV32 PROCESSOR CONFIG WARNING: TRNG uses SIMULATION mode!
../../rtl/core/neorv32_trng.vhd:408:3:@0ms:(assertion note): << neoTRNG V2 - A Tiny and Platform-Independent True Random Number Generator for any FPGA >>
../../rtl/core/neorv32_trng.vhd:409:3:@0ms:(assertion note): neoTRNG note: Post-processing enabled.
../../rtl/core/neorv32_trng.vhd:410:3:@0ms:(assertion warning): neoTRNG WARNING: Simulation mode (PRNG!) enabled!
../../rtl/core/neorv32_trng.vhd:734:5:@0ms:(assertion warning): neoTRNG WARNING: Implementing simulation-only PRNG (LFSR)!
../../rtl/core/neorv32_trng.vhd:734:5:@0ms:(assertion warning): neoTRNG WARNING: Implementing simulation-only PRNG (LFSR)!
../../rtl/core/neorv32_trng.vhd:734:5:@0ms:(assertion warning): neoTRNG WARNING: Implementing simulation-only PRNG (LFSR)!
ART-NG-MCE-1.0 / art_reg_if-v1.0
```

The very last console line is the actual UART output, which looks fine, right?

So maybe there is configuration issue (hardware or software) on your side?!

* What version of the core are you using?
* Did you change the default linker script?
* What GCC toolchain / version are you using?
* What core configuration do you use (peripherals, ISA extensions, memory sizes, boot configuration)?

agamez, 2023-06-02:
Right, here's something strange.

If I simulate with your command line:
`
make USER_FLAGS+=-DUART0_SIM_MODE clean_all sim
`

Then my simulation output is exactly as yours: I see the whole string on the console and the message is the full `ART-NG-MCE-1.0 / art_reg_if-v1.0`

If I simulate without defining DUART0_SIM_MODE:

`
make clean_all sim
`

Then I see what I wrote on my post above, the characters one by one and the message that is being output is wrong:
```
uart_rx.simple.vhd:59:15:@504885ns:(report note): uart0.tx: N
uart_rx.simple.vhd:59:15:@1077765ns:(report note): uart0.tx: G
uart_rx.simple.vhd:59:15:@1650645ns:(report note): uart0.tx: -
uart_rx.simple.vhd:59:15:@2223525ns:(report note): uart0.tx: M
uart_rx.simple.vhd:59:15:@2796405ns:(report note): uart0.tx: C
uart_rx.simple.vhd:59:15:@3369285ns:(report note): uart0.tx: E
uart_rx.simple.vhd:59:15:@3942165ns:(report note): uart0.tx: -
uart_rx.simple.vhd:59:15:@4515045ns:(report note): uart0.tx: 1
uart_rx.simple.vhd:59:15:@5087925ns:(report note): uart0.tx: .
uart_rx.simple.vhd:59:15:@5660805ns:(report note): uart0.tx: 0
uart_rx.simple.vhd:59:15:@6233685ns:(report note): uart0.tx: r
uart_rx.simple.vhd:59:15:@6806565ns:(report note): uart0.tx: t
uart_rx.simple.vhd:59:15:@7379445ns:(report note): uart0.tx: _
uart_rx.simple.vhd:59:15:@7952325ns:(report note): uart0.tx: r
uart_rx.simple.vhd:59:15:@8525205ns:(report note): uart0.tx: e
uart_rx.simple.vhd:59:15:@9098085ns:(report note): uart0.tx: g
uart_rx.simple.vhd:59:15:@9670965ns:(report note): uart0.tx: _
```

Is this a bug in the UART code then?

agamez, 2023-06-02:
> Here, address `9d4` is missing. This causes an offset in `.rodata`. So all the addresses / data there are "off" by 4 bytes.

That definitely explains why a working program wouldn't fail ever but a wrong one would fail even in the region of code that was not being executed. With a little bit of luck, only the strings were misaligned, but code could still execute, otherwise instructions would all be mangled up and the existence of const pointers to functions probably made it easier to spot.

> Thank you very much for finding this (heisen-)bug! This is a very interesting corner case you have found! :wink:

Well, my. Thanks to you, definitely. I wouldn't have ever guessed any of that. I truly appreciate your assistance. The quality of your work is more than remarkable!

stnolting, 2023-06-02:
I have started implementing a fix for this in #626.

closed 2023-06-02 by stnolting