stnolting/neorv32 #640: feature request - Zb* sub-extensions

biosbob, 2023-06-28:
the `Zbc` sub-extension (carry-less multiplication) is quite different from the other sub-extensions, which really do "bit manipulation" in the more traditional sense...

not critical, but it might be helpful to enable the `Zb*` sub-extensions individual -- as this also has impact on the overall size of the synthesized core....

a placeholder for the future....

stnolting, 2023-06-30:
For the records: this is being discussed in https://github.com/stnolting/neorv32/issues/633#issuecomment-1604684303

biosbob, 2023-06-30:
#633 talks about the `Zc*` sub-extensions; this issue concerns the `Zb*` sub-extensions....

perhaps the data-sheet could not only list the extensions currently supported, but those on (or NOT on) the roadmap....

stnolting, 2023-06-30:
Oh sorry, seems like I have mixed up something 😅

`B` vs. `Zb*` is something I need to take care of because the `B` extensions basically does not exist (anymore): https://github.com/riscv/riscv-bitmanip/issues/183

So we should replace `CPU_EXTENSION_RISCV_B` by several `CPU_EXTENSION_RISCV_Zb*` generics to fine tune the bitmanip extension. Actually, this fine-grained tuning is already implemented:

https://github.com/stnolting/neorv32/blob/a5bf1c822ff02482029f3aa93e8ca07e91a45a77/rtl/core/neorv32_cpu_cp_bitmanip.vhd#L74-L79

But theses switches are not propagated to the top nor to the control unit to check for legal/illegal instructions.

ecstrema, 2025-03-14:
I've just tried using zbb, and couldn't have it work:

code:

```c
  uint32_t trailing_zeros;
  asm volatile("ctz %0, %1" : "=r"(trailing_zeros) : "r"(mask));
  neorv32_uart0_printf("%d\n", trailing_zeros);
```

ecstrema, 2025-03-14:
Enabled extensions: 

```vhdl
      -- RISC-V CPU Extensions --
      RISCV_ISA_C      => true, -- Compressed ISA (16-bit instructions)
      RISCV_ISA_M      => true, -- Integer Multiplication and Division
      RISCV_ISA_U      => true, -- Integer base ISA
      RISCV_ISA_ZICNTR => true, -- Base counters extension
      RISCV_ISA_Zbb    => true, -- Bit manipulation extension (used to count trailing bits)
```

Error printed to the console:

```
<NEORV32-RTE> [cpu0|M] Illegal instruction @ PC=0x0000024C, MTINST=0x60161613, MTVAL=0x00000000 </NEORV32-RTE>
```

ecstrema, 2025-03-14:
Note that MTINST (`0x6016123`) corresponds to `ctz` ([link](https://five-embeddev.com/riscv-bitmanip/1.0.0/bitmanip.html#insns-ctz))

Is there anything else I need to make this work?

Unike267, 2025-03-14:
Maybe related with:

https://github.com/stnolting/neorv32/blob/330b00bcabaac79dafa5cb5bb769412b2d72c9c6/rtl/core/neorv32_cpu_cp_bitmanip.vhd#L138

According to `30.5.18. ctz`.  

`Funct12` of ctz is `011000000001` no `011000000000` 

[image]

Now I'll confirm, and if this is the issue, I'll make a PR!

Source: https://github.com/riscv/riscv-isa-manual/releases/download/riscv-isa-release-2c2a793-2025-03-07/riscv-unprivileged.pdf

Unike267, 2025-03-14:
Hi, I reproduced the issue in simulation:

[image]

> Funct12 of ctz is 011000000001, not 011000000000.

Sorry, my bad. This is not the issue because it refers to `ctrl_i.ir_funct12(11 downto 1)`, which is 11 bits, ignoring the LSB.

I'll keep investigating.

ecstrema, 2025-03-14:
What that was quick. I haven't taken the time to simulate yet, because the setup with my fpga was so simple, and the testing cycle is so fast I up to that point didn't need it! Plus the instructions everywhere were so clear that most things worked first shot!

Since i only need the `ctz` instruction at compile time, I ended up using gcc's `__builtin_ctz` instead.

Unike267, 2025-03-14:
>  I ended up using gcc's __builtin_ctz instead.

Oh nice workaround! 

> What that was quick.

I followed these steps https://github.com/stnolting/neorv32/issues/1188#issuecomment-2666305346

But adding to the top:

> -- RISC-V CPU Extensions --
>       RISCV_ISA_C      => true, -- Compressed ISA (16-bit instructions)
>       RISCV_ISA_M      => true, -- Integer Multiplication and Division
>       RISCV_ISA_U      => [...]

And compiling:

```c
#include <neorv32.h>

#define BAUD_RATE 19200

int main() {

  // capture all exceptions and give debug info via UART
  // this is not required, but keeps us safe
  neorv32_rte_setup();

  // setup UART at default baud rate, no interrupts
  neorv32_uart0_setup(BAUD_RATE, 0);

  uint32_t trailing_zeros;
  uint32_t mask = 0x00000001;
  asm volatile("clz %0, %1" : "=r"(trailing_zeros) : "r"(mask));
  neorv32_uart0_printf("%d\n", trailing_zeros);

  return 0;
}
```

Through:

`podman run --rm -itv $(pwd):/wrk:Z -w /wrk stnolting/neorv32/sim:latest sh -c "make -C sw/example/neorv32-640 clean_all MARCH=rv32imac_zicsr_zifencei_zbb info image"` (Adding zbb)

Cheers!

stnolting, 2025-03-14:
@ecstrema 

I guess you have `CPU_FAST_SHIFT_EN` disabled in your setup, right?

Maybe the iterative shifter of the `B` extension is broken for this specific (= `Zbb`) configuration... 🤔 I'll have a look.

stnolting, 2025-03-14:
I can confirm this bug. The shift operations (inlcuding `clz` and `ctz`) of the `Zbb` instructions are broken if `CPU_FAST_SHIFT_EN ` and `RISCV_ISA_Zbkc` are both disabled. Damn cross-dependencies... 🙈😅

I'll fix that ASAP.

Thanks for finding this bug!

ecstrema, 2025-03-14:
Wow. You guys are incredible. I indeed have fast shift disabled.

stnolting, 2025-03-14:
> On the debugging way, I noticed that the opcode of the ctz instruction shown in the ctrl.ir_opcode signal is 0x13 (19), but you're defining it as 3: [...]

Think of these `op_*` constants as an `enum`. Basically, this is a one-hot bit vector that indicates the decoded instruction type (not the actual instruction itself). So it is just a decoding helper and the individual bit positions are not related to any opcode patterns.

Btw, `op_cz_c` stands for "operation ..." not "opcode" (if I remeber correctly 😅).

Unike267, 2025-03-14:
> Basically, this is a one-hot bit vector that indicates the decoded instruction type (not the actual instruction itself). So it is just a decoding helper and the individual bit positions are not related to any opcode patterns.

Ok, I got it!

> Btw, op_cz_c stands for "operation ..." not "opcode" (if I remeber correctly 😅).

Perfect, now everything fits, enjoy the weekend! 😄