stnolting/neorv32 #663: NEORV32 Multicore processor TMR implementation

Luperior, 2023-08-01:
### Discussed in https://github.com/stnolting/neorv32/discussions/661

<div type='discussions-op-text'>

<sup>Originally posted by **Luperior** July 31, 2023</sup>
Good afternoon everyone, 
for my Master Thesis I am working on a multicore version of the Neorv32 in order to reduce the error probability due to bit flipping in the bitstream. My idea is to **triplicate the CPU (done already)** and then introduce a **Triple Modular Redundancy "security gate"** before writing data into memory. Basically I want to compare the results obtained by the three different cores and then store the one which, in case of disagreement, is most obtained (eg 2 times out of 3). 
In order to do so, I modified the _neorv32_cpu_bus.vhd_ file by adding a comparison between the result coming from the same core ALU (_wdata_i_) and the ones coming from the other two cores ALUs (_wdata_ix_ and _wdata_iy_). Should the three CPUs all produce different results, every single one will store the data obtained by their own ALU. This is the code:

```
mem_do_reg: process(rstn_i, clk_i)
  begin
    if (rstn_i = '0') then
      d_bus_wdata_o <= (others => '0');
      d_bus_ben_o   <= (others => '0');
    elsif rising_edge(clk_i) then
      if (ctrl_i.bus_mo_we = '1') then
	  d_bus_ben_o <= (others => '0'); -- default
	    -- Check for a majority vote (2 out of 3 match)
		if (wdata_i = wdata_ix) or (wdata_i = wdata_iy) then
        case ctrl_i.ir_funct3(1 downto 0) is
          when "00" => -- byte
            d_bus_wdata_o(07 downto 00) <= wdata_i(7 downto 0);
            d_bus_wdata_o(15 downto 08) <= wdata_i(7 downto 0);
            d_bus_wdata_o(23 downto 16) <= wdata_i(7 downto 0);
            d_bus_wdata_o(31 downto 24) <= wdata_i(7 downto 0);
            d_bus_ben_o(to_integer(unsigned(addr_i(1 downto 0)))) <= '1';
          when "01" => -- half-word
            d_bus_wdata_o(15 downto 00) <= wdata_i(15 downto 0);
            d_bus_wdata_o(31 downto 16) <= wdata_i(15 downto 0);
            if (addr_i(1) = '0') then
              d_bus_ben_o <= "0011"; -- low half-word
            else
              d_bus_ben_o <= "1100"; -- high half-word
            end if;
          when others => -- word
            d_bus_wdata_o <= wdata_i;
            d_bus_ben_o   <= "1111";
        end case;
        elsif wdata_ix = wdata_iy then
          case ctrl_i.ir_funct3(1 downto 0) is
          when "00" => -- byte
            d_bus_wdata_o(07 downto 00) <= wdata_ix(7 downto 0);
            d_bus_wdata_o(15 downto 08) <= wdata_ix(7 downto 0);
            d_bus_wdata_o(23 downto 16) <= wdata_ix(7 downto 0);
            d_bus_wdata_o(31 downto 24) <= wdata_ix(7 downto 0);
            d_bus_ben_o(to_integer(unsigned(addr_i(1 downto 0)))) <= '1';
          when "01" => -- half-word
            d_bus_wdata_o(15 downto 00) <= wdata_ix(15 downto 0);
            d_bus_wdata_o(31 downto 16) <= wdata_ix(15 downto 0);
            if (addr_i(1) = '0') then
              d_bus_ben_o <= "0011"; -- low half-word
            else
              d_bus_ben_o <= "1100"; -- high half-word
            end if;
          when others => -- word
            d_bus_wdata_o <= wdata_ix;
            d_bus_ben_o   <= "1111";
        end case;
        else
          -- No majority vote, there's a disagreement
        case ctrl_i.ir_funct3(1 downto 0) is
          when "00" => -- byte
            d_bus_wdata_o(07 downto 00) <= wdata_i(7 downto 0);
            d_bus_wdata_o(15 downto 08) <= wdata_i(7 downto 0);
            d_bus_wdata_o(23 downto 16) <= wdata_i(7 downto 0);
            d_bus_wdata_o(31 downto 24) <= wdata_i(7 downto 0);
            d_bus_ben_o(to_integer(unsigned(addr_i(1 downto 0)))) <= '1';
          when "01" => -- half-word
            d_bus_wdata_o(15 downto 00) <= wdata_i(15 downto 0);
            d_bus_wdata_o(31 downto 16) <= wdata_i(15 downto 0);
            if (addr_i(1) = '0') then
              d_bus_ben_o <= "0011"; -- low half-word
            else
              d_bus_ben_o <= "1100"; -- high half-word
            end if;
          when others => -- word
            d_bus_wdata_o <= wdata_i;
            d_bus_ben_o   <= "1111";
        end case;
        end if;
	  end if;
    end if;
  end process mem_do_reg;
```

Clearly, in order to make all of this possible, I added two extra IN signals for each CPU (wdata_ix and wdata_iy) and modified the "**rs2**" signal to be (IN)OUT since it represents the ALU outcome and has to be sent to the other cores. So for example the first core takes in _cpu2.rs2_ and _cpu3.rs2_ and outputs _cpu1.rs2_ and so on... however it does not work.
I am using Vivado to deploy it on a XILINX PYNQ Z2 FPGA and Synthesis and Implementation complete without any major warning (nor error obviously), and the Implemented Design makes total sense. I already tested the code and bistream end everything works fine without the TMR implementation so it shouldn't be a problem caused by the core triplication.
How is it possible then that I obtain nothing coming from the UART0 when I should see the first CPU output (after being "TMR-checked")?
Thank you all in advance!

Edit: I also tried other different approaches, like using Shared Variables or istantiating a new component which played the role of the TMR majority voter. However, let alone that neither of them worked, the first one is likely more subject to errors due to bit flipping and the second one (probably) had synchronisation issues since the values were available a few clocks after they were needed. 
Should you have any other suggestions I am open to hear them, thanks!</div>

**ISSUE UPDATE**
I am opening an Issue starting from discussion since I just noticed something weird in the Implemented Design Netlist (picture):
[image]

1) Despite being instatiated 3 times in the code (and in the Vivado hierarchy), the neorv32_busswitch is only created once (blue);
2) Same thing goes for the icache component (red), in contrast with the dcache which is correctly istantiated three times;
3) Only the first core has an ALU: the second and third do not possess one (green).

How is all of this possible? What is going on? I guess this is why I see nothing on wdata_ix and wdata_iy and so on...

stnolting, 2023-08-01:
I recommend to start with a setup **without** any caches. These can be added later after your bus modifications are done (and verified).

> Despite being instatiated 3 times in the code (and in the Vivado hierarchy), the neorv32_busswitch is only created once (blue);

I think this might be caused by incomplete signal/bus assignments, e.g. the CPU's instruction buses are not correctly mapped resulting in a removal of the i-cache.

Do you have your source code in a public repository? Then we could have a closer look at it.

Luperior, 2023-08-01:
> I recommend to start with a setup **without** any caches

Enabling them unfortunately proved to be necessary as discovered here (https://github.com/stnolting/neorv32/issues/655#issuecomment-1655225209).

> Do you have your source code in a public repository?

Here is the link (https://www.dropbox.com/scl/fo/cy6xkl9ys2hpzcbbpilh7/h?rlkey=30c11po4r3xcbzh36btxbwmar&dl=0) containing all the 3 files I modified.

I also quote your words from here (https://github.com/stnolting/neorv32/discussions/661#discussioncomment-6608786) : "Btw, you are building a setup with three identical cores doing the same stuff at the same time. There is a (tiny?) chance that the synthesis tool is smart enough to identify this redundancy removing all but one cores." 
I guess this may just be the case but, if this is really what happens, how can I fix it? After searching online it appears that Vivado optimization can't be disabled, but there must be a way to obtain redundancy in my design if that is what I really wish...

stnolting, 2023-08-02:
> Enabling them unfortunately proved to be necessary as discovered here (https://github.com/stnolting/neorv32/issues/655#issuecomment-1655225209).

I still think this is just a wiring issue 😉

> Here is the link (https://www.dropbox.com/scl/fo/cy6xkl9ys2hpzcbbpilh7/h?rlkey=30c11po4r3xcbzh36btxbwmar&dl=0) containing all the 3 files I modified.

I just scrolled through your `neorv32_top.vhd` file. `core_req2` and `core_req3` are driven by the according core-complexes but they are not connected to any module. That's why synthesis will trim all the according logic. All three `core_req` buses need to combined into a final SoC bus.

That is what I was suggesting in #661: just add another bus module (the TMR controller) and connect these three buses to it. The module makes the majority vote and then forwards the request(s) via single/unified system bus to the rest of the processor.

Luperior, 2023-08-04:
> That is what I was suggesting in #661: just add another bus module (the TMR controller) and connect these three buses to it. The module makes the majority vote and then forwards the request(s) via single/unified system bus to the rest of the processor.

So what you're suggesting is to implement the TMR verification module (let's say) in the _neorv32_intercon file_, gather the three `core_reqX` and then output what is needed for the neorv32_busswitch component right? This way I'd have only one bus mux and `core_req2` and `core_req3` would be actually connected to an instanced module. 

There is one thing though that's bothering me, wouldn't I add a 1-T_clk delay?

stnolting, 2023-08-04:
That's right. Add a new TMR module to the `neorv32_intercon.vhd` file that takes care of the majority vote. This is what the entity of that module might look like:

```vhdl
entity neorv32_bus_tmr is
  port (
    -- global control --
    clk_i        : in  std_ulogic;
    rstn_i       : in  std_ulogic;
    -- core complex ports --
    core_a_req_i : in  bus_req_t;
    core_a_rsp_o : out bus_rsp_t;
    core_b_req_i : in  bus_req_t;
    core_b_rsp_o : out bus_rsp_t;
    core_c_req_i : in  bus_req_t;
    core_c_rsp_o : out bus_rsp_t;
    -- system port --
    sys_req_o    : out bus_req_t;
    sys_rsp_i    : in  bus_rsp_t
  );
end neorv32_bus_tmr;
```

Connect your three core complexes to that module and wire the system port to the processor's GATEWAY module.

Inside the TMR module you just basically just need to _gate_ the `sys_req_o.we` signal. You can do that in an asnychronous way (no need to register that) so you won't have any additional delay - we are doing the same in the reservation set controller.

But even if you add one or more register stages - that should not hurt too much because of the CPU caches.

Luperior, 2023-08-05:
> Inside the TMR module you just basically just need to _gate_ the `sys_req_o.we` signal. You can do that in an asnychronous way (no need to register that) so you won't have any additional delay - we are doing the same in the reservation set controller.

This is how I decided to handle things, starting from your entity declaration:

```
architecture neorv32_bus_tmr_rtl of neorv32_bus_tmr is
begin  
  write_en: process(core_a_req_i.we, core_b_req_i.we, core_c_req_i.we)
  begin
    if (core_a_req_i.we = '1' and core_b_req_i.we = '1' and core_c_req_i.we = '1') then
	-- Check for a majority vote (2 out of 3 match)
        if (core_a_req_i.data = core_b_req_i.data) or (core_a_req_i.data = core_c_req_i.data) then
          sys_req_o <= core_a_req_i;
        elsif core_b_req_i.data = core_c_req_i.data then
          sys_req_o <= core_b_req_i;
        else
        -- No majority vote, there's a disagreement
          sys_req_o <= core_a_req_i;
        end if;
	core_a_rsp_o <=  sys_rsp_i;
	core_b_rsp_o <=  sys_rsp_i;
	core_c_rsp_o <=  sys_rsp_i;
	end if;
end process write_en;
end architecture neorv32_bus_tmr_rtl;
```

I then connected the system ports to the host ports of the gateway module.

Now when I compile everything on Vivado the obtained design makes much more sense, with ALUs and busswitches actually present and I have to thank you a lot for that. However my program still doesn't work, and IMEM and DMEM are still instantiated only once. Did I miss or misunderstand something?  
I am thinking I may be needing three gateway modules (to attach IMEMs and DMEMs) but then wouldn't I be missing the point of unifying the three core requests in the TMR module? Let alone that probably I'd have concurrent signal assignments errors.
Thank you in advance, I am sorry to bother you this much but I was left totally unsupervised and/or assisted for this Thesis...

stnolting, 2023-08-05:
> This is how I decided to handle things, starting from your entity declaration:

Look good, but there are some issues here:

* the sensitivity list is incomplete: you are using several signals (e.g. `core_a_req_i.data`) as input for the process logic - not just `core_a_req_i.we, core_b_req_i.we, core_c_req_i.we`. Better use the the entire requests buses in the list:
```vhdl
write_en: process(core_a_req_i, core_b_req_i, core_c_req_i)
```

* `sys_rsp_i` is not in the sensitivity list. Better make this concurrent (outside of the process):

```vhdl
...
    end if;
  end process write_en;

  core_a_rsp_o <=  sys_rsp_i;
  core_b_rsp_o <=  sys_rsp_i;
  core_c_rsp_o <=  sys_rsp_i;
```

* I am not really sure, but I think there might be one case missing in your majority logic: 😅
  * `A.data = B.data` OR `A.data = C.data`: `SYS <= A`
  * `B.data = A.data` OR `B.data = C.data`: `SYS <= B`
  * `C.data = A.data` OR `C.data = B.data`: `SYS <= C`

* A general question regarding your TMR concept: Do you just want to check the actual data or do you also want to check the remaining bus signal? For instance, there could also be a discrepancy in the address lines.

> I then connected the system ports to the host ports of the gateway module.

Sounds good.

> Now when I compile everything on Vivado the obtained design makes much more sense, with ALUs and busswitches actually present and I have to thank you a lot for that.

:+1:

> However my program still doesn't work, and IMEM and DMEM are still instantiated only once. Did I miss or misunderstand something?

Well, it depends on your system architecture. I thought you want to build a uniform system based on three cores that **share** all the memory-mapped modules. Hence, you only have one IMEM, one DMEM, one peripheral X, one peripheral Y, etc.

tmeissner, 2023-08-05:
> the sensitivity list is incomplete: you are using several signals (e.g. core_a_req_i.data) as input for the process logic - not just core_a_req_i.we, core_b_req_i.we, core_c_req_i.we. Better use the the entire requests buses in the list:

Or even better, use the keyword all in the sensitivity list of combinatorial processes:

process (all) is
begin
...
end process;

Luperior, 2023-08-13:
First of all sorry for getting back to you so late, it was my summer-break week. Now, when it comes to this project:

> * the sensitivity list is incomplete: you are using several signals (e.g. `core_a_req_i.data`) as input for the process logic - not just `core_a_req_i.we, core_b_req_i.we, core_c_req_i.we`. Better use the the entire requests buses in the list:

Thank you, I changed that and tried both this way and with "all" in the process sensitivity list but none of them worked.

> * `sys_rsp_i` is not in the sensitivity list. Better make this concurrent (outside of the process):

Thank you again for that.

> * I am not really sure, but I think there might be one case missing in your majority logic: 😅

I tried with a couple examples and I'm not sure it's wrong, but eventually I'll try and test it more deeply.

> * A general question regarding your TMR concept: Do you just want to check the actual data or do you also want to check the remaining bus signals? For instance, there could also be a discrepancy in the address lines.

I'd like to start with a functioning system that checks only what is supposed to be stored, then should I have enough time I'd be happy to improve the verification method.

> I thought you want to build a uniform system based on three cores that **share** all the memory-mapped modules. Hence, you only have one IMEM, one DMEM, one peripheral X, one peripheral Y, etc.

You're right, thank you once more...

Despite all this, however, it still doesn't work. It looks like everything that's needed is actually implemented in the design, but the test program still does not produce any output. What else could I try?

Luperior, 2023-08-13:
> Or even better, use the keyword `all` in the sensitivity list of combinatorial processes:

Thank you for the nice suggestion!

Luperior, 2023-08-13:
**Update**

I tried commenting the whole process containing the TMR verification code (pasted below) and just assigning, outside the process, sys_req_o to the different core_X_req_i and it worked for all three cores. I'm then guessing it must be a problem of the first IF condition 

```
  write_en: process(core_a_req_i, core_b_req_i, core_c_req_i, clk_i, rstn_i, sys_rsp_i)
  -- write_en: process(all) -- not supported before VHDL 2008 in Vivado
  begin
    if (core_a_req_i.we = '1' and core_b_req_i.we = '1' and core_c_req_i.we = '1') then   -- unreachable
		-- Check for a majority vote (2 out of 3 match)
        if (core_a_req_i.data = core_b_req_i.data) or (core_a_req_i.data = core_c_req_i.data) then
          sys_req_o <= core_a_req_i;
        elsif core_b_req_i.data = core_c_req_i.data then
          sys_req_o <= core_b_req_i;
        else
        -- No majority vote, there's a disagreement
          sys_req_o <= core_a_req_i;
        end if;
    else  -- solved the unreachable clause issue
        sys_req_o <= core_b_req_i;
    end if;
end process write_en;
```

since adding the else branch solved it. So I'm pretty sure the problem is that the three cores do not share a time instant when all three _.we_ signals are asserted, and as a consequence sys_req_o was left undefined. However, in this way, the TMR verification is not performed since this else statement is always executed. How could I solve it?

stnolting, 2023-08-14:
> and just assigning, outside the process, sys_req_o to the different core_X_req_i and it worked for all three cores.

That sounds fine.

> So I'm pretty sure the problem is that the three cores do not share a time instant when all three .we signals are asserted, or at least it doesn't happen every time the process is called (clearly) and as a consequence sys_req_o was left undefined.

The CPU does not provide any kind of "lock-step" mechanism to synchronize with other cores so we'll need to ensure synchronous execution via a different mechanism. Basically, all three cores are identical - so they should behave identical, too.

What kind of boot scenario do you use? I assume the cores are booting from the pre-initialized IMEM - so no bootloader and no on-chip debugger, right?

> However, in this way, the TMR verification is not always performed since this else statement is almost always executed. How could I solve it?

This is related to a more general question: what happens if - for example - one core goes crazy? Some time ago I worked on a similar TMR setup also implementing three cores. In case of one core going rogue the according core was disabled and the remaining one were informed via an IRQ.

> Edit: I checked and saw that the .we signals are defined as "single shot", so I may need to extend their duration or find another way to keep track of their activation. Do you have any suggestions? Where can I find the exact behaviour of these signals?
Thank you in advance.

These "single-shot" signals have to be single-shot. If you extend them, e.g. asserting them for several cycles, then you would trigger **several** read/write transactions, which would compromise data integrity.

The CPU/processor bus interface is described in the data sheet: [DOC: Bus Interface](https://stnolting.github.io/neorv32/#_bus_interface)

Luperior, 2023-08-14:
> Basically, all three cores are identical - so they should behave identical, too.

My point exactly, thank you.

> I assume the cores are booting from the pre-initialized IMEM - so no bootloader and no on-chip debugger, right?

Yes, you're right.

> This is related to a more general question: what happens if - for example - one core goes crazy?

I see your point, an else statement is needed to handle strange and unpredictable behaviour.

Looking at the IF condition `` if (core_a_req_i.we = '1' and core_b_req_i.we = '1' and core_c_req_i.we = '1') then `` I tried to substitute the ``and`` gates with ``or`` ones, and even using ``rising_edge(core_X_req_i.we)`` to see if anything changed in the behaviour but got nothing. It looks like it never enters the if branch but that would be impossible since the process is triggered (confirmed by the fact that the else condition, when present, solves everything despite making the TMR gate pointless). At least sometimes the ``.we`` signals must reach value 1 in order to trigger it, so how come it never enters the IF branch even with the ``or`` logic? Or maybe it enters it but just _not enough times_ to make the system work? Like when the IF condition is false it may stilll need some signal to be assigned to `sys_req_o` but then it would TMR check only some times and not for every data (as seen with the else branch for example) ...

stnolting, 2023-08-14:
Your TMR logic can only be active if all three cores issue a write request (via `.we`) at the same time. But there might be the case where - for example - the address bus of one core is corrupted during the current write request. So why not check **all** request bus signals for a match?

I am not sure if VHDL can compare records, so I would use a custom comparison function. Maybe something like this:

```vhdl
  function cmp_bus_req_f(x : bus_req_t; y : bus_req_t) return boolean is
  begin
    if (x.addr = y.addr) and
       (x.data = y.data) and
       (x.ben  = y.ben)  and
       (x.we   = y.we)   and
       (x.re   = y.re)   and
       (x.src  = y.src)  and
       (x.priv = y.priv) and
       (x.rvso = y.rvso) then
      return true;
    else
      return false;
    end if;
  end function cmp_bus_req_f;
```

With this function you can compare the entire request buses to check for an actual majority:

```vhdl
tmr: process(core_a_req_i, core_b_req_i, core_c_req_i)
begin
  if (cmp_bus_req_f(core_a_req_i, core_b_req_i) = true) or
     (cmp_bus_req_f(core_a_req_i, core_c_req_i) = true) then
    sys_req_o <= core_a_req_i;
    tmr_err   <= '0';
  elsif (cmp_bus_req_f(core_b_req_i, core_a_req_i) = true) or
        (cmp_bus_req_f(core_b_req_i, core_c_req_i) = true) then
    sys_req_o <= core_b_req_i;
    tmr_err   <= '0';
  elsif (cmp_bus_req_f(core_c_req_i, core_a_req_i) = true) or
        (cmp_bus_req_f(core_c_req_i, core_b_req_i) = true) then
    sys_req_o <= core_c_req_i;
    tmr_err   <= '0';
  else -- discrepancy detected
    sys_req_o <= req_terminate_c; -- no system bus access is performed at all
    tmr_err   <= '1';
  end if;
end process tmr;
```

The `tmr_err` signal is set when there is a discrepancy - but ignore this signal for the moment.

To ensure that all three cores are in sync you can do a simulation and inspect all three request buses (`core_a_req_i`, `core_b_req_i`, `core_c_req_i`) inside a waveform.

Luperior, 2023-08-15:
Thank you so much, it is now actually working! What's more, I only ran a quick simulation for now but it looks like the error rate dropped from over 50% (single core) to about 25% (triple core with TMR verification): it really is amazing!

I only have another question, when I now implemented the new design on Vivado it gave me this critical warning _"The design failed to meet the timing requirements"_ . I don't remember setting any time requirement or constraint, is it something I should be worried about? Can it be fixed? Or can I ignore it given that everything seems to be working more than fine?
Below is a more detailed screenshot of the issue.

[image]

GideonZ, 2023-08-15:
1.2 ns can certainly not be ignored. For testing you might ignore a few ps
(not ns!) of negative slack, but certainly never for production builds. You
will have to either lower the clock frequency or fix your critical timing
paths, or play with optimization settings to achieve better results. This
is a no-go.

stnolting, 2023-08-15:
> Thank you so much, it is now actually working!

Great to hear! 🎉

> What's more, I only ran a quick simulation for now but it looks like the error rate dropped from over 50% (single core) to about 25% (triple core with TMR verification): it really is amazing!

What do you mean with error rate? Do you actually simulate faults in one ore more cores?

> I only have another question, when I now implemented the new design on Vivado it gave me this critical warning "The design failed to meet the timing requirements" . I don't remember setting any time [...]
Below is a more detailed screenshot of the issue.

Maybe you did not set an explicit timing constraint, but a PLL or DCM might add an "implicit" timing constraint. How do you generate the core's clock signal? Do you use a PLL for that?

The timing results looks quite... bad... very bad... 😅 Please check the timing reports for the critical path. I think the comparator logic might be a problems here, so maybe you should add at least one register stage to the system bus:

```vhdl
tmr: process(rstn_i, clk_i)
begin
  if (rstn_i) then
    sys_req_o <= req_terminate_c;
    tmr_err   <= '1';
  elsif rising_edge(clk_i) then
    if (cmp_bus_req_f(core_a_req_i, core_b_req_i) = true) or
       (cmp_bus_req_f(core_a_req_i, core_c_req_i) = true) then
      sys_req_o <= core_a_req_i;
      tmr_err   <= '0';
    elsif (cmp_bus_req_f(core_b_req_i, core_a_req_i) = true) or
          (cmp_bus_req_f(core_b_req_i, core_c_req_i) = true) then
      sys_req_o <= core_b_req_i;
      tmr_err   <= '0';
    elsif (cmp_bus_req_f(core_c_req_i, core_a_req_i) = true) or
          (cmp_bus_req_f(core_c_req_i, core_b_req_i) = true) then
      sys_req_o <= core_c_req_i;
      tmr_err   <= '0';
    else -- discrepancy detected
      sys_req_o <= req_terminate_c; -- no system bus access is performed at all
      tmr_err   <= '1';
    end if;
  end if;
end process tmr;
```

Obviously, this will increase the latency but that should not be too bad as you have the caches enabled.

Luperior, 2023-08-15:
> 1.2 ns can certainly not be ignored. For testing you might ignore a few ps (not ns!) of negative slack, but certainly never for production builds. You will have to either lower the clock frequency [...]

Thank you for explaining that to me!

Luperior, 2023-08-15:
> What do you mean with error rate? Do you actually simulate faults in one ore more cores?

I usually simulate N faults in the active parts of the bitstream (after "seeing" an image of it), and then test the system 10000 times to evaluate the error rate (each time with different flipped bits). Comparing the new result with the old one, at least when it comes to 30 faults which was my first choice, the error rate dropped from 50% to 25%.
 

> How do you generate the core's clock signal? Do you use a PLL for that?

I left untouched the default clock generation.

> I think the comparator logic might be a problems here, so maybe you should add at least one register stage to the system bus

Thank you, I just tested it and it all works fine without the timing issue! 

I really can't thank you enough, I'll make sure to quote you in my acknowledgements section. Also, should it be possible to donate something to your for your work and effort, I'd be more than happy to!

stnolting, 2023-08-15:
> Thank you, I just tested it and it all works fine without the timing issue!

That's great! 🚀
So, can we close this issue?

> I really can't thank you enough

You're welcome.

closed 2023-08-15 by Luperior

GideonZ, 2023-08-15:
>
> Comparing the new result with the old one, at least when it comes to 30
> faults which was my first choice, the error rate dropped from 50% to 25%.
>
And the next time it will be 60% and some other time 5%. I think changing
an FPGA design like this is too dependent on routing and mapping. Maybe I
am wrong on this, but it doesn't sound like a useful measure at all.

I may have missed the entire purpose of this thread though. I lurk in here
and sometimes pick things up. :-)