stnolting/neorv32 #668: ISA Zfinx extension

Snorny, 2023-08-10:
Hello,

I'd like to ask you a question. I would like to use the Zfinx ISA extension to perform float calculations.
The only problem is that my code compiles whether I add zfinx to the compile flag or not.
(-march=rv32im_zfinx)

My question is how can I be sure that the float calculations are used by the hardware and not the software?

Thank you in advance for your answers.

GideonZ, 2023-08-10:
The easiest way to check this is disassembling the generated code with
riscv32-unknown-elf-objdump -d and check for the float operations.

As you said you can compile twice, so you can make the dump twice and diff
them.

Snorny, 2023-08-10:
It seems that the operations are the same, so I deduce that I haven't activated the zfinx extension properly.

stnolting, 2023-08-10:
You need to check first if your toolchain actually supports the `Zfinx` extension. I think support has been added with GCC 12.something?!

Anyway, you can also "force" the compiler to infer `Zfinx` instructions using the provided [intrinsics library](https://github.com/stnolting/neorv32/blob/main/sw/example/floating_point_test/neorv32_zfinx_extension_intrinsics.h). This is also possible even if the toolchain does not support `Zfinx` yet.

Snorny, 2023-08-10:
> You need to check first if your toolchain actually supports the `Zfinx` extension. I think support has been added with GCC version 12.something?!
> 
> Anyway, you can also "force" the compiler to infer `Zfinx` instructions using the provided [intrinsics library](https://github.com/stnolting/neorv32/blob/main/sw/example/floating_point_test/neorv32_zfinx_extension_intrinsics.h). This is also possible even if the toolchain does not support `Zfinx` yet.

Thanks for those precision, how can i know if the toolchain support it ?

stnolting, 2023-08-10:
Good question 😅 

Unfortunately, it is quite hard to make GCC emit a list off all supported `Z*` sub-extensions... Better compare your version of GCC/binutils with the official release lists.

Or just try to compile a simple floating-point program with `Zfinx` in the `march` string and scan the resulting assembly code for any floating point instructions.

stnolting, 2023-08-17:
> Other question, does the riscv32-unknown-elf-gcc can use this extension ?

No, not yet.

According to this list https://gcc.gnu.org/gcc-13/changes.html support for the `Zfinx` was introduced this year with GCC 13.

However, you can still utilize this extension with older GCC versions by using intrinsics.

Snorny, 2023-08-18:
Thanks a lot for your help ! 

> However, you can still utilize this extension with older GCC versions by using intrins
Yeah thanks for this precision ! It could be helpful.

closed 2023-08-18 by Snorny