stnolting/neorv32 #700: `make check` passes but `make bin` fails with pre-built toolchain

lianakoleva, 2023-10-10:
**Describe the bug**
When trying to build an executable from the examples using the provided pre-built 32-bit toolchain (`cd $NEORV32_HOME/sw/example/demo_blink_led; make clean_all bin`), I experience error `` ABI is incompatible with that of the selected emulation: target emulation `elf32-littleriscv' does not match `elf64-littleriscv' `` even though `make check` passes.

**To Reproduce**
Download [pre-built toolchain `32i-4.0.0`](https://github.com/stnolting/riscv-gcc-prebuilt). Extract with `sudo tar -xzf riscv32-unknown-elf.gcc-12.1.0.tar.gz -C /opt/riscv/` in download directory. Install by appending `export PATH=$PATH:/opt/riscv/bin` to `~/.bashrc`, then running `source ~/.bashrc` (`echo $PATH` contains `/opt/riscv/bin`). `riscv32-unknown-elf-gcc -v` returns 
```
Using built-in specs.
COLLECT_GCC=riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv32-unknown-elf/12.1.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /tmp/rv_gcc/riscv-gnu-toolchain/gcc/configure --target=riscv32-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g1ea978e3066 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=/tmp/rv_gcc/riscv-gnu-toolchain/gcc --disable-multilib --with-abi=ilp32 --with-arch=rv32i --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.1.0 (g1ea978e3066) 
```
Change `RISCV_PREFIX` to `riscv32-unknown-elf-`. Run `make check` -- it succeeds with output
```
---------------- Check: NEORV32_HOME folder ----------------
NEORV32_HOME: ../../..
---------------- Check: Shell ----------------
/bin/sh
/usr/bin/dash
---------------- Check: riscv32-unknown-elf-gcc ----------------
Using built-in specs.
COLLECT_GCC=riscv32-unknown-elf-gcc
COLLECT_LTO_WRAPPER=/opt/riscv/libexec/gcc/riscv32-unknown-elf/12.1.0/lto-wrapper
Target: riscv32-unknown-elf
Configured with: /tmp/rv_gcc/riscv-gnu-toolchain/gcc/configure --target=riscv32-unknown-elf --prefix=/opt/riscv --disable-shared --disable-threads --enable-languages=c,c++ --with-pkgversion=g1ea978e3066 --with-system-zlib --enable-tls --with-newlib --with-sysroot=/opt/riscv/riscv32-unknown-elf --with-native-system-header-dir=/include --disable-libmudflap --disable-libssp --disable-libquadmath --disable-libgomp --disable-nls --disable-tm-clone-registry --src=/tmp/rv_gcc/riscv-gnu-toolchain/gcc --disable-multilib --with-abi=ilp32 --with-arch=rv32i --with-tune=rocket --with-isa-spec=2.2 'CFLAGS_FOR_TARGET=-Os   -mcmodel=medlow' 'CXXFLAGS_FOR_TARGET=-Os   -mcmodel=medlow'
Thread model: single
Supported LTO compression algorithms: zlib
gcc version 12.1.0 (g1ea978e3066) 
---------------- Check: riscv32-unknown-elf-objdump ----------------
GNU objdump (GNU Binutils) 2.39
Copyright (C) 2022 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
---------------- Check: riscv32-unknown-elf-objcopy ----------------
GNU objcopy (GNU Binutils) 2.39
Copyright (C) 2022 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
---------------- Check: riscv32-unknown-elf-size ----------------
GNU size (GNU Binutils) 2.39
Copyright (C) 2022 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) any later version.
This program has absolutely no warranty.
---------------- Check: NEORV32 image_gen ----------------
<<< NEORV32 executable image generator >>>
by Stephan Nolting
Three arguments are required.
1st: Option
 -app_bin : Generate application executable binary (binary file, little-endian, with header) 
 -app_img : Generate application raw executable memory image (vhdl package body file, no header)
 -raw_hex : Generate application raw executable (ASCII hex file, no header)
 -raw_bin : Generate application raw executable (binary file, no header)
 -bld_img : Generate bootloader raw executable memory image (vhdl package body file, no header)
2nd: Input file (raw binary image)
3rd: Output file
4th: Project folder (optional)
---------------- Check: Native GCC ----------------
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/lib/gcc/x86_64-linux-gnu/11/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none:amdgcn-amdhsa
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-linux-gnu
Configured with: ../src/configure -v --with-pkgversion='Ubuntu 11.4.0-1ubuntu1~22.04' --with-bugurl=file:///usr/share/doc/gcc-11/README.Bugs --enable-languages=c,ada,c++,go,brig,d,fortran,objc,obj-c++,m2 --prefix=/usr --with-gcc-major-version-only --program-suffix=-11 --program-prefix=x86_64-linux-gnu- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --enable-bootstrap --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-vtable-verify --enable-plugin --enable-default-pie --with-system-zlib --enable-libphobos-checking=release --with-target-system-zlib=auto --enable-objc-gc=auto --enable-multiarch --disable-werror --enable-cet --with-arch-32=i686 --with-abi=m64 --with-multilib-list=m32,m64,mx32 --enable-multilib --with-tune=generic --enable-offload-targets=nvptx-none=/build/gcc-11-XeT9lY/gcc-11-11.4.0/debian/tmp-nvptx/usr,amdgcn-amdhsa=/build/gcc-11-XeT9lY/gcc-11-11.4.0/debian/tmp-gcn/usr --without-cuda-driver --enable-checking=release --build=x86_64-linux-gnu --host=x86_64-linux-gnu --target=x86_64-linux-gnu --with-build-config=bootstrap-lto-lean --enable-link-serialization=2
[... 98 lines cut ...]
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-mlock.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf32-littleriscv' does not match `elf64-littleriscv'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-mlock.o)
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/bin/ld: /opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-sbrkr.o): ABI is incompatible with that of the selected emulation:
  target emulation `elf32-littleriscv' does not match `elf64-littleriscv'
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/bin/ld: failed to merge target specific data of file /opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/lib/libc.a(lib_a-sbrkr.o)
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/bin/ld: main.elf(.text.startup.main): relocation "neorv32_gpio_port_set+0x0 (type R_RISCV_CALL)" goes out of range
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/bin/ld: ./main.c.o: file class ELFCLASS32 incompatible with ELFCLASS64
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/bin/ld: final link failed: file in wrong format
collect2: error: ld returned 1 exit status
make: *** [../../../sw/common/common.mk:196: main.elf] Error 1
```

**Expected behavior**
I expect `make clean_all bin` to complete without error.

**Environment:**
 - OS: Ubuntu 22.04.01
 - GCC Version 12.1.0 (RISCV), GCC Version 11.4.0 (Native)

**Hardware:**
 - x86_64 processor

**Additional context**
I noticed that `riscv32-unknown-elf` has been configured with `--disable-multilib`. When I attempted to build the toolchain myself, I used the configuration command `--enable-multilib` because I thought that was the cause of the error, but this did not solve it.

stnolting, 2023-10-10:
Hey @lianakoleva!

I am using the same toolchain over here and I have never experienced this behavior...
Can you please run `make info` and post the output?

> Change RISCV_PREFIX to riscv32-unknown-elf-

No need to change that as `riscv32-unknown-elf-` should be the default. 🤔
https://github.com/stnolting/neorv32/blob/715917ddb4f26ff9d53a52f19c4c22c972f30222/sw/common/common.mk#L51

Which version of the processor are you using?

lianakoleva, 2023-10-10:
`make info` outputs
```
------------------------------------------------------
-- Project
------------------------------------------------------
Project folder:        demo_blink_led
Source files:          ./main.c   
Include folder(s):     -I .
ASM include folder(s): -I .
------------------------------------------------------
-- NEORV32
------------------------------------------------------
NEORV32 home folder (NEORV32_HOME): ../../..
IMAGE_GEN: ../../../sw/image_gen/image_gen
Core source files:
../../../sw/lib/source/neorv32_cfs.c ../../../sw/lib/source/neorv32_cpu_amo.c ../../../sw/lib/source/neorv32_cpu.c ../../../sw/lib/source/neorv32_cpu_cfu.c ../../../sw/lib/source/neorv32_crc.c ../../../sw/lib/source/neorv32_dma.c ../../../sw/lib/source/neorv32_gpio.c ../../../sw/lib/source/neorv32_gptmr.c ../../../sw/lib/source/neorv32_mtime.c ../../../sw/lib/source/neorv32_neoled.c ../../../sw/lib/source/neorv32_onewire.c ../../../sw/lib/source/neorv32_pwm.c ../../../sw/lib/source/neorv32_rte.c ../../../sw/lib/source/neorv32_sdi.c ../../../sw/lib/source/neorv32_slink.c ../../../sw/lib/source/neorv32_spi.c ../../../sw/lib/source/neorv32_trng.c ../../../sw/lib/source/neorv32_twi.c ../../../sw/lib/source/neorv32_uart.c ../../../sw/lib/source/neorv32_wdt.c ../../../sw/lib/source/neorv32_xip.c ../../../sw/lib/source/neorv32_xirq.c ../../../sw/lib/source/syscalls.c ../../../sw/common/crt0.S
Core include folder:
../../../sw/lib/include
------------------------------------------------------
-- Objects
------------------------------------------------------
Project object files:
./main.c.o ../../../sw/lib/source/neorv32_cfs.c.o ../../../sw/lib/source/neorv32_cpu_amo.c.o ../../../sw/lib/source/neorv32_cpu.c.o ../../../sw/lib/source/neorv32_cpu_cfu.c.o ../../../sw/lib/source/neorv32_crc.c.o ../../../sw/lib/source/neorv32_dma.c.o ../../../sw/lib/source/neorv32_gpio.c.o ../../../sw/lib/source/neorv32_gptmr.c.o ../../../sw/lib/source/neorv32_mtime.c.o ../../../sw/lib/source/neorv32_neoled.c.o ../../../sw/lib/source/neorv32_onewire.c.o ../../../sw/lib/source/neorv32_pwm.c.o ../../../sw/lib/source/neorv32_rte.c.o ../../../sw/lib/source/neorv32_sdi.c.o ../../../sw/lib/source/neorv32_slink.c.o ../../../sw/lib/source/neorv32_spi.c.o ../../../sw/lib/source/neorv32_trng.c.o ../../../sw/lib/source/neorv32_twi.c.o ../../../sw/lib/source/neorv32_uart.c.o ../../../sw/lib/source/neorv32_wdt.c.o ../../../sw/lib/source/neorv32_xip.c.o ../../../sw/lib/source/neorv32_xirq.c.o ../../../sw/lib/source/syscalls.c.o ../../../sw/common/crt0.S.o
------------------------------------------------------
-- RISC-V CPU
------------------------------------------------------
MARCH:      rv32imac_zicsr
MABI:       ilp32
------------------------------------------------------
-- Toolchain
------------------------------------------------------
Toolchain:  
CC:         riscv32-unknown-elf-gcc
OBJDUMP:    riscv32-unknown-elf-objdump
OBJCOPY:    riscv32-unknown-elf-objcopy
SIZE:       riscv32-unknown-elf-size
DEBUGGER:   riscv32-unknown-elf-gdb
------------------------------------------------------
-- GDB Arguments
------------------------------------------------------
GDB_ARGS:   -ex target extended-remote localhost:3333
------------------------------------------------------
-- Libraries
------------------------------------------------------
LIBGCC:
/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/libgcc.a
SEARCH-DIRS:
install: /opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/
programs: =/opt/riscv/libexec/gcc/riscv32-unknown-elf/12.1.0/:/opt/riscv/libexec/gcc/riscv32-unknown-elf/12.1.0/:/opt/riscv/libexec/gcc/riscv32-unknown-elf/:/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/:/opt/riscv/lib/gcc/riscv32-unknown-elf/:/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/bin/riscv32-unknown-elf/12.1.0/:/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/bin/
libraries: =/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/:/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/lib/riscv32-unknown-elf/12.1.0/:/opt/riscv/lib/gcc/riscv32-unknown-elf/12.1.0/../../../../riscv32-unknown-elf/lib/:/opt/riscv/riscv32-unknown-elf/lib/riscv32-unknown-elf/12.1.0/:/opt/riscv/riscv32-unknown-elf/lib/:/opt/riscv/riscv32-unknown-elf/usr/lib/riscv32-unknown-elf/12.1.0/:/opt/riscv/riscv32-unknown-elf/usr/lib/
------------------------------------------------------
-- Flags
------------------------------------------------------
USER_FLAGS: 
CC_OPTS:    -march=rv32imac_zicsr -mabi=ilp32 -Os -Wall -ffunction-sections -fdata-sections -nostartfiles -mno-fdiv -mstrict-align -mbranch-cost=10 -g -Wl,--gc-sections 
------------------------------------------------------
-- Libraries
------------------------------------------------------
USER_LIBS: 
LD_LIBS:   -lm -lc -lgcc  
```
RE `riscv32-unknown-elf-`: I must've changed it earlier, sorry for the confusion!
RE processor version: I am on the main branch.

stnolting, 2023-10-10:
So everything is working fine now? 😉

lianakoleva, 2023-10-10:
Yes, on the compilation front! `make all` created all the expected files. I am getting the following in `cutecom` though: `Comm error: Could not start sz`.

closed 2023-10-10 by lianakoleva