stnolting/neorv32 #710: Provide a neorv32_uart_vprintf function

jpf91, 2023-10-22:
**Is your feature request related to a problem? Please describe.**

In C/C++, it is not possible to write a function which forwards to printf style functions (https://stackoverflow.com/a/150626 ). This means it's currently not possible to write a wrapper  function for `neorv32_uart_printf`. Howerver, such a wrapper is required to make the printf task-safe in FreeRTOS applications.

**Describe the solution you'd like**
Move most code of the printf function into a new vprintf function. Then make the old printf function call the vprintf one.

**Describe alternatives you've considered**
I'd prefer a solution which works with the current codebase.  The only other one I know is copying the printf function and dependencies entirely.

stnolting, 2023-10-23:
Hey @jpf91!

I was not aware of something like `vprintf` (I'm no software expert), but it sounds like a good idea!

> Howerver, such a wrapper is required to make the printf task-safe in FreeRTOS applications.

Could you explain "task-safe" in that context a little bit more?

> Move most code of the printf function into a new vprintf function. Then make the old printf function call the vprintf one.

Shouldn't be too complicated. I'll draft a PR for that.

stnolting, 2023-10-23:
What do you think about this?

https://github.com/stnolting/neorv32/blob/688a429332fc6f3ccb0bf1a877cc04621564e2de/sw/lib/source/neorv32_uart.c#L328-L389

The `printf`-to-`vprintf` wrapper seems to work fine.

https://github.com/stnolting/neorv32/blob/688a429332fc6f3ccb0bf1a877cc04621564e2de/sw/lib/source/neorv32_uart.c#L400-L406

jpf91, 2023-10-23:
> If I understand the vprintf function correctly, a NEORV32-flavored version would look like this, right?

Exactly :-)

> Could you explain "task-safe" in that context a little bit more?

Of course :-) It's just like thread-safe in non-embedded programming. Although in this case, the CPU does not have real multiprocessing (no to threads / tasks executing in parallel), FreeRTOS does have preemption (stopping tasks at random times and running other tasks, interleaving). Because of this "interleaving" a printf call in task 1 can be paused / interleaved by task2, which might also call printf. As a result, you get interleaved output. To avoid this, calls need to be synchronized using a mutex.

> Shouldn't be too complicated. I'll draft a PR for that.

Great, thanks a lot :smile:

jpf91, 2023-10-23:
> What do you think about this?

Looks great :+1: Curiously there seems to be little information about how to implement a vprintf like function. But I think your code is correct.

stnolting, 2023-10-23:
> Although in this case, the CPU does not have real multiprocessing (no to threads / tasks executing in parallel), FreeRTOS does have preemption (stopping tasks at random times and running other [...]

This is where I would simply turn off interrupts... 😅 But no question, the mutex approach seems to be the better/safer way to handle this.

> Curiously there seems to be little information about how to implement a vprintf like function. But I think your code is correct.

I was also wondering about this... 🤔

closed 2023-10-24 by stnolting