stnolting/neorv32 #720: Application does not persist

fedy0, 2023-11-03:
**Describe the bug**
The app.bin flashed into the FPGA (ULX3S) does not persist in memory after reset or power on-off

**To Reproduce**
My setup can be found [here](https://github.com/fedy0/neo.git)

**Expected behavior**
app.bin to persist in memory

**Screenshots**
My Screenshot can be found [here](https://github.com/fedy0/neo#test)

**Environment:**
 - Ubuntu 22.04
 - GCC Version (RISC-V and native): riscv64-unknown-elf-gcc (g1ea978e3066) 12.1.0
 - Libraries used: newlib

**Hardware:**
 - Hardware version: latest

**Additional context**
None

stnolting, 2023-11-04:
Hey @fedy0!

I think I am not quite sure what this is about. When you upload an executable via the bootloader it will be stored to the processor's instruction memory. By default, this is implemented as RAM. If you power-cycle the FPGA all content - including the RAM data - will be lost.

> The app.bin flashed into the FPGA (ULX3S) does not persist in memory after reset or power on-off

What do you mean by "flashed into the FPGA"?

> My Screenshot can be found [here](https://github.com/fedy0/neo#test)

I'm curious. Seems like your are explicitly calling the image generator after `make all`. Do you use a customized version fo it?

fedy0, 2023-11-04:
> By default, this is implemented as RAM

Ok now I know why.

```bash
cd ./neorv32/sw/example/hello_world
make all
../../image_gen/image_gen -app_bin main.bin hello.bin
sudo sh ../../image_gen/uart_upload.sh /dev/ttyUSB0 hello.bin
sudo apt install gtkterm
gtkterm --port /dev/ttyUSB0 --speed 19200
```
I call `make all` to build `image_gen` as it does not exist by default in the repo/project
Also I used `-app_bin` to add header to the generated binary as the bootloader does not recognize the default-generated bin (without header) as a valid bin

stnolting, 2023-11-04:
> I call make all to build image_gen as it does not exist by default in the repo/project

Right, that's normal as I try to keep pre-built binaries out of the repo. :wink:

> Also I used -app_bin to add header to the generated binary as the bootloader does not recognize the default-generated bin (without header) as a valid bin

Have you modified the default bootloader?

`make exe` should build the image generator and convert the compiled application into a valid executable that can be uploaded via the bootloader - so no additional calling of `image_gen` required.

> `sudo sh ../../image_gen/uart_upload.sh /dev/ttyUSB0 hello.bin`

Oh this script! I'm always surprised that people actually use it. I had already forgotten that it even existed. 😅

fedy0, 2023-11-04:
Yeah `make exe` produces binaries with header

1. Can we make it programmable like every other MCU that uses UART, a BOOT pin and a RESET pin rather than the command? 
2. Can we make it such that the binaries can be read from an external flash? This means the bootloader will support moving the exe from UART port to the Flash.
3. Make user modify the pin locations and Flash address at will.

This will enable me to do the porting we discussed earlier.

stnolting, 2023-11-06:
> Yeah make exe produces binaries with header

:+1:

> Can we make it programmable like every other MCU that uses UART, a BOOT pin and a RESET pin rather than only from the commands issued to the bootloader?

Interesting idea! What do you suggest the BOOT pin should do? Boot from IMEM (or more general: boot from address zero) or run the bootloader?

> Can we make it such that the binaries can be read from an external flash? This means the bootloader will support moving the exe from UART to the Flash during bin upload.

The bootloader already provides an option to program an executable to an external SPI flash: https://stnolting.github.io/neorv32/ug/#_programming_an_external_spi_flash_via_the_bootloader

> Make user modify the pin locations and Flash address at will.
This will enable me to do the porting we discussed earlier.

There are several options to customize the bootloader (flash base boot address etc.): https://stnolting.github.io/neorv32/ug/#_customizing_the_internal_bootloader

> What do you use (recommend) if you've forgotten this script? 😅

Most of the time I use the on-chip debugger. But when I play with the bootloader I just use a simple terminal program to upload files (like _Tera Term_).

fedy0, 2023-11-11:
> What do you suggest the BOOT pin should do? Boot from IMEM (or more general: boot from address zero) or run the bootloader?

To act almost exactly as shown in the picture attached (from ESP32 datasheet similar to other MCUs too).

[image]

stnolting, 2023-11-18:
Sorry for the late reply... 🙈

> To act almost exactly as shown in the picture attached (from ESP32 datasheet similar to other MCUs too).

Basically, the processor already provides something like a boot select, but in the form of a generic + the bootloader's auto-boot sequence.

When the bootloader is enabled, it will try to load an executable via SPI if there is no user interaction. Furthermore, the bootloader console allows to boot an application directly via XIP, but this actions has to be manually triggered. Maybe this could also be added to the auto-boot sequence 🤔

Anyway, there is no option yet to use a pre-initialized IMEM (executable "burned" into the bitstream" **and** the bootloader in parallel. Is that the use case you are pointing at?

fedy0, 2023-11-18:
Yes, that option.

This will also make it easier to implement the SPI boot and/or  SPI firmware download sequence without the command line. But I still find the boot command prompt useful.

Also, I'll try and test again because occasionally the start memory for the SPI is missed on boot

stnolting, 2023-12-04:
The basic idea behind the "boot-via-bootloader OR boot-from-memory" configuration was to simplify this boot setup for the user. Of course, there are much more application-specific boot scenarios. The question is how do we provide a comfortable setup without constraining too much?

The most flexible way would be to
* provide a generic that controls implementation of the boot ROM
* provide a generic that controls implementation of a **per-initialized** IMEM
* provide another generic to manually set the boot address
* add a new command to the bootloader to jump right to the beginning of the pre-initialized IMEM

But this would also complicate the general core setup. So I am not sure which way to go here.
Proposals are highly welcome :wink:

> Also, I'll try and test again because occasionally the start memory for the SPI is missed on boot

What do you mean here?
The bootloader sometimes fails to fetch an image from SPI flash??

fedy0, 2023-12-09:
Sorry for the late reply I will take a closer look at your suggestions.

The SoC occasionally fails to boot. I'll try to recreate it and send a screenshot

stnolting, 2024-02-11:
Do you need any additional help or can we close this? ;)

fedy0, 2024-02-14:
Sorry for the late reply. I have not been able to recreate the problem. You may close it. Thank you for your support. BTW, I'll provide an update soon.

closed 2024-02-14 by stnolting