stnolting/neorv32 #732: Configure Bootloader.c booting from address other than start of IMEM (Ex: 0x2000, etc)

SteveZ-Cal, 2023-11-20:
**FPGA Setup**

NEORV32 on Cyclone V De0-CV
IMEM: 128KB
DMEM: 64KB

**Hardware**

Version: V1.8.6 (Purpose: compatible to Zephyr latest version)

**Bootloader Console**

I just followed "5.Uploading and Starting of a Binary Executable Image via UART" using TeraTerm for windows

Set Baud Rate to 19200

**Describe the bug**

_When booting from address other than start of IMEM, neorv32_uart0_printf won't print the desired output, even though the binary content starting from that booting address is correct_

**Ex: Booting from 0x2000**

I just configured the bootloader.c with its inline assembly starting from 0x2000 instead of NEORV32_SYSINFO->ISPACE_BASE

Program to be uploaded -> a simple hello_world file with "neorv32_port_set(2) to indicate if the PC actually start/execute the file I uploaded given it's booting from address other than 0x0000"

When upload its corresponding bin file to the terminal program, even though the program is executing ( given that the led is on from the neorv32_port_set(2), which eliminated the execution error ), there's nothing to be print out or garbage to be print out. (I also checked that it's not a file misaligned issue)

stnolting, 2023-11-20:
Hey @SteveZ-Cal.

So you want to make the bootloader boot from address 0x00002000, right? Did you also modify the "ROM" base address in the [linker script](https://stnolting.github.io/neorv32/#_linker_script)? The compiler might (for sure) emit code that is **not** position-independent. Hence, the linker scripts is always required to correspond to the physical memory layout.

SteveZ-Cal, 2023-11-21:
Thanks a lot!!

That resolved all of issues I encountered :)

closed 2023-11-21 by SteveZ-Cal