stnolting/neorv32 #750: Nexys 4 DDR FPGA board flashing zephyr RTOS hello_world no output on default UART

cryptobharathi, 2023-12-11:
I have build the the neorv32 for Nexys 4 DDR FPGA board with the TCL provided - https://github.com/stnolting/neorv32-setups/blob/main/vivado/nexys-a7-test-setup/create_project.tcl with vivado version 2019.2

I am able to make the bit file, flashing works fine, i am able to see the bootloader screen and i aboted autoboot.

I am able to successfully load the below two software programs, works as expected
1. demo_blink_led - https://github.com/stnolting/neorv32/tree/main/sw/example/demo_blink_led
2. hello_world - https://github.com/stnolting/neorv32/tree/main/sw/example/hello_world

Attached screenshot for reference
[image]

I am working with zephyr RTOS for other target hardware for more than 2 years.
I tried running zephyr RTOS hello_world application (Uploading via UART) on neorv32 as described - https://docs.zephyrproject.org/latest/boards/riscv/neorv32/doc/index.html 

I see the upload works fine, but i dont see any output in the UART console (screenshot attached)
I have used zephyr latest version (3.5.99)
[image]

Looking forward for support.

stnolting, 2023-12-11:
Hey there!

We have changed the processor's memory map since [version 1.8.6](https://github.com/stnolting/neorv32/blob/main/CHANGELOG.md). The upstream Zephyr port is not compatible with that (yet). However, there is a PR to update the port's device tree according to the new memory map: https://github.com/zephyrproject-rtos/zephyr/pull/65627

cryptobharathi, 2023-12-12:
@stnolting 
Thanks for the update.

I have applied the PR (https://github.com/zephyrproject-rtos/zephyr/pull/65627) as patch

[image]

I have made a fresh build after applying the patch. 

I have also used  release 1.9.2 (https://github.com/stnolting/neorv32/releases/tag/v1.9.2) for generating the bit stream.

There is no output in the UART console (attached reference screenshot)

[image]

Please note, i have rename the `zephyr_exe.bin` to `neorv32_exe.bin`

stnolting, 2023-12-23:
Looks good so far. What kind of application program are you compiling? Maybe the GPIO port also needs to be updated? 🤔 

**FYI** I just saw that GitHub user @VolodymyrSvintozelskyi made a PR (https://github.com/zephyrproject-rtos/zephyr/pull/66722) to update the processor's Zephyr port according to the latest memory layout 🎉

cryptobharathi, 2023-12-24:
@stnolting 
I am trying to compile zephyr hello world application.

Am i missing something ? 

Let me do another fresh re-try and update you the status.

VolodymyrSvintozelskyi, 2023-12-24:
Hi @cryptobharathi ,
The issues you're faced with are likely due to memory mapping changes, accumulated in the different neorv32 releases starting from 1.8.7.  
I've fixed this within Zephyr for neorv32 of version 1.9.2, as was pointed out by @stnolting.

To try this out, you must compile my branch of zephyr, which hasn't yet been merged. To do that, use

`west init -m https://github.com/VolodymyrSvintozelskyi/zephyr.git --mr vsvintoz_neorv32_offsets_update zephyrproject
`

instead of normal 

`west init`

Everything else should be as in the official guide on the Zephyr RTOS installation webpage.

cryptobharathi, 2023-12-24:
@VolodymyrSvintozelskyi,

Thanks for pointing out, sure will make a build out of the MR branch you referred.

Will keep you posted once done.

cryptobharathi, 2023-12-25:
@VolodymyrSvintozelskyi 

I have tried the following, 

clone the branch 
```
linumiz@linumiz-lenovo:~/letmetry/c/riscv/zephyrproject/zephyr/samples/hello_world/build/zephyr$ git branch
* vsvintoz_neorv32_offsets_update
```

build 
```
west build -b neorv32 . -- -DCMAKE_PROGRAM_PATH=/home/linumiz/letmetry/c/riscv/neorv32/sw/image_gen/
```

list the build
```
linumiz@linumiz-lenovo:~/letmetry/c/riscv/zephyrproject/zephyr/samples/hello_world/build/zephyr$ ls -lrt
total 2720
-rw-rw-r--  1 linumiz linumiz    910 Dec 25 11:19 snippets_generated.cmake
drwxrwxr-x  3 linumiz linumiz   4096 Dec 25 11:19 include
-rw-rw-r--  1 linumiz linumiz   5058 Dec 25 11:19 zephyr.dts.pre
-rw-rw-r--  1 linumiz linumiz    572 Dec 25 11:19 zephyr.dts.d
-rw-rw-r--  1 linumiz linumiz 325900 Dec 25 11:19 edt.pickle
-rw-rw-r--  1 linumiz linumiz   3797 Dec 25 11:19 zephyr.dts
-rw-rw-r--  1 linumiz linumiz  20570 Dec 25 11:19 dts.cmake
drwxrwxr-x  3 linumiz linumiz   4096 Dec 25 11:19 kconfig
drwxrwxr-x  3 linumiz linumiz   4096 Dec 25 11:19 misc
drwxrwxr-x  5 linumiz linumiz   4096 Dec 25 11:19 cmake
-rw-rw-r--  1 linumiz linumiz    891 Dec 25 11:19 runners.yaml
drwxrwxr-x  6 linumiz linumiz   4096 Dec 25 11:19 CMakeFiles
-rw-rw-r--  1 linumiz linumiz   3400 Dec 25 11:19 cmake_install.cmake
drwxrwxr-x  5 linumiz linumiz   4096 Dec 25 11:19 arch
drwxrwxr-x  8 linumiz linumiz   4096 Dec 25 11:19 lib
drwxrwxr-x 23 linumiz linumiz   4096 Dec 25 11:19 subsys
drwxrwxr-x  4 linumiz linumiz   4096 Dec 25 11:19 soc
drwxrwxr-x  4 linumiz linumiz   4096 Dec 25 11:19 boards
drwxrwxr-x 14 linumiz linumiz   4096 Dec 25 11:19 drivers
-rw-rw-r--  1 linumiz linumiz   7624 Dec 25 11:19 linker_zephyr_pre0.cmd.dep
-rw-rw-r--  1 linumiz linumiz  12474 Dec 25 11:19 linker_zephyr_pre0.cmd
-rw-rw-r--  1 linumiz linumiz 802406 Dec 25 11:19 libzephyr.a
drwxrwxr-x  3 linumiz linumiz   4096 Dec 25 11:19 kernel
-rw-rw-r--  1 linumiz linumiz 206778 Dec 25 11:19 zephyr_pre0.map
-rwxrwxr-x  1 linumiz linumiz 383076 Dec 25 11:19 zephyr_pre0.elf
-rw-rw-r--  1 linumiz linumiz   7612 Dec 25 11:19 linker.cmd.dep
-rw-rw-r--  1 linumiz linumiz  12464 Dec 25 11:19 linker.cmd
-rw-rw-r--  1 linumiz linumiz   2048 Dec 25 11:19 isr_tables.c
-rw-rw-r--  1 linumiz linumiz 206519 Dec 25 11:19 zephyr_final.map
-rwxrwxr-x  1 linumiz linumiz 382708 Dec 25 11:19 zephyr.elf
-rw-rw-r--  1 linumiz linumiz 206519 Dec 25 11:19 zephyr.map
-rw-rw-r--  1 linumiz linumiz  30606 Dec 25 11:19 zephyr.hex
-rwxrwxr-x  1 linumiz linumiz  10844 Dec 25 11:19 zephyr.bin
-rw-rw-r--  1 linumiz linumiz  35731 Dec 25 11:19 zephyr.vhd
-rw-rw-r--  1 linumiz linumiz   4441 Dec 25 11:19 zephyr.stat
-rw-rw-r--  1 linumiz linumiz  10856 Dec 25 11:19 zephyr_exe.bin
```
i have rename the `zephyr_exe.bin` to `neorv32_exe.bin`

Uploaded the `neorv32_exe.bin` file,
[image]

command `u` to upload the file,
[image]

command `e` to execute
[image]

No output seen on serial console.

Am i missing something here ?

VolodymyrSvintozelskyi, 2023-12-25:
@cryptobharathi, maybe try to increase the sizes of IMEM & DMEM to 64 and 32 KB respectively? 

Apart from that, everything looks fine to me...

cryptobharathi, 2023-12-26:
@VolodymyrSvintozelskyi 

You are correct, it was a size issue. 

After update the size of IMEM and DMEM to 64 and 32 KB, it works as expected

[image]

cryptobharathi, 2023-12-26:
Thanks for the support @stnolting and @VolodymyrSvintozelskyi 

Closing the issue

closed 2023-12-26 by cryptobharathi

stnolting, 2023-12-31:
Great to see this has been resolved!
@VolodymyrSvintozelskyi thank you very much for your support!