stnolting/neorv32 #757: Illegal instruction in coremark

mahdi259, 2023-12-22:
**Describe the bug**
Hi,
I used neorv32_test_setup_bootloader and compiled coremark example program once with: 
`make exe MARCH=rv32i`
and then with:
`make exe MARCH=rv32imc`

The problem is that it prints errors of illegal instructions in terminal for both cases. Is there any suggestion?

**To Reproduce**
One should use neorv32_test_setup_bootloader as topmodule, ISE for synthesis and compilation, Tera Term VT for UART communication and Posedge one board for implementation. The board has a Spartan 6 SLX9 chip on it.

**Expected behavior**
I expected that both cases were executed normally.

**Environment:**
 I used RISC-V pre-built compiler available at:

https://github.com/stnolting/riscv-gcc-prebuilt/releases/download/rv32i-4.0.0/riscv32-unknown-elf.gcc-12.1.0.tar.gz

**Hardware:**
 neorv32_test_setup_bootloader has M and C extensions enabled by default.

[image]

Thanks for this great projects...

stnolting, 2023-12-23:
Hey there!

An illegal instruction exception gets raised when
* a program does not have the rights to conduct certain operations (this is not relevant here as the program is execute in privileged machine mode)
* a program is executing instructions that are not available for example because the according ISA extension is not implemented

When CoreMark is starting it tries to setup some hardware performance monitors (HPMs) for benchmarking the run:

https://github.com/stnolting/neorv32/blob/745eec3c3ab030e564c48d88caa4d5edd679da21/sw/example/coremark/core_portme.c#L146-L159

It seems like your hardware configuration has no HPMs enabled (`CPU_EXTENSION_RISCV_Zicond` is false and/or `HPM_NUM_CNTS` is zero). Unfortunately, changing the compiler's `MARCH` flags does not help here as the according isntructions are "hard coded".

Anyway, I will add some code to the setup function to avoid those confusing exceptions when no HPMs are available... 🙈

stnolting, 2023-12-23:
I've updated the code base. There should be no more HPM-related exceptions (even if none are implemented).

mahdi259, 2023-12-23:
Thanks for your reply.
I examined coremark example with your modification and it run successfully.
I also added codes below in top module. I think you meant zihpm right?
```
CPU_EXTENSION_RISCV_Zihpm    => true
HPM_NUM_CNTS			      => 13
```
Enabling hardware performance monitoring lets coremark print more intuitive information.
[image]

[image]

stnolting, 2023-12-23:
> I also added codes below in top module. I think you meant zihpm right?

That is correct. These two generics control the implemented HPMs.

> Now the former coremark without new modification was also run successfully. Enabling hardware performance monitoring lets coremark print more intuitive information.

Looks great! :+1:

mahdi259, 2023-12-24:
If you agree, we can close this issue. In fact it was really a question and not a bug.
Thanks a lot

closed 2023-12-24 by stnolting