stnolting/neorv32 #774: CSR reads fail for un-implemented features like HPM counters

mikaelsky, 2024-01-25:
**Describe the bug**
When performing a CSR read command from a CSR register that is tied to a disabled feature, like HPM counters, the core traps with an illegal instruction. 
RISCV spec requires that all un-implemented CSRs read back 0, which the core does for everything but CSRs tied to disabled features.

--- Log out from our RVVI trace compare:
Warning (RISCV_CSR_NWA) CPU 'refRoot/cpu' 0x0000026a c9d51073 csrw    hpmcounterh29,x10: CSR has no write access
Warning (RISCV_CSR_NWA) CPU 'refRoot/cpu' 0x0000026e c9e51073 csrw    hpmcounterh30,x10: CSR has no write access
Warning (RISCV_CSR_NWA) CPU 'refRoot/cpu' 0x00000272 c9f51073 csrw    hpmcounterh31,x10: CSR has no write access
Info (IDV) Instruction executed prior to mismatch '0x502(__neorv32_rte_core+11e): 30200073 mret'
Error (IDV) PC mismatch (HartId:0, PC:0x00000276 main+148):
Error (IDV) Mismatch 0>
Error (IDV)   . dut:0x000003e4 __neorv32_rte_core+0
Error (IDV)   . ref:0x00000276 main+148
Error (IDV) Insn. bit pattern mismatch (HartId:0, PC:0x00000276 main+148):
Error (IDV) Mismatch 1>
Error (IDV)   . dut:715d     addi    x2,x2,-80
Error (IDV)   . ref:c03027f3 csrr    x15,hpmcounter3
Error (IDV) GPR register value mismatch (HartId:0, PC:0x00000276 main+148):
Error (IDV) Mismatch 2> GPR x2
Error (IDV)   . dut:0x00003f7c
Error (IDV)   . ref:0x00003fcc (not updated)
Error (IDV) Mismatch 3> GPR x15
Error (IDV)   . dut:0x00003ff0 (not updated)
Error (IDV)   . ref:0x00000000
Error (IDV) CSR register value mismatch (HartId:0, PC:0x00000276 main+148):
Error (IDV) Mismatch 4> CSR 300 (mstatus)
Error (IDV)   . dut:0x00001800 SD:0 XS:0(Off) FS:0(Off) MPP:3 VS:0(Off) MPIE:0 MIE:0
Error (IDV)   . ref:0x00001880 SD:0 XS:0(Off) FS:0(Off) MPP:3 VS:0(Off) MPIE:1 MIE:0 (not updated)
Error (IDV) Mismatch 5> CSR 343 (mtval)
Error (IDV)   . dut:0xc03027f3
Error (IDV)   . ref:0xc9f51073 (not updated)
Error (IDV) tb.riscv_tb.idv_trace2api.state_compare @ 1080420.00 ns: MISMATCH

The reference and the dut get out of sync as the dut takes an illegal instructions trap after the MRET instruction.

**To Reproduce**
Disable HPM counters.
Read an HPM counter CSR.

Example code:
 26e:   c9e51073                csrw    hpmcounter30h,a0
 272:   c9f51073                csrw    hpmcounter31h,a0
  asm volatile ("csrr %[result], %[input_i]" : [result] "=r" (csr_data) : [input_i] "i" (csr_id));
 276:   c03027f3                csrr    a5,hpmcounter3
 27a:   c04027f3                csrr    a5,hpmcounter4

The 2 CSR writes must cause an illegal instruction trap
The 2 CSR read must return 0 and not cause an illegal instruction trap.

**Expected behavior**
CSR reads must always return 0.

**Screenshots**
NA

**Environment:**
 - RHEL 7
 - GCC Version 12.2.0
 - Libraries used newlib

**Hardware:**
 - Hardware version (1.9.3.0)

**Additional context**
I am currently running a full regression test to ensure the bug fix below doesn't break anything else. If regressions pass I will submit a pull request for this issue.

The fix is to basically gate csr_reg_valid with csr.re_nxt to ensure that csr_reg_valid is only used for CSR writes.

neorv32_cpu_control.v 
```
        -- The RISCV spec requires that CSR registers are always readable and should return 0 if not implemented.
        -- To ensure that CSR reads do not result in an illegal instruction trap we gate csr_reg_valid with csr.re_nxt
        elsif ((csr_reg_valid = '0') and (csr.re_nxt = '0')) or (csr_rw_valid = '0') or (csr_priv_valid = '0') or -- invalid CSR access?
              (execute_engine.ir(instr_funct3_msb_c downto instr_funct3_lsb_c) = funct3_csril_c) then -- invalid CSR access instruction?
          illegal_cmd <= '1';
        else
          illegal_cmd <= '0';
        end if;

```

stnolting, 2024-01-25:
> RISCV spec requires that all un-implemented CSRs read back 0, which the core does for everything but CSRs tied to disabled features.

This depends on the priv. spec. version I think.

This is what the current (ratified) priv. spec. says:

> Attempts to access a non-existent CSR raise an illegal instruction exception.

[image]

Right now, this is how CSR accesses are handled by NEORV32.
* If `CPU_EXTENSION_RISCV_Zihpm` is disabled, any access to HPM-related CSR will trap.
* If `CPU_EXTENSION_RISCV_Zihpm` is enabled but `HPM_NUM_CNTS` is zero any read access to a HPM-related CSR will return zero and any write access to a HPM-related CSR will basically do nothing.
* If `CPU_EXTENSION_RISCV_Zihpm` is enabled but `HPM_NUM_CNTS` is non-zero any read access to a HPM-related CSR above "HPM_NUM_CNTS" will return zero and any write access to a HPM-related CSR above "HPM_NUM_CNTS" will basically do nothing.

However, the upcoming priv. spec. version 1.13 relaxes this and makes this requirement more platform-specific:

[github.com/riscv/riscv-isa-manual/issues/1116: Accessing an unimplemented CSR is reserved, not necessarily illegal](https://github.com/riscv/riscv-isa-manual/issues/1116)

mikaelsky, 2024-01-25:
Interesting. I'm currently on spec 1.13 I think, but release 04/2023. Still has the same wording:
[image]

Now where this becomes interesting is that our reference model actually flags this as an error, meaning they expect the read to pass and not trap. But I see your point though.

I think the conflicts ties in the spec volume 1 chapter 11 Zicsr where the CSRRS instruction specifies that it shall only cause an illegal instruction on a write but never on a read... sorta as it also states its only for "read only CSRs" so yeah fun times :) So not really specifying what should happen on an un-implemented CSR
[image]

It further notes that CSRR is mapped in CSRRS:
[image]

So reading volume could be read as: Issuing a CSRR read from any CSR should not result in an illegal instruction trap... but I digress.

stnolting, 2024-01-25:
The specs are really horrible at this point... 🙈 I've followed a lot of discussions about this and in the end the answer was always something like "but that is platform-defined". So I am not sure which way to go here.

I think it is quite handy to trap if a program is accessing an unimplemented CSR. This indicates that the hardware configuration the software was written for does not match the actual hardware configuration - so it is an actual software _fault_.

> [...] the CSRRS instruction specifies that it shall only cause an illegal instruction on a write but never on a read [...]

As far as I understand the specs this is only true if `rs` or `uimm5` is zero, right? And I think this is correctly implemented in NEORV.

However, this is really a hard one. So, how should we handle this?

mikaelsky, 2024-01-25:
Good question. I'm not actually disagreeing with your interpretation. I've taken action to check with our reference model vendor to get their perspective and see why they opted for the CSR read must always return 0 interpretation. Sometimes its good to hear someone else's read on it. For now I'm leaning more towards not a bug, than a bug. But let me get someone else perspective.

Where I think folks could stumble is this part:
> For both CSRRS and CSRRC, if rs1=x0, then the instruction will not write to the CSR at all, and so shall
> not cause any of the side effects that might otherwise occur on a CSR write, nor raise illegal
> instruction exceptions on accesses to _**read-only**_ CSRs.

As the GCC generated CSRR is shorthand for CSRRS rd, csr, rs1 with rs1 = x0 so basically 0 per the above section.
Now what annoys me is the "read-only" section. I can see how if you squint and sort miss the sentence a bit you would read "nor raise illegal... on read". In reality I think its "if you read a read-only CSR register you want get an illegal instruction trap.

Where it gets murky is the "any access to an undefined CSR" in the volume II as it collides a bit with the above.

For now I have an internal "fix" mostly to get the dut to match the ref to allow my DV team to progress. I'll hold back on issuing a pull request for this one until we get more clarification.

mikaelsky, 2024-01-26:
After some internal reviews the conclusion is that the current behavior of the core when accessing undefined CSRs is correct. It should be illegal instructions. We will fix our internal model to reflect this.
The issue can be closed :)

mikaelsky, 2024-01-26:
Issue can be closed as the current behavior of the core is correct as per the RISCV specification.

closed 2024-01-26 by mikaelsky

stnolting, 2024-01-26:
> Issue can be closed as the current behavior of the core is correct as per the RISCV specification.

Great to hear!