stnolting/neorv32 #781: Cannot upload  neorv32_exe.bin for demo CFS on NexysA7

JC-LL, 2024-01-30:
**Describe the bug**
cannot upload neorv32_exe.bin for demo CFS with nexys-a7-test-setup. The upload process freezes

**To Reproduce**
Switch IO_CFS_EN=true in neorv32_top.vhd then restart synthesis : 
cd neorv32-setups/vivado/nexys-a7-test-setup/
vivado -mode tcl -source create_project.tcl # and  push bitstream 
cd neorv32/sw/example/demo_cfs 
make exe 
run neo on FPGA and interrupt the boot process to 'u'pload  neorv32_exe.bin (me : using gtkterm or minicom)
the process freezes

**Expected behavior**
The upload should finish.

**Screenshots**
[image]

umarcor, 2024-01-30:
FTR, we are also having issues with the bootloader on an Arty-A7. For now, we are disabling the bootloader so that the app is started straightaway. However, that requires resynthesising if we want to change the software.

/cc @Unike267

stnolting, 2024-01-30:
Hey @JC-LL!

Hmm this problem sounds familiar...

Does the upload only freeze when you upload your _specific_ program? Can you try compiling a simpler one (e.g. [sw/example/hello_world](https://github.com/stnolting/neorv32/tree/main/sw/example/hello_world)) and try uploading that?

I guess your executable is larger than 4kB, which is a known problem with GTKTerm: https://github.com/stnolting/neorv32/pull/215#issuecomment-975836572

Can you try a different terminal program like cutecom or picocom?

@umarcor

What terminal program are you using?

JC-LL, 2024-01-30:
Hello @stnolting !

First, congrats for this interesting project. 

Yes, smaller binary program work fine, like the LED one that works like a charm.

Gtkterm is the issue here. However, I'v just tried picocom, that also seem to freeze (I hope I have used it correctly for sending the code). On my Linux Mint, cutecom does not seem to simply react when booting Neorv32... 

I'm still investigating.

JC-LL, 2024-01-30:
Ok, cutcom now reacts, but gives this :

CMD:> u
Awaiting neorv32_exe.bin... <0x07>
ERR_EXE

JC-LL, 2024-01-30:
Things are a bit weird. My first attempt with picocom (with the correct neorv32_exe.bin) went wrong...

I just retried, and I finally managed to upload the executable. Hurray !

Well...not perfect still.

```
CMD:> u
Awaiting neorv32_exe.bin... 
*** file: /home/jcll/JCLL/exp/neorv32/sw/example/demo_cfs/neorv32_exe.bin
$ ascii-xfr -s -n /home/jcll/JCLL/exp/neorv32/sw/example/demo_cfs/neorv32_exe.bin
ASCII upload of "/home/jcll/JCLL/exp/neorv32/sw/example/demo_cfs/neorv32_exe.bin"

*** exit status: 0 ***
OK
CMD:> e
Booting from 0x00000000...

Error! No CFS synthesized!
```

stnolting, 2024-01-30:
> I just retried, and I finally managed to upload the executable. Hurray !

Great to hear! 🎉

> Well...not perfect still. [...] `Error! No CFS synthesized!`

Seems like the CFS was not enabled before synthesizing. You need to set the according configuration generic to "true" in your instance of the processor:

https://github.com/stnolting/neorv32/blob/8e5223485da326448500f97e4c3ebbe45f845470/rtl/core/neorv32_top.vhd#L139

JC-LL, 2024-01-30:
That surprised me much because, IO_CFS_EN was indeed correctly set to **true**...That was my initial goal !  I probably made a mistake renaming bitstream. I'll try again tomorrow.

closed 2024-01-30 by JC-LL

stnolting, 2024-02-03:
> It works. It seems that modifying generic definition (which I did) is not sufficient. I don't see why. It really needs to be set in the generic map

Right, the instantiation will override the defaults from the entity declaration.

> For completeness, here is the uart log 

:+1: :rocket:

Unike267, 2024-02-05:
Hi @stnolting 

> Which terminal program are you using?

We are using **cutecom** terminal. The problem was that the type of file that we set in cutecom was Script and if you will set this type the cutecom repons with **ERR_EXE**.

So to fix it you should be set **Plain** type. As shown in the following image:

[image]

/cc @umarcor

stnolting, 2024-02-05:
Thanks for the information!