stnolting/neorv32 #787: Instruction to halt TRNG operation

mahdi259, 2024-02-02:
Hi @stnolting

**Is your feature request related to a problem? Please describe.**
About TRNG module, As you stated in datasheet, it's possible to keep enable_i signal low to reduce dynamic power of the system. Is there any specific instruction that enables or disables TRNG? I looked at demo_trng example and it seems that you didn't stop TRNG there.

**Describe the solution you'd like**
Using **start** and **stop** commands in demo_trng example.

Thanks

mahdi259, 2024-02-02:
I found instructions to enable/disable TRNG unit in 
_**neorv32-main/sw/lib/source/neorv32_trng.c**_
Thanks

closed 2024-02-02 by mahdi259

stnolting, 2024-02-02:
The TRNG is a SoC module and not part of the actual CPU core so there is no dedicated instruction to turn on/off the TRNG.

> I found instructions to enable/disable TRNG unit in
neorv32-main/sw/lib/source/neorv32_trng.c

Right! Just for the records: here is the link to the TRNG's software "API" documentation: https://stnolting.github.io/neorv32/sw/neorv32__trng_8c.html