stnolting/neorv32 #789: ERR_EXE when uploading neorv32_exe.bin for demo_blink_led on DE2-115 Board

DatNguyen97-VN, 2024-02-03:
Dear Admin, 

**Describe the bug**
When I send neorv32_exe.bin file on Tera Term , I receive a notification of ERR_EXE

**To Reproduce**
```
 cd demo_blink_led
make exe
```
result to 
```
Memory utilization:
   text    data     bss     dec     hex filename
   1080       0       0    1080     438 main.elf
Compiling ../../../sw/image_gen/image_gen
Executable (neorv32_exe.bin) size in bytes:
1092
```
Finally, I connected Tera Term  to de2 and when sending the neorv32_exe.bin file to it an error appeared as err_exe

[image]

Please help me.

Thanks,
Dat Nguyen

stnolting, 2024-02-03:
Hey @DatNguyen97-VN!

`ERR_EXE` indicates a signature error. So maybe you are just uploading the wrong file (see #781). Which file are you uploading (`neorv32_exe.bin`, **not** `main.bin`)?

DatNguyen97-VN, 2024-02-03:
Thank you very much @stnolting. I forgot to select "option binary" so I got err_exe.
[image]

closed 2024-02-03 by DatNguyen97-VN