stnolting/neorv32 #847: Syntax error near "context".

mahdi259, 2024-03-14:
**Describe the bug**
The updated testbench contains `context` keyword that is not supported in older software like ISE. 

**To Reproduce**
I used neorv32_tb.vhd file as benchmark. 

**Expected behavior**
Normal simulation.

**Screenshots**
[image]

mahdi259, 2024-03-14:
Oh. I think this testbench file is only usable by VUnit, right?

stnolting, 2024-03-14:
The testbench files in [`sim`](https://github.com/stnolting/neorv32/tree/main/sim) **require VHDL-2008** (or newer) as standard. Please use the functional-equivalent "simple" testbench in [`sim/simple`](https://github.com/stnolting/neorv32/tree/main/sim/simple) for toolchain that do not support this standard.

Btw, we really should add a note regarding this to the README 😅

stnolting, 2024-03-14:
> Oh. I think this testbench file is only usable by VUnit, right?

Well, yes, and VUnit requires VHDL-2008 (as far as I can see).

mahdi259, 2024-03-14:
Thanks for your reply.

closed 2024-03-14 by mahdi259

stnolting, 2024-03-14:
You're welcome! :)

umarcor, 2024-03-15:
VUnit uses VHDL 2008 by default, but it does have a VHDL 1993 mode. The problem is that simulators from AMD/Xilinx traditionally do not support enough of VHDL 2008 to use the default, and neither support enough of VHDL 1993 to use the compatility mode. So, in order to use those tools, a non-standard variation of a 30y old version of the language needs to be used.

I would argue that it is less of a headache to use some simulator which does support VHDL 2008, such as GHDL or NVC. For synthesis, `ghdl synth` allows converting VHDL 2008 sources to some flavour that AMD/Xilinx can understand.

stnolting, 2024-03-15:
> For synthesis, ghdl synth allows converting VHDL 2008 sources to some flavour that AMD/Xilinx can understand.

Very good point!