stnolting/neorv32 #848: Changing name of memory size variables

mahdi259, 2024-03-14:
**Is your feature request related to a problem? Please describe.**
I think `MEM_INT_DMEM_SIZE` and `MEM_INT_IMEM_SIZE` should be general variables that should apply even we use exrernal memories i.e. when `MEM_INT_DMEM_EN` and `MEM_INT_IMEM_EN` are false. 

**Describe the solution you'd like**
So `INT` refers to Internal memory is it right? Can we remove `INT` to generalize these variables?

stnolting, 2024-03-14:
> So INT refers to Internal memory is it right?

That's right.

> Can we remove INT to generalize these variables?

What do you mean with "generalize"? All the generics are used to configure **processor-internal** features. There is no way to use the generics of a module to configure something that is outside of that module (like **processor-external** memory).

mahdi259, 2024-03-14:
I thought that the memory sizes are stored in `neorv32.neorv32_sysinfo` in both cases (Internal or external memory) and the size variables (`MEM_INT_IMEM_SIZE` and `MEM_INT_DMEM_SIZE`) should be initialized with accurate values in both cases.

stnolting, 2024-03-14:
Ah ok, I think I understand your use case now :wink:

The SYSINFO module is intended to provide information about the processor configuration itself (i.e. all the **internal** stuff). Hence, only the internal memory configuration is available from that module.

If a setup uses processor-external memories (note that there could be a lot of different memories attached to processor) the designer is responsible to provide any kind of hardware discovery mechanism (if required at all).

For example, you could add another memory to the external bus system that features a tiny ROM (just like SYSINFO) that shows the sizes (and maybe the base addresses) of all the memories that you have attached to the core.

closed 2024-03-14 by mahdi259