stnolting/neorv32 #856: Makefile: USER_FLAGS only for C++

jpf91, 2024-03-21:
**Is your feature request related to a problem? Please describe.**

When compilling C++ code, I sometimes need flags which are not valid for C code, for example `-fno-rtti -fno-use-cxa-atexit -fno-threadsafe-statics`. I can just pass them to `USER_FLAGS`, but then they also get passed when compiling C files. in that case, GCC complains with warnings:
`cc1: warning: command-line option '-fno-rtti' is valid for C++/D/ObjC++ but not for C`

**Describe the solution you'd like**
Split `CC_OPTS` into `C_OPTS` and `CXX_OPTS`. Introduce `CXX_USER_FLAGS` and only use those for `CXX_OPTS` and in cpp compilation.

Though I guess there are other options to implement this as well. In the end, anything that allows specifying flags in the user Makefile that only get passed to the C++ compiler should work.

jpf91, 2024-03-22:
In addition, it would be nice to have a variable only affecting link flags. AFAIK `-fwhole-program` behaves differently at compile and link time.

stnolting, 2024-03-23:
Hey @jpf91!

Sounds like a good idea! However, I am not an experienced C++ user... 🙈 Would you draft a PR so we can see which way this is going? :wink:

jpf91, 2024-03-24:
Hey @stnolting 

I've drafted a PR in #862 . It seems those variables even have somewhat standardized names, I always thought it's just a convention.

I didn't any flags yet, as that is a completely different and probably controversial topic :-)
For example, I recently experimented with -flto and -fwhole-program which worked fine when executing code from IRAM but broke execution in XIP mode. I still don't know why ;-)

But this change alone should enable users to specify any C++ and/or linker flags by themselves. I guess we can continue the discussion in #862

PR #862 (2024-03-26): Support tool-specific standard flags in makefile
This introduces the following new makefile variables: `CFLAGS` for C, `CXXFLAGS` for C++, `ASFLAGS` for Assembler and `LDFLAGS` for the linker. Names follow  conventions from  https://www.gnu.org/software/make/manual/html_node/Implicit-Variables.html

These variables can then be set either in the user application makefile (before including `common.mk`) or on the command line when calling make.

Fixes #856
closed 2024-03-24 by jpf91