stnolting/neorv32 #866: A question on simple testbech

mahdi259, 2024-03-31:
Hi

**My question**
I imported NeoRV32 micro controller into a questa simulation project and used `sim/simple/neorv32_tb.simple.vhd` and `sim/simple/uart_rx.simple.vhd` for simulation. I wondered why UART doesn't show activity. Here `int_bootloader` is disabled and `application_image` is for hello_world. 

**To Reproduce**
Using `sim/simple/neorv32_tb.simple.vhd` and `sim/simple/uart_rx.simple.vhd` files for simulation in questasim.

**Expected behavior**
UART activity for printing hello world!

**Screenshots**
[image]

stnolting, 2024-03-31:
Hey there!

Your warning.txt looks fine so far. How did you compile/install the hello_world memory image?

mahdi259, 2024-03-31:
I use this command
`make clean MARCH=rv32imc_zicsr_zifencei install`
and copy application_image content into `neorv32_application_image.vhd`.

stnolting, 2024-03-31:
Ah I see, that might be the problem.

According to your warnings.txt the CPU is configured **without** the `C` extension:

> Note: [NEORV32] CPU ISA: rv32imabu_zicsr_zicntr_zicond_zifencei_zfinx_zihpm_zxcfu_sdext_sdtrig_smpmp

So if you use the same ISA configuration (no "c") for compiling the code the simulation should work (better use `clean_all` here to remove all artifacts from previous builds):

```
$ make clean_all MARCH=rv32im_zicsr_zifencei install
```

By the way, the `install` target already copies the local memory initializationimage into the `core/rtl` folder overriding the default `neorv32_application_image.vhd`.

mahdi259, 2024-03-31:
You're right. Thanks a lot.
[image]

stnolting, 2024-04-01:
Looks better now! :wink:

I guess we can close this?

mahdi259, 2024-04-02:
Yes thanks.

closed 2024-04-02 by mahdi259