stnolting/neorv32 #878: ASIC Processor boot conccept

ucycg, 2024-04-18:
I'm currently exploring the possibility to create an ASIC from a specific NeoRV32 configuration. As I'm still a student without much experience in digital design I still have a lot of things to figure out on this endeavour. 

One thing I currently came across, as it is fundamental for the ASIC processor to be actually usable is the boot concept. I was always thinking I would use the bootloader to upload binaries from an external flash via the auto boot SPI process. But today I realized that the bootloader itself is contained in the neorv32_boot_rom.vhd which isn't exactly synthesizable either.

closed 2024-04-18 by ucycg

ucycg, 2024-04-19:
Yeah so I managed to synthesize a netlist in our tech process and simulated the startup of the bootloader successfully which tells me that the boot_rom is successfully synthesized in the GTL netlist. I would still like to find out how much area is occupied by that memory, which i couldnt figure out until now sadly.
[image]

stnolting, 2024-04-21:
That looks great! :+1:
Keep us updated if you like :wink:

stnolting, 2024-04-23:
Sounds like a cool project!

If you "just" want to support read accesses (i.e. no flash write) you only need to implement the READ command, so the logic should be quite simple. Hit me up if I can help you in any way.

jtplaarj, 2024-04-24:
Picorv32 has such a simulation module por XIP: https://github.com/YosysHQ/picorv32

I have successfully used it with neorv32.

stnolting, 2024-04-24:
You mean this one? https://github.com/YosysHQ/picorv32/blob/main/picosoc/spiflash.v

I thinks it looks quite promising.

jtplaarj, 2024-04-25:
Yes. I have done small changes, it looks like neorv32 implementation of XIP and that of picorv32 are not exactly the same. the spi simulator from picorv32 waits for one command to initialize the flash, while neorv32 does not send it. Because of that, I have changed line 63 and initialize `reg powered_up = 1` to avoid the issue.

stnolting, 2024-04-25:
So the SPI model is waiting for some "get out of standby mode" command?

Is that a common command understood by most flash chips? If yes, then we should add this to the bootloader. 🤔

ucycg, 2024-04-26:
That was a thing that I already noticed while implementing my SPI flash. Maybe you are talking about somehting else, but in the bootloader code there is already a spi_flash_wakeup() function called sending cmd 0xAB, which isn't mentioned in the datasheet currently.

stnolting, 2024-04-26:
> Maybe you are talking about somehting else, but in the bootloader.c code there is already a spi_flash_wakeup() function [...]

Oh, right... Seems like I have forgotten that 🙈😅

https://github.com/stnolting/neorv32/blob/8ca618c351b4ff42b8633488e88ab9e0f1723bed/sw/bootloader/bootloader.c#L177

ucycg, 2024-04-26:
I wrote a crude simple flash emulation module and in simulation the auto boot scenario is working :smile:  The NeoRV32 now 
reads the data from the flash module via SPI and starts executing the blink_led_example binary. 
[image]
One thing I don't quite understand currently is that a first setup where I decreased the memory to 8KB IMEM and 2KB DMEM was loading the software as well, but couldn't execute it. I only changed the memory size inside my wrapper module. I dont quite understand what else I have to change to use different memory sizes. I there something in the linker that I have to change ? Or do I have to respect the memory sizes when I compile my application? I can also formulate this in another issue if that is better.

stnolting, 2024-04-27:
> I wrote a crude simple flash emulation module and in simulation the auto boot scenario is working 😄 The NeoRV32 now
reads the data from the virtual flash module via SPI and starts executing the blink_led_example binary.

That's great! 🚀

> One thing I don't quite understand currently is that a first setup where I decreased the memory to 8KB IMEM and 2KB DMEM was loading the software as well, but couldn't execute it. I only changed the [...]

You will need to recompile your application if you change the memory size of the hardware (actually only if you shrink it). Since this is a common pitfall, it might be good to clarify this in another issue.