stnolting/neorv32 #884: Neorv32 synthesis in Design compiler

mahdi259, 2024-04-24:
Hi,
This is actually a question. I have synthesized Neorv32 in Design Compiler. My problem is with the resulting design file that I produce with design compiler. It actually gives 'X' signal in simulation for some outputs. Have you ever synthesized Neorv32 in Design compiler @stnolting?
Thanks

mahdi259, 2024-04-25:
Yes, I synthesized the same neorv32_top that I used in neorv32_tb_simple. The UART signals are getting 'X'. If I solve it, I will describe that here.

mahdi259, 2024-04-25:
The problem was with hold time of implementation. I relaxed the constraints and it worked.

closed 2024-04-25 by mahdi259

stnolting, 2024-04-25:
Great to hear! :+1:

mahdi259, 2024-05-03:
Hi @stnolting 
Do you have a friend that is familiar with synopsis DC?

stnolting, 2024-05-04:
Unfortunately not. 🙈
But maybe someone else here can help?!

mahdi259, 2024-05-04:
Ok thanks

mahdi259, 2024-05-07:
For anyone who reads this later. Generating sdf file of synthesized design and using it for simulation in Questasim improves maximum operational frequency of design.