stnolting/neorv32 #890: UART0 issue after #883 upd

Polaris-III, 2024-05-02:
**Describe the bug**
After #883 i discover that neorv32 determines any printed character as "?".
Looks like #883 uart_rx upd broke something

**To Reproduce**
Check "sreg" bus of "rx_engine_t" record in "neorv32_uart.vhd" file.
before #883 - sreg[9:0]
after #883 - sreg[8:0]  <- where this issue happened

**Expected behavior**
Correct character determination

**Screenshots**
I push "e" button to abort autoboot sequence
Screen 1 - incorrect symbol.
[image]

Screen 2 - correct after bus fix
[image]

**Environment:**
 - OS Ubuntu 
 - Xilinx xc7k325 fpga
 - ft232rl as uart2usb bridge
 - stock bootloader image

**Hardware:**
 - Hardware version (1.9.8.9)

**Additional context**
I changed sreg back to "9 downto 0" and reimplement design - works fine

Unike267, 2024-05-02:
Hi everyone!

I've reproduced this bug in Arty A7 35t.

As shown in the following result via CuteCom:

[image]

I hope it will be solved soon,

Cheers!

Unike267, 2024-05-02:
Hi,

I've fixed this in local changing this line:

https://github.com/stnolting/neorv32/blob/e2fa5c7579444b1142de635ae0cf2fd82dd30ab5/rtl/core/neorv32_uart.vhd#L289

From `rx_engine.sreg(8 downto 1);` to `rx_engine.sreg(7 downto 0);`

Cheers! 😃

Polaris-III, 2024-05-02:
Hi @Unike267, I'll check your solution tomorrow, thank you

stnolting, 2024-05-02:
Seems like I have messed up the UART receiver 🙈

@Polaris-III thanks for finding this bug!

@Unike267 your fix looks great! Would you open a PR? :wink:

closed 2024-05-02 by stnolting