stnolting/neorv32 #906: Capture variable value in simulation

mahdi259, 2024-05-18:
Hi
This is actually a question. I use simple testbench for simulation. What is the best way to capture the value of a variable of C code without using UART to print it?

**Describe the solution you'd like**
I want to know the result of a operation in simulation mode without needing to print it using UART.

Thanks.

stnolting, 2024-05-18:
Hey @mahdi259!

Hmm that's an interesting question! Are we talking about a variable in main memory?

Let's assume you have something like this:

```c
volatile uint32_t some_variable;
```

This variable **will** be placed in main memory at a fixed address as it is marked as volatile. You can use `make elf_info` to find out the actual memory address of this variable:

[image]

In this example, `some_variable` is placed at 0x800000a4. Then, you could connect some VHDL report block to the CPU's data interface (in `neorv32_top.vhd`) to track write accesses to this address:

```vhdl
process(clk_i)
begin
  if rising_edge(clk_i) then
    if (cpu_d_req.stb = '1') and (cpu_d_req.rw = '1') and (cpu_d_req.addr = x"800000a4") then
      report "MEM 0x" & to_hstring32_f(cpu_d_req.data) & " -> [0x" & to_hstring32_f(cpu_d_req.addr) & "]" severity note;
    end if;
  end if;
end process;
```

Now, when running some C code like this

```c
some_variable = 3;
some_variable = 5;
```

you'll get the according simulator reports in the console:

[image]

mahdi259, 2024-05-18:
Thanks for your response. Unfortunately `make elf_info` doesn't show some_variable. 
[1.txt](https://github.com/stnolting/neorv32/files/15366676/1.txt)
[code.txt](https://github.com/stnolting/neorv32/files/15366679/code.txt)

mahdi259, 2024-05-18:
Ok I should define it as global variable outside main...

stnolting, 2024-05-18:
> Ok I should define it as global variable outside main...

Right, your variable is just **local** so it will be put on the stack. Thus, it might not have a static address.

mahdi259, 2024-05-18:
You really solved my question. I wish great success for you in your career and life.

mahdi259, 2024-05-21:
Hi. Sorry can you mention how to use this code in testbench with xbus signals? It seems that address changes.

mahdi259, 2024-05-22:
Hi. Sorry can you mention how to use this code in testbench with xbus signals? It seems that it doesn't redirect address and data on xbus.

stnolting, 2024-05-22:
> Sorry can you mention how to use this code in testbench with xbus signals? It seems that it changes address on xbus.

Just put that somewhere into `neorv32_top.vhd`.

I'm not really sure what you want to monitor? What does your processor configuration look like? Do you use internal memories (IMEM / DMEM) or do you use some external memory coupled via XBUS?

mahdi259, 2024-05-22:
[image]
[image]
Hi. As mentioned above, I monitor value change of `volatile int temp_i` variable with address specified in `make elf_info` that is 0x80000000.  I use neorv32_tb.simple.vhd that utilizes external memories and I disable instruction and data caches.

I wonder why the same writing isn't seen on xbus signals.  
[codes.zip](https://github.com/stnolting/neorv32/files/15407434/codes.zip)

stnolting, 2024-05-22:
For the default testbench setup, address 0x80000000 is mapped to the **internal** DMEM. Hence, when accessing this variable / address there will be NO activity on the **external** bus interface (XBUS).

mahdi259, 2024-05-22:
Sorry, you're right.

closed 2024-05-22 by mahdi259