stnolting/neorv32 #961: openocd only supports Debug Module version 2 (0.13) and 3 (1.0), not 0 (dmstatus=0x0). This error might be caused by a JTAG signal issue. Try reducing the JTAG clock speed

UmeshChandargi, 2024-07-22:
Hello,
I am trying openocd with jlink JTAG adapter.
My openocd_neorv32.cfg - 
```
# NEORV32 on-chip debugger openOCD configuration file

# ----------------------------------------------
# Physical interface configuration
# -> ADJUST THIS FOR YOUR ADAPTER / SETUP
# ----------------------------------------------
adapter driver jlink
#adapter driver ftdi
#ftdi vid_pid 0x0403 0x6010
#ftdi channel 0
#ftdi layout_init 0x0038 0x003b

adapter speed 4000
transport select jtag

# ----------------------------------------------
# Target configuration
# ----------------------------------------------
set _CHIPNAME neorv32
jtag newtap $_CHIPNAME cpu -irlen 5
set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME.0 riscv -chain-position $_TARGETNAME

# expose NEORV32-specific CSRs
riscv expose_csrs 2048=cfureg0
riscv expose_csrs 2049=cfureg1
riscv expose_csrs 2050=cfureg2
riscv expose_csrs 2051=cfureg3
riscv expose_csrs 4032=mxisa

# enable memory access error reports
gdb_report_data_abort enable

# ----------------------------------------------
# Start session
# ----------------------------------------------
init
halt

echo "Target HALTED. Ready for remote connections."
```
I initally connected to the openocd many times and it worked fine.
But then i disabled the internal bootloader and ran the program again, Igot this error "openocd only supports Debug Module version 2 (0.13) and 3 (1.0), not 0 (dmstatus=0x0). This error might be caused by a JTAG signal issue. Try reducing the JTAG clock speed."

And then onwards I am unable to connect to openocd.

Below the image where it was working initally and then stopped connecting.
[image]

I rechecked the pin connections and the pins are pulled up in the FPGA. My board is Xilinx Zedboard
```
## External JTAG Interface
set_property -dict {PACKAGE_PIN AA9 IOSTANDARD LVCMOS33 PULLTYPE PULLUP } [get_ports jtag_trst_i]
set_property -dict {PACKAGE_PIN Y11 IOSTANDARD LVCMOS33 PULLTYPE PULLUP } [get_ports jtag_tck_i]
set_property -dict {PACKAGE_PIN Y10 IOSTANDARD LVCMOS33 PULLTYPE PULLUP } [get_ports jtag_tdi_i]
set_property -dict {PACKAGE_PIN AA11 IOSTANDARD LVCMOS33 PULLTYPE PULLUP } [get_ports jtag_tms_i]
set_property -dict {PACKAGE_PIN AB11 IOSTANDARD LVCMOS33 PULLTYPE PULLUP } [get_ports jtag_tdo_o]
```
Could you please help me with this?

stnolting, 2024-07-22:
Hey @UmeshChandargi.

Which [version](https://github.com/stnolting/neorv32/blob/main/CHANGELOG.md) of the core are you using (I see that in your setup the `jtag_trst_i` signal is still present)?

> I initally connected to the openocd many times and it worked fine.
But then i disabled the internal bootloader and ran the program again, Igot this error "openocd only supports Debug Module version 2 (0.13) and 3 (1.0), not 0 (dmstatus=0x0). This error might be caused by a JTAG signal issue. Try reducing the JTAG clock speed."

So this error only occurs if the bootloader is disabled? Does the openOCD connection work again when you re-enable the bootloader?

UmeshChandargi, 2024-07-23:
> Which [version](https://github.com/stnolting/neorv32/blob/main/CHANGELOG.md) of the core are you using (I see that in your setup the `jtag_trst_i` signal is still present)?

core version is 01.09.09.01

> So this error only occurs if the bootloader is disabled? Does the openOCD connection work again when you re-enable the bootloader?
> 
> What happens if you re-upload the FPGA bitstream? Can you establish a connection?

Actually I had established a connection when I had disabled the bootloader for my application. I apologise for the miscommunication earlier. I have updated the comments.

Now I am facing issues even when i enable the bootloader or disable it. 
After I upload the new bitstream and reset  the FPGA I am always getting this error:
 
```
ml013:~> openocd -f /home.filer/chandargi/GIT_repo/neorv32-setups/neorv32/sw/openocd/openocd_neorv32.cfg
Open On-Chip Debugger 0.12.0+dev-00634-g23c33e1d3 (2024-07-12-12:09)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : J-Link ARM V8 compiled Sep 22 2014 23:26:43
Info : Hardware version: 8.00
Info : VTarget = 3.338 V
Info : clock speed 100 kHz
Error: JTAG scan chain interrogation failed: all zeroes
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: neorv32.cpu: IR capture error; saw 0x00 not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: dtmcontrol is 0. Check JTAG connectivity/board power.
Error: [neorv32.cpu.0] Examination failed
Warn : target neorv32.cpu.0 examination failed
Info : [neorv32.cpu.0] starting gdb server on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet
```

I reduced the clock to 100kHz. 
I removed the `jtag_trst_i` connection in my block design and connected a constant of 1 as shown below

[image]

But I am facing the issue despite of all these changes.

stnolting, 2024-07-23:
> core version is 01.09.09.01

If possible, I would suggest updating to the latest version. :wink:

> Actually I had established a connection when I had disabled the bootloader for my application. I apologise for the miscommunication earlier. I have updated the comments.

So you were able to establish a successful connection via openOCD? What (hardware configuration) have you changed since then?

> Now I am facing issues even when i enable the bootloader or disable it.
After I upload the new bitstream and reset the FPGA I am always getting this error:

The wiring in your block design looks good. What hardware configuration do you use? Could you make a screenshot from the configuration options from the customization GUI?

I see that you have connected the UART lines. Do you have the bootloader enabled? Can you see the bootloader console after reset?

UmeshChandargi, 2024-07-24:
Hello,
I solved it. I checked with the oscilloscope for tck, tdi, tdo on the PMOD. I got no readings so I checked the IO report. None of the JTAG signals appeared in the report. After analysing a little I saw that vivado had duplicated the wrapper file from the block design (I do no know why it suddenly happened now). All the JTAG signals were updated in the duplicated wrapper.vhd. I cleaned it and now the openocd works fine.
It's a bit embarrassing, I apologise for taking your time. 
Thank you for all your help.

closed 2024-07-24 by UmeshChandargi

stnolting, 2024-07-24:
> It's a bit embarrassing, I apologise for taking your time.

No worries! I know these problems well enough myself... Vivado can be tricky sometimes. 😅

Anyway, it's great to hear that your setup is working now!