stnolting/neorv32 #968: Datasheet error in the DMA section

Sam-Vervaeck, 2024-07-29:
**Describe the bug**
The descriptions of the DMA_TTYPE_QSEL does not match over two locations:
[image]
[image]

**Expected behavior**
I assume the first screenshot is the correct description and should be carried over.

**PS:**
I am putting this here now as a bug, do you prefer if I create a discussion for it instead like in the past?

Sam-Vervaeck, 2024-07-29:
To add another error since opening another ticket seems unnecessary, this figure seems to be misplaced right now:
[image]

Sam-Vervaeck, 2024-07-29:
I want to stress though that the general quality of documentation is really good. Working with this project has been a breeze compared to trying to work with Ibex previously!

stnolting, 2024-07-29:
Thanks for finding these!

> I want to stress though that the general quality of documentation is really good. Working with this project has been a breeze compared to trying to work with Ibex previously!

Thank you very much! :)

commit f096fddb (2024-07-29): [docs] minor fixes

#968
closed 2024-07-29 by stnolting