stnolting/neorv32 #973: [HELP WANTED] Reorganize HDL files? 🤔

stnolting, 2024-08-03:
I am struggling with the current organization of the core's HDL files. There are several different setups that use different file sets. Right now, only the memories allow an easy replaced (by setup- or platform-specific files). But there are other files that might be replaced by a user depending on the actual setup:

* instruction memory (IMEM)
* data memory (DMEM)
* custom functions subsystem (CFS)
* custom functions unit (CFU)
* clock-gating switch (CLOCKGATE)

For example, in one setup we just want to replace the default memories by some ASIC-macros. In another setup, we might want to replace the default CFU to add custom and setup-specific instructions. And all this should be based on wildcards or file-lists so we do not have to add each file individually. How can we do this in a nice and easy to understand way? :thinking:

Furthermore, the splittings splitting of entity and architecture (#151) seems to confuse some user (recently, I got a lot of messages regarding this). Should we keep this?

I have been thinking about a reorganization of the HDL files. Here are two ideas:

### Option 1

* use individual folders
  * e.g. `rtl/core/mem` for the default memory modules
  * e.g. `rtl/core/cfs` for the default custom functions subsystem
  * and so on...
* alternatively, use a single folder for all setup/platform-specific files (`rtl/core/platform`)

Thus, we would have one folder for all the core / setup-independent HDL file (`rtl/core`) and individual folders for each exchangeable module.

### Option 2

* put **all** HDL files into `rtl/core` - no more subfolders
* rename all setup-independent files -> `*.core.vhd`
* rename all setup-specific files -> `*.default.vhd` (these are the default files that can be replaced)

The nice thing here is that we can still use wildcards to add all the setup-independent files. The setup-specific files are outnumbered, so it seems fine to add them one by one.

---------------------------

Feedback (or even better ideas) are highly welcome! :)

NikLeberg, 2024-08-05:
For the list of files that a user might want to overwrite: add also the debug module (DM).

About the entity / architecture splitting: Thats just VHDL, users have to learn it eventually. What better way to encounter this problem than with this super well documented project? 😃

About Option 1:
 - with the individual folders: would that not leave us with folders with one single file in them? e.g. `glockgate` only contains `neorv32_clockgate.vhd`, or would you add the entity there as well?
 - Having a folder just for a single file is a bit moot.
 - Here I like the `platform` folder idea better.

About Option 2:
 - are you implying to rename e.g. the `neorv32_cpu_alu.vhd` to `neorv32_cpu_alu.core.vhd`?
 - Doable, but I dont realy like it...

About Option 3:
 - this would liberate the users to choose for themself.
 - the other options kinda assume what the user might want to overwrite, if he then wants to overwrite yet another module he has no significant advantage and has to modify his fileset anyway.

IMHO, option 1 with a common `platform` folder sounds the best. But option 3 is also very much VHDL like.

Hot take: why not remove the specific entity/archs entirely? This would help users that just started with VHDL with the whole _fileset problem_ and _choosing mem implementation_. Advanced users that might want to replace an architecture can be expected to know how to do it at instatiation with things like:
```vhd
neorv32_debug_dtm_inst : entity neorv32.neorv32_debug_dtm(custom_architecture)
        generic map (
           ...
```

stnolting, 2024-08-05:
Thanks for your feedback, @NikLeberg!

> For the list of files that a user might want to overwrite: add also the debug module (DM).

The DTM and DM are part of the SoC list:

https://github.com/stnolting/neorv32/blob/e72c254bf60a2bddf41d7d44e9853a7d565951b0/rtl/file_list_soc.f#L42-L43

We cannot add them to the CPU list as the DTM relies on a specific memory mapping, which is not available in a CPU-only setup.

Maybe we should drop the CPU HDL list anyway and stick to the "NEORV32 top as CPU wrapper" idea? 🤔 

> About the entity / architecture splitting: Thats just VHDL, users have to learn it eventually. What better way to encounter this problem than with this super well documented project? 😃

I think I do not like this splitting concept anymore... I need to open two files to understand a unit's interface.

Oh, and thanks! ❤️

> **Option 1**
> with the individual folders: would that not leave us with folders with one single file in them? e.g. glockgate only contains neorv32_clockgate.vhd, [...] Having a folder just for a single file is a bit moot.

You are right, that would be too much. However, I also like the "platform folder" idea.

> **Option 2**
> are you implying to rename e.g. the neorv32_cpu_alu.vhd to neorv32_cpu_alu.core.vhd?

Correct. Basically, this would have the same effect as the "platform folder" approach but with all files in the same directory.

> **Option 3**
> this would liberate the users to choose for themself. the other options kinda assume what the user might want to overwrite, if he then wants to overwrite yet another module he has no significant advantage and has to modify his fileset anyway.

I also like this concepts.

The big problem here is that we need to have a list of all HDL files for **each new setup**. If a new module is added, this must be adapted in all projects... But we wouldn't have this problem if we had wildcards, separate folders or special file suffixes.

> Nevermind.. forgot that this would then necessitate changing the neorv32 files themself.

:+1:

stnolting, 2024-08-09:
> But option 3 [_all files in one folder_] is also very much VHDL like.

I think this is the best approach - it is clean, straightforward, easy to understand.

> the other options kinda assume what the user might want to overwrite, if he then wants to overwrite yet another module he has no significant advantage and has to modify his fileset anyway.

Good point! Who am I to dictate which files can be replaced/customized? 😅

I am preparing a PR for this. It will probably be a lot of work to adapt all repositories, so this might take a while.

closed 2024-08-09 by stnolting