stnolting/neorv32 #997: [idea] Add a generic RAM primitive for **all** memories?

stnolting, 2024-08-26:
Right now I am trying to port the core to a specific ASIC technology. Unfortunately, inferring RAM is ASIC world is not as easy as in FPGA world, so I need to replace all (larger) memory arrays by specific IP blocks.

I wonder if it would make sense to have a generic RAM module (e.g. `neorv32_ram.vhd`) which is then instantiated in all places where memory arrays are used (IMEM, DMEM, caches). Then you would only have to adapt a single file to map all those large memories to specific primitives.

What do you think? 🤔
Maybe I'm making this too easy for myself or am I missing something?

NikLeberg, 2024-08-27:
> Right now I am trying to port the core to a specific ASIC technology.

So cool! What technology node are you using?

So your idea is replacing things like

```vhd
signal mem_ram_b0, ... : mem8_t(0 to DMEM_SIZE/4-1);
```

With an ram entity that takes an input addess and then provides the data at this address? Would that entity need to be registered to work in ASIC toolchains?

If it makes it tremedeously more easy for ASIC setups and only slighly (if at all) less easy for FPGA setups, then I see no reason against.
I think that we'd need to double check that all common FPGA tools accept the new entity and are able to map it to block ram.

What about FIFOs? Are they manageable as they already are contained within a single entity?

stnolting, 2024-08-28:
> So cool! What technology node are you using?

It is!! 😅
A 180nm technology and maybe a 130nm technology as well.

> So your idea is replacing things like [...]

That's right.

> With an ram entity that takes an input addess and then provides the data at this address?

Correct. Maybe something like this:

```vhdl
entity neorv32_ram is
  generic (
    RAM_SIZE : natural -- RAM size in bytes, has to be a power of 2
  );
  port (
    clk_i  : in  std_ulogic; -- triggering on rising edge
    we_i   : in  std_ulogic; -- write-enable
    ben_i  : in  std_ulogic_vector(3 downto 0); -- byte-enable
    addr_i : in  std_ulogic_vector(index_size_f(RAM_SIZE/4)-1 downto 0); -- word access address
    data_i : in  std_ulogic_vector(31 downto 0); -- write data
    data_o : out std_ulogic_vector(31 downto 0)  -- read data
  );
end neorv32_ram;
```

> Would that entity need to be registered to work in ASIC toolchains?

That is indeed a good question! There are RAM compilers (like [OpenRAM](https://github.com/VLSIDA/OpenRAM)) that can generate you (arbitrary) RAM layouts. But there are also pre-defined IPs (like a 256x8-bit macro) that you have to interconnect yourself to form larger blocks.

Register stages and access latencies are indeed a bit tricky. The RAMs I have seen do not have any registers at all - these must be added explicitly if you need them. But there are also pre-defined RAM macros that already have register - maybe just one output register or maybe even a second register to buffer the address.

So we would need a generic RAM module that provides some mechanism for "exporting" the latency to the actual instance (e.g. the DMEM).

Furthermore, ASIC RAMs often also provide advanced power-saving features like disabling the clock, powering down certain amplifiers, etc. To make actual use of this features the generic RAM module would also need to provide some kind of (generic) interface for this...

So, on a second thought, the whole idea here might not work out... 😅

> I think that we'd need to double check that all common FPGA tools accept the new entity and are able to map it to block ram.

Basically, we already have some (very minimal) mechanism for that:

https://github.com/stnolting/neorv32/blob/47948ac8c2f61922aafe3ae256411dbe02675659/rtl/core/neorv32_dmem.vhd#L32-L33

> What about FIFOs? Are they manageable as they already are contained within a single entity?

Another good question!

FIFOs might require a very specific timing (e.g. an async read so new data just "falls through"). For rather small FIFOs I would just use standard cells (i.e. no RAM primitives) at all. But the term "rather small" is of course very application-specific here...

stnolting, 2024-08-28:
FYI

Here is a nice pre-defined IP block from Efabless: https://platform.efabless.com/design_catalog/ip_block/40
Actually, this one already has one output register stage.

closed 2024-09-29 by stnolting