stnolting/neorv32 #1011: CLZ Timing Documentation Clarification

CharlesAverill, 2024-09-11:
**Is your feature request related to a problem? Please describe.**
In Section 3.8, page 148 of the NEORV32 manual, the worst-case clock cycle count provided for the `CLZ` instruction is `3 + 0..32`. I'd like to clarify that this means `3 + <number of leading zeroes in rs>`.

stnolting, 2024-09-13:
> I'd like to clarify that this means 3 + <number of leading zeroes in rs>.

That's right. Maybe the documentation is a little bit unclear here...

If the `FAST_SHIFT_EN` tuning option is disabled, `clz` is implemented as a simple shift register that is loaded with the operand. The register is shifted one position per clock cycle and a simple counter keeps incrementing until the MSB is set. Hence, the total execution time is related to the number of leading zeros in the operand data.

If the `FAST_SHIFT_EN` tuning option is enabled, the number of leading zeros is computed within a single clock cycle regardless of the number of leading zeros.

CharlesAverill, 2024-09-13:
Thank you for the clarification, and neat insight into the instruction implementation!

closed 2024-09-13 by CharlesAverill