stnolting/neorv32 #1016: Access Denied Error When Building on Eclipse

AbdulrahmanKH7, 2024-09-14:
Hi,

Thanks a lot for this very special open-source MCU. My issue is I am getting the following error when trying to build on both a preconfigured project and a project from scratch (in Eclipse):
04:46:28 **** Incremental Build of configuration Debug for project test2-risc-v ****
make all 
Building file: ../src/main.c
Invoking: GNU RISC-V Cross C Compiler
riscv-none-elf-gcc -msmall-data-limit=8 -mno-save-restore -O0 -fmessage-length=0 -fsigned-char -ffunction-sections -fdata-sections -g3 -std=gnu11 -MMD -MP -MF"src/main.d" -MT"src/main.o" -c -o "src/main.o" "../src/main.c"
c:/users/abdul/downloads/xpack-windows-build-tools-4.4.1-3-win32-x64/xpack-windows-build-tools-4.4.1-3/bin/sh: riscv-none-elf-gcc: Permission denied
make: *** [src/subdir.mk:20: src/main.o] Error 126
"make all" terminated with exit code 2. Build might be incomplete.

04:46:28 Build Failed. 1 errors, 0 warnings. (took 571ms)

I noticed that the first time I built, the antivirus (Avast) stopped the build process to scan a file (I believe it is the sh file) and said it's safe. Then I disabled the antivirus and still the problem persists.

Your support is highly appreciated!
Thank you

stnolting, 2024-09-14:
Hey @AbdulrahmanKH7.

> Thanks a lot for this very special open-source MCU. 

Thank you for the nice words! :)

> `elf-gcc: Permission denied`

Hm, I never say this... I assume you are using Windows? Can you try to run the "make_clean" target from the project's list of targets before you start a new build?

AbdulrahmanKH7, 2024-09-14:
Yes I am using Windows. I ran the "make_clean" target and it cleaned successfully but when I build I get the same error. I checked the permissions of this file and it got all of them and I also ran Eclipse as admin and still. I tried also different solutions from internet like checking if this denied executable is still running (by task manager) and the antivirus quarantine folder.

stnolting, 2024-09-14:
That's strange... Can you run the "info" target or does that also fail?

Where did you get the GCC binaries from? Maybe X-Pack?

Please open a system console (cmd.exe), navigate to the `bin` folder of the GCC toolchain and try to execute one of the binaries from the console.

AbdulrahmanKH7, 2024-09-14:
I ran the info and got the following (it failed as well):

10:20:14 **** Build of configuration Default for project eclipse_example ****
make info 
BusyBox v1.37.0.git-5398-g89ae34445 (2024-09-13 08:43:13 UTC)

Usage: basename FILE [SUFFIX] | -a FILE... | -s SUFFIX FILE...

Strip directory path and SUFFIX from FILE

	-a		All arguments are FILEs
	-s SUFFIX	Remove SUFFIX (implies -a)
******************************************************
Project / Makfile Configuration
******************************************************
Project folder: 
Source files: ./main.c   
Include folder(s): -I .
ASM include folder(s): -I .
NEORV32 home folder (NEORV32_HOME): ../../..
IMAGE_GEN: ../../../sw/image_gen/image_gen
Core source files:
../../../sw/lib/source/neorv32_aux.c ../../../sw/lib/source/neorv32_cfs.c ../../../sw/lib/source/neorv32_cpu.c ../../../sw/lib/source/neorv32_cpu_amo.c ../../../sw/lib/source/neorv32_cpu_cfu.c ../../../sw/lib/source/neorv32_crc.c ../../../sw/lib/source/neorv32_dma.c ../../../sw/lib/source/neorv32_gpio.c ../../../sw/lib/source/neorv32_gptmr.c ../../../sw/lib/source/neorv32_mtime.c ../../../sw/lib/source/neorv32_neoled.c ../../../sw/lib/source/neorv32_newlib.c ../../../sw/lib/source/neorv32_onewire.c ../../../sw/lib/source/neorv32_pwm.c ../../../sw/lib/source/neorv32_rte.c ../../../sw/lib/source/neorv32_sdi.c ../../../sw/lib/source/neorv32_slink.c ../../../sw/lib/source/neorv32_spi.c ../../../sw/lib/source/neorv32_sysinfo.c ../../../sw/lib/source/neorv32_trng.c ../../../sw/lib/source/neorv32_twi.c ../../../sw/lib/source/neorv32_uart.c ../../../sw/lib/source/neorv32_wdt.c ../../../sw/lib/source/neorv32_xip.c ../../../sw/lib/source/neorv32_xirq.c ../../../sw/common/crt0.S
Core include folder:
../../../sw/lib/include
Project object files:
./main.c.o ../../../sw/lib/source/neorv32_aux.c.o ../../../sw/lib/source/neorv32_cfs.c.o ../../../sw/lib/source/neorv32_cpu.c.o ../../../sw/lib/source/neorv32_cpu_amo.c.o ../../../sw/lib/source/neorv32_cpu_cfu.c.o ../../../sw/lib/source/neorv32_crc.c.o ../../../sw/lib/source/neorv32_dma.c.o ../../../sw/lib/source/neorv32_gpio.c.o ../../../sw/lib/source/neorv32_gptmr.c.o ../../../sw/lib/source/neorv32_mtime.c.o ../../../sw/lib/source/neorv32_neoled.c.o ../../../sw/lib/source/neorv32_newlib.c.o ../../../sw/lib/source/neorv32_onewire.c.o ../../../sw/lib/source/neorv32_pwm.c.o ../../../sw/lib/source/neorv32_rte.c.o ../../../sw/lib/source/neorv32_sdi.c.o ../../../sw/lib/source/neorv32_slink.c.o ../../../sw/lib/source/neorv32_spi.c.o ../../../sw/lib/source/neorv32_sysinfo.c.o ../../../sw/lib/source/neorv32_trng.c.o ../../../sw/lib/source/neorv32_twi.c.o ../../../sw/lib/source/neorv32_uart.c.o ../../../sw/lib/source/neorv32_wdt.c.o ../../../sw/lib/source/neorv32_xip.c.o ../../../sw/lib/source/neorv32_xirq.c.o ../../../sw/common/crt0.S.o
LIBGCC:
process_begin: CreateProcess(C:\xpack\xpack-riscv-none-elf-gcc-14.2.0-2\bin\riscv-none-elf-gcc, riscv-none-elf-gcc -print-libgcc-file-name, ...) failed.
make (e=193): Error 193
make: *** [../../../sw/common/common.mk:348: info] Error 193
"make info" terminated with exit code 2. Build might be incomplete.

10:20:18 Build Failed. 1 errors, 0 warnings. (took 3s.670ms)

------------------------------------------------------------------------------------------------------

I got the binaries from xpack.
I tried to execute one of the files in bin folder from cmd and got this:

Microsoft Windows [Version 10.0.22631.4169]
(c) Microsoft Corporation. All rights reserved.

C:\Users\abdul>cd C:\xpack\xpack-riscv-none-elf-gcc-14.2.0-2\bin

C:\xpack\xpack-riscv-none-elf-gcc-14.2.0-2\bin>riscv-none-elf-gdb
'riscv-none-elf-gdb' is not recognized as an internal or external command,
operable program or batch file.

C:\xpack\xpack-riscv-none-elf-gcc-14.2.0-2\bin>C:\xpack\xpack-riscv-none-elf-gcc-14.2.0-2\bin start riscv-none-elf-gdb
'C:\xpack\xpack-riscv-none-elf-gcc-14.2.0-2\bin' is not recognized as an internal or external command,
operable program or batch file.

stnolting, 2024-09-14:
Maybe you have installed the Linux version instead of the Windows one? For the Windows version all binaries should have an `.exe` suffix:

```
xpack-riscv-none-elf-gcc-14.2.0-2/bin$ ls
DLLs                          riscv-none-elf-ar.exe          riscv-none-elf-gcc.exe            riscv-none-elf-lto-dump.exe
libgcc_s_seh-1.dll            riscv-none-elf-as.exe          riscv-none-elf-gcov-dump.exe      riscv-none-elf-nm.exe
libgmp-10.dll                 riscv-none-elf-c++.exe         riscv-none-elf-gcov-tool.exe      riscv-none-elf-objcopy.exe
libmpc-3.dll                  riscv-none-elf-c++filt.exe     riscv-none-elf-gcov.exe           riscv-none-elf-objdump.exe
libmpfr-6.dll                 riscv-none-elf-cpp.exe         riscv-none-elf-gdb-add-index      riscv-none-elf-ranlib.exe
libstdc++-6.dll               riscv-none-elf-elfedit.exe     riscv-none-elf-gdb-add-index-py3  riscv-none-elf-readelf.exe
libwinpthread-1.dll           riscv-none-elf-g++.exe         riscv-none-elf-gdb-py3.exe        riscv-none-elf-size.exe
libzstd.dll                   riscv-none-elf-gcc-14.2.0.exe  riscv-none-elf-gdb.exe            riscv-none-elf-strings.exe
python312.dll                 riscv-none-elf-gcc-ar.exe      riscv-none-elf-gprof.exe          riscv-none-elf-strip.exe
python312.zip                 riscv-none-elf-gcc-nm.exe      riscv-none-elf-ld.bfd.exe
riscv-none-elf-addr2line.exe  riscv-none-elf-gcc-ranlib.exe  riscv-none-elf-ld.exe
```

AbdulrahmanKH7, 2024-09-14:
My bad xD.. sorry about that, it is working now!
Thanks man you are the best!

stnolting, 2024-09-14:
> it is working now!

Great to hear! 👍

closed 2024-09-14 by stnolting