stnolting/neorv32 #1024: Check code quality in a continuous way

racodond, 2024-09-19:
Hi,

I analyzed the VHDL code from your NEORV32 project with Linty: https://oss.linty-services.com/dashboard?id=neorv32&codeScope=overall

Do not get scared by the number of issues :-)
It's just to provide you with data to browse. I activated all the rules, some are mutually opposed, others are not properly configured with your requirements (such as naming conventions).

There might be a potential bug behind that one though: https://oss.linty-services.com/project/issues?issues=0bea7495-7740-4097-91c3-baac7057b3f4&open=0bea7495-7740-4097-91c3-baac7057b3f4&id=neorv32

We offer Linty for free for open-source projects. I'd be glad to organize a call if you're interested in checking code quality in a continuous way through GitHub actions.
We also offer our VS Code extension for free: https://hdl-designer.linty-services.com/

I'm looking forward to hearing from you.

Thank you

Have a good day

David Racodon
CTO at Linty Services

racodond, 2024-09-20:
> Wow, this looks really amazing! Linting has been on my to-do list for quite some time... 😅

You'll see that Linty does a lot more than linting -:)
CDC, RDC, etc.
See https://oss.linty-services.com/coding_rules?languages=hdl and https://oss.linty-services.com/coding_rules?languages=vhdl

> That sounds really great and I appreciate your offer. Can you provide a link to the according GH action? Or maybe a link to some documentation?

You're welcome!

Here's the documentation: https://doc.linty-services.com
To scan your code: https://doc.linty-services.com/scan.html (see GitHub section)

If you provide me with a name and an email address, I'll create an account for you on our OSS cloud platform (https://oss.linty-services.com/) so that you'll be able to update your quality profile (set of rules) and run Linty analyses.

stnolting, 2024-09-20:
> You'll see that Linty does a lot more than linting -:)
CDC, RDC, etc.

That sounds really great! :+1:

> Here's the documentation: https://doc.linty-services.com/
To scan your code: https://doc.linty-services.com/scan.html (see GitHub section)

Thanks - I'll have a look at those.

> If you provide me with a name and an email address, I'll create an account for you on our OSS cloud platform (https://oss.linty-services.com/) so that you'll be able to update your quality profile (set of rules) and run Linty analyses.

Can I just create an OSS account over there?

I am really curious about your OSS plan in general. Please don't get me wrong, but I had some bad experiences with a certain CI platform in the past. They had a strong commitment to open source, but then they started limiting OSS resources more and more, and in the end you had to send emails when resources were exhausted and politely ask for more additional server time, which is an absolute no-go.

Again, please don't get me wrong. I am very grateful for the offer - I'm just curious. :wink:

racodond, 2024-09-20:
> Can I just create an OSS account over there?

You cannot create an account by yourself (at least for now).
Here's the link to send name and email privately to me: https://linty-services.com/contact.html

> I am really curious about your OSS plan in general. Please don't get me wrong, but I had some bad experiences with a certain CI platform in the past. They had a strong commitment to open source, but [...]
Again, please don't get me wrong. I am very grateful for the offer - I'm just curious. 😉

No problem :-)
Any question is worth asking.

Our objective is to keep the OSS platform really free with no limitation. When you run a Linty analysis, most of the computing is done on your GitHub runners. So costs on our side should not be exponential as we add more projects (and thus analyses) to the Linty platform.

At Linty Services, we're more tech guys that sales guys. We believe that Linty could bring a lot of value and we'd really like Linty to be widely used. That's why we decided to make Linty free for open-source projects.

We can schedule a call later on to get to know each other better.

stnolting, 2024-09-20:
> You cannot create an account by yourself (at least for now).
Here's the link to send name and email privately to me: https://linty-services.com/contact.html

👍

> Our objective is to keep the OSS platform really free with no limitation. When you run a Linty analysis, most of the computing is done on your GitHub runners. So costs on our side should not be [...]

Now, I am really pleased to read this! :)
Thanks for the feedback.

stnolting, 2025-09-06:
I apologize for not addressing this issue for so long... I somehow lost sight of it.

I think your tool is a cool thing - but unfortunately, it's too time-consuming for this project and for me as the only administrator who has access to the tool.

I understand that the operators of such tools also have bills to pay, and so I would like to say thank you again for providing this free access.

closed 2025-09-06 by stnolting

racodond, 2025-09-10:
Thank you for trying Linty.
Feel free to reach out if you have more time to use it in the future.