stnolting/neorv32 #1043: UART0 not working after post implementation timing simulation under Vivado 2021.2

H1alus, 2024-09-29:
I am new to the neorv32 platform, so I'm still trying to get my feet wet, I am working on extending a risc-v processor for my thesis work.

I created a minimal processor configuration starting from the neorv32_ProcessorTop_Minimal.vhd, then I created a testbench for the said top module while also simulating the uart_rx_simple.vhd . The hello_world example works perfectly in behavioural  simulation.

When i run the post implementation timing simulation I can´t get any ouput, the uart_txd signal is stuck on high.

I disabled incremental compilation from the simulation settings as explained in the user guide.
The hello_world program was both compiled with USER_FLAGS+=-DUART0_SIM_MODE and without.

I need the post implementation simulation because I don´t have an fpga board unless I'm in the university's lab and my thesis work requires to have some power estimations.
Here's the files I made.

[files.zip](https://github.com/user-attachments/files/17180083/files.zip)

H1alus, 2024-09-29:
Strange thing, the USER_FLAGS+=-DUART0_SIM_MODE was still set after i did a make clean and make clean_all, i solved by USER_FLAGS-=-DUART0_SIM_MODE, closing and reopening the terminal should work too.

closed 2024-09-29 by H1alus

stnolting, 2024-09-29:
Hey @H1alus.

> I am new to the neorv32 platform

Welcome! :)

> When i run the post implementation timing simulation I can´t get any ouput, the uart_txd signal is stuck on high.

Your HDL files look good so far. Just two tips:

* Don't mix `std_logic[_vector]` and `std_ulogic[_vector]` without having explicitly enabled VHDL2008 support (which seems to be the default in newer Vivado versions?!).
* For all enabled modules, connect tie unused inputs to a default value (`uart0_cts_i` in your case).

I threw your top and your testbench into my Vivado setup. The post-implementation simulation takes quite a long time, but I can see activity on the UART TXD signal:

[image]

> The hello_world program was both compiled with USER_FLAGS+=-DUART0_SIM_MODE and without.

Don't use the `UART0_SIM_MODE` if you want to have "physical" UARt output (see [📚 Primary Universal Asynchronous Receiver and Transmitter (UART0)](https://stnolting.github.io/neorv32/#_primary_universal_asynchronous_receiver_and_transmitter_uart0) section "Simulation Mode").

stnolting, 2024-09-29:
> Strange thing, the USER_FLAGS+=-DUART0_SIM_MODE was still set after i did a make clean and make clean_all, i solved by USER_FLAGS-=-DUART0_SIM_MODE, closing and reopening the terminal should work too.

Oh, timing! 😅

Good to hear that the setup is working now.

H1alus, 2024-09-29:
Thank you so much for the fast reply and the tips you gave me!

stnolting, 2024-09-29:
You're welcome!