stnolting/neorv32 #1050: Invalid carachter when generating file list with GHDL using proposed script

H1alus, 2024-10-07:
**Describe the bug**
On latest commit there's a character not UTF-8 encoded in neorv32_cpu_cp_bitmanip.vhd line 139 coloumn 172

**To Reproduce**
Pull latest commit and after replacing the placeholder inside rtl/generate_file_lists.sh running the script the following message is displayed:
```
core/neorv32_cpu_cp_bitmanip.vhd:139:172:error: invalid character, even in a comment
/usr/bin/ghdl-mcode:error: importation has failed due to compilation error
```

**Expected behavior**
The script shouldn't fail and complete the file list generation

**Environment:**
 - OS UBUNTU 22.04
 - GHDL 0.37

**Hardware:**
 - neorv32 e68842475217033b757b7949cf0a207ac31c88f8 commit

stnolting, 2024-10-07:
Oh, how I hate it when that happens... 🙈

I think this is the problem (invalid escape sequence):

[image]

I'm just wondering why I don't have a problem with those scripts... Maybe it's a Windows/WSL thing?! 🤔 

Anyway, I'll fix this. Thanks for finding!

H1alus, 2024-10-07:
I guess so since windows/wsl uses ISO encoding for what I know. Are you using vim?

closed 2024-10-07 by H1alus

stnolting, 2024-10-07:
> Are you using vim?

Absolutely not! I always have to google how to get out of it when I accidentally open vim... 😄

I'm a big notepad++ fan.

H1alus, 2024-10-07:
Ah ok, you have to love the vim style to use it

stnolting, 2024-10-07:
I guess so... But I don't like emacs either. nano is fine, but nothing beats notepad++ (IMHO)! 😅