stnolting/neorv32 #1061: A question on fence and fence.i

mahdi259, 2024-10-17:
**Describe the bug**
Hi. As you described in datasheet, fence and fence.i instructions act in the same manner.

> The NEORV32 treats both fence instructions (fence = data fence, fence.i = instruction fence) in exactly the same way. Both instructions cause a flush of the CPU’s instruction prefetch buffer and also send a fence request via the system bus (see [Bus Interface](https://stnolting.github.io/neorv32/#_bus_interface)). This system bus fence operation will, for example, clear/flush all downstream caches.

My understanding is that it doesn't matter if you execute fence or fence.i instruction. Executing each of them clears all caches (dcache, icache, and xcache) and instruction prefetch buffer. But isn't fence.i specifically for instruction cache? and fence for data cache?

Thanks

stnolting, 2024-10-17:
> My understanding is that it doesn't matter if you execute fence or fence.i instruction. Executing each of them clears all caches (dcache, icache, and xcache) and instruction prefetch buffer.

That's right - at least for NEORV32.

However, to keep you code platform-independent (and to make clear you intention of each FENCE operation) I recommend to keep using individual `fence` and `fence.i` instructions.

> But isn't fence.i specifically for instruction cache? and fence for data cache?

It's difficult to generalize, because it depends a lot on the platform.

On a single-hart setup (just one CPU core; no other instances that can alter memory) `fence` can be implemented as simple `nop`. On another single-hart platform a `fence` might put the CPU into halt mode until outstanding DMA transfers have completed...

mahdi259, 2024-10-18:
Thanks a lot.

stnolting, 2024-10-18:
If a platform does not have any caches nor any other mechanism that might cause outstanding or re-ordering of memory accesses any FENCE instruction can also be implemented as `nop` because there is nothing a fence could do.

stnolting, 2024-10-18:
Furthermore, the point that fence instructions cause cache flushes is also platform-specific and not explicitly dictated by the RISC-V spec. To be precisely, there is an additional ISA extension that contain all the cache management operations.

mahdi259, 2024-10-18:
Thanks a lot

closed 2024-10-18 by mahdi259