stnolting/neorv32 #1070: bootloader makefile

mahdi259, 2024-10-19:
Hi
I appreciate utilization of parameters in bootloader image generator. Unfortunately some parameters are not recognized and I'm not able to change them in make command. Any idea?

`make USER_FLAGS+=-DUART_BAUD=115200 USER_FLAGS+=-SPI_FLASH_SECTOR_SIZE=4096 clean_all bootloader`

And the result is:

```
/content/neorv32/sw/bootloader
riscv32-unknown-elf-gcc: error: unrecognized command-line option '-SPI_FLASH_SECTOR_SIZE=4096'
make: *** [../../sw/common/common.mk:172: bootloader.c.o] Error 1
```

stnolting, 2024-10-19:
You need to put a `-D` in front of each define that you want to alter. Just add this to your SPI flash definition and it should work:

```
make USER_FLAGS+=-DUART_BAUD=115200 USER_FLAGS+=-DSPI_FLASH_SECTOR_SIZE=4096 clean_all bootloader
```

mahdi259, 2024-10-19:
Thanks a lot

mahdi259, 2024-10-19:
May I ask what `-D` does?

closed 2024-10-19 by mahdi259

stnolting, 2024-10-19:
It is just a GCC preprocessor flag: https://gcc.gnu.org/onlinedocs/gcc/Preprocessor-Options.html

mahdi259, 2024-10-19:
Thanks a lot