stnolting/neorv32 #1078: Application image zero padding

mahdi259, 2024-10-24:
**Is your feature request related to a problem? Please describe.**
Hi 
I'm trying to use external SRAM memory instead of internal IMEM. The problem is that it's blocked at 
`Booting from 0x00000000`
I found out that SRAM memory is not cleared after power off and addresses beyond program binary contain non zero data. Is it necessary that they get zero?

mahdi259, 2024-10-24:
I cleared SRAM manually and then uploaded binary file. Although other memory values are zeroed, it is again blocked at `Booting from 0x00000000`.

mahdi259, 2024-10-24:
After enabling `XBUS_CACHE_EN`, it works now! Does anyone have any idea about the reason of blockage in case of not using XBUS cache?!

stnolting, 2024-10-28:
> The problem is that it's blocked at `Booting from 0x00000000`

This can have a lot of reasons... At first, the (external) DMEM is used to setup the stack. If that already fails the further boot process crashes.

> I found out that SRAM memory is not cleared after power off and addresses beyond program binary contain non zero data. Is it necessary that they get zero?

Basically, no memory is cleared nor have to be cleared at all except for the `.bss` segment. See [:books: 4.7. Start-Up Code (crt0)](https://stnolting.github.io/neorv32/#_start_up_code_crt0)

> (I checked memory values by bus_explorer in external flash (XIP method), and the application binary is placed correctly in SRAM.)

Did you also try to access the external SRAM with the bus explorer?

> After enabling XBUS_CACHE_EN, it works now! Does anyone have any idea about the reason of blockage in case of not using XBUS cache?!

This might be caused by the SRAM controller. Maybe it does not support single-word accesses (and just block accesses that are used by the cache)?

closed 2024-11-08 by mahdi259