stnolting/neorv32 #1080: Questions reg. UART and SPI demo

vinith32, 2024-10-28:
1). Hi, your work on NEORV32 is amazing. I am trying to simulate the examples. Im getting hello world and hello cpp output via UART, but when i run _DEMO_SPI, the simulation is freezed and getting max runtime error.

2). so every example, sw/simple/testbench's output is displayed in the console VIA UART. So, actually where it is defined.

3). In the example files, can we call them as "c test cases" ?

4). So, we have a uart receiver in sim/simple. So, how to send value from the testbench, so that the uart receiver can receive?

stnolting, 2024-10-29:
Hey @vinith32.

> Hi, your work on NEORV32 is amazing. 

Thank you very much! :)

> I am trying to simulate the examples. Im getting hello world and hello cpp output via UART, but when i run _DEMO_SPI, the simulation is freezed and getting max runtime error.

The [demo_spi](https://github.com/stnolting/neorv32/blob/main/sw/example/demo_spi/main.c) program provides an _interactive_ console (via UART) to allow the user to execute manual SPI transmissions;

https://github.com/stnolting/neorv32/blob/f9a2801c5c5764f26e57867ee67d7a6eebbd941b/sw/example/demo_spi/main.c#L40

When simulating this program, there is no UART input. So this program will just wait for input until the simulation terminates.

> so every example, sw/simple/testbench's output is displayed in the console VIA UART. So, actually where it is defined.

That depends on the compile switches. If you compile with `USER_FLAGS+=-DUART0_SIM_MODE` all UART0 outputs are redirected to the simulator console. We tried to explain this in [:books: 2.7.13. Primary Universal Asynchronous Receiver and Transmitter (UART0)](https://stnolting.github.io/neorv32/#_primary_universal_asynchronous_receiver_and_transmitter_uart0) (subsection "Simulation Mode").

> In the example files, can we call them as "c test cases" ?

Yes, basically all example programs are also some kind of test programs as they target each individual peripheral module.

> So, we have a uart receiver in sim/simple. So, how to send value from the testbench, so that the uart receiver can receive?

Unfortunately, there is no default sim_tx -> core_rx mechanism right now. We just have a simulated UART receiver. But it should not be too hard to add the TX counterpart I think. 🤔

vinith32, 2024-11-01:
Ok sir. So, here demo_spi can be simualted by enabling the inbuilt SPI using the testbench or it needed a real external spi device?

stnolting, 2024-11-02:
Sure, you can simulate the `demp_spi` program, but nothing will happen in the simulation :wink:

Furthermore, the SPI port is not connected to any simulated device inside the testbench - there is just some kind of local echo that connects the SPI module (SPI host) to the SDI (SPI device):

https://github.com/stnolting/neorv32/blob/cbfd4730def232758eb10e58079aa3916a1d7569/sim/neorv32_tb.vhd#L320-L329

vinith32, 2024-11-04:
The UART provides a simulation-only mode to dump console data as well as raw data directly to a file. When the simulation mode is enabled (by setting the UART_CTRL_SIM_MODE bit) there will be no physical transaction on the uart0_txd_o signal.

I got this content from the datasheet sir. So, in which VHDL file can i find the UART_CTRL_SIM_MODE bit declaration? I searched but not found.

stnolting, 2024-11-04:
It is defined right inside the UART module (`UART_CTRL_SIM_MODE` is the name/alias for the software framework; the actual VHDL signal has a slightly different name):

https://github.com/stnolting/neorv32/blob/06f3177249dcf29cebc9b5c88ab04d6731bde77f/rtl/core/neorv32_uart.vhd#L88

This is the actual "simulation transmitter":

https://github.com/stnolting/neorv32/blob/06f3177249dcf29cebc9b5c88ab04d6731bde77f/rtl/core/neorv32_uart.vhd#L471-L503

vinith32, 2024-11-04:
Ok sir. So, the example is used to just generate the application image file and add to the rtl folder and run using the default testbench.

 Is there any relation between the signals used in the rtl folder (ex. uart.vhd) and the library file(uart.c, uart.h)?

stnolting, 2024-11-04:
The UART simulation mode cna be used for simulation to speed up UART output; instead of transmitting the UART data bit by bit via the TX line (using the programmed Baud rate) the data is just written to the simulator console with no transmission overhead.

> Is there any relation between the signals used in the rtl folder (ex. uart.vhd) and the library file(uart.c, uart.h)?

Yes, the software libraries provide a hardware abstraction layer (HAL) that, for example, reflects the UART's memory mapped registers.

This is the definition of the memory-mapped registers:

https://github.com/stnolting/neorv32/blob/06f3177249dcf29cebc9b5c88ab04d6731bde77f/sw/lib/include/neorv32_uart.h#L28-L34

There are also "bit-definitions" for control register (`CTRL`):

https://github.com/stnolting/neorv32/blob/06f3177249dcf29cebc9b5c88ab04d6731bde77f/sw/lib/include/neorv32_uart.h#L40-L75

These definitions match with the HDL-based control register layout in the VHDL file:

https://github.com/stnolting/neorv32/blob/06f3177249dcf29cebc9b5c88ab04d6731bde77f/rtl/core/neorv32_uart.vhd#L48-L72

vinith32, 2024-11-04:
Thank you sir. !!!!

vinith32, 2024-11-04:
So, like UART i need to verify the functionality of spi. the demo_spi runs but no output. Can you help me to check SPI ? Can we check using existing setups? Or anything additionally needed?

stnolting, 2024-11-04:
I think that depends on what you actually want to check.

For example, you could add a simulation model (like a simple SPI flash) to the testbench and write a program for accessing it that will also be simulated.

Or you could move to "real hardware" and put a NEORV32 + SPI into you FPGA of choice. If you connect a real SPI device to it you could use the spi_demo program to perform SPI transactions via the UART terminal.

closed 2024-11-14 by vinith32