stnolting/neorv32 #1082: Rework processor boot configuration - ideas?

stnolting, 2024-10-30:
I would like to rework the boot configuration of the processor as (IMHO) it is way too complicated to understand and also not very flexible.

My idea (so far) is to add two new top generics:
```vhdl
BOOT_CONFIG      : natural range 0 to 2           := 0;           -- boot configuration select
BOOT_ADDR_CUSTOM : std_ulogic_vector(31 downto 0) := x"00000000"; -- custom boot address if BOOT_CONFIG = 2
```

Now `BOOT_CONFIG` selects the actual boot mode:

0. **Bootloader** - enable the bootloader ROM, implement the internal instruction memory (IMEM) as blank RAM and adjust the CPU boot address to beginning of the bootloader ROM
1. **IMEM as ROM** - implement IMEM as ROM pre-initialized with the application image and adjust the CPU boot address to beginning of the IMEM
2. **Custom** - use `BOOT_ADDR_CUSTOM` as CPU boot address (do not enable bootloader ROM, implement IMEM as blank RAM)

What do you think about this? Are there any other configurations that should be covered?

closed 2024-11-02 by stnolting

stnolting, 2024-11-18:
Hey @Unike267.

> I have successfully loaded IMEM using a .hex file instead of .vhd, but to do it you must add an extra function in neorv32_package.vhd to read the hex file through std.textio.all library (only supported in VHDL 2008).

That looks great! I think it is quite similar to what we are doing in the RISCOF testbench: https://github.com/stnolting/neorv32-riscof/blob/4c2b91ebcd324f01b8418ff98bce27bc0d4231f1/sim/neorv32_riscof_tb.vhd#L79

> The second option (IMEM as ROM) can be subdivided into two. One to load the ROM from app_image (.vhd) and another one to load the ROM from hexadecimal file (.hex).

I also thought about this and I think we had already several discussions about this. The main drawback is the "portability". You need to specify a file path for the HEX image which has to be adjusted for each single processor setup (so we cannot provide a generic version).

That's why there is no "default" or "built-in" option for that. But as you can see in the RISCOF example, boot mode 1 (boot from custom address) can be used together with an external IMEM for arbitrary memory initialization schemes - including load form HEX file.

Btw, sorry for the late reply 🙈

Unike267, 2024-11-19:
Hi Stephan!

> That looks great! I think it is quite similar to what we are doing in the RISCOF testbench: https://github.com/stnolting/neorv32-riscof/blob/4c2b91ebcd324f01b8418ff98bce27bc0d4231f1/sim/neorv32_riscof_tb.vhd#L79

Yeah!

> The main drawback is the "portability". You need to specify a file path for the HEX image which has to be adjusted for each single processor setup (so we cannot provide a generic version).

That's right! But, if you modify the `common.mk` file a little bit adding new argument, for instance `image-hex` , to locate the generated `hex` file into the core, the path:

```vhdl
constant ROM_INIT_FILE : string := "neorv32_raw_exe.hex";
```
is going to be constant, right? As similar way that the argument `image` does.

In addition to that, in the same way that you provide by default the boot-image in `.vhd`, you could provide it in `.hex`.

Therefore, the boot configuration could be:

0. BOOTLOADER - Load from default `.vhd` (implement IMEM as blank RAM, load it from Prompt)
1. BOOTLOADER-HEX - Load from default `.hex`  (implement IMEM as blank RAM, load it from Prompt)
2. IMEM as ROM - Implement IMEM as ROM loaded from `.vhd`, `make image` overwrite the default `.vhd`
3. IMEM as ROM-HEX - Implement IMEM as ROM loaded from `.hex`, `make image-hex` overwrite the default `.hex`
4. Custom

> Btw, sorry for the late reply 🙈

Ohh, don't worry about it. 😁 😁 😁 

Cheers!

stnolting, 2024-11-20:
> That's right! But, if you modify the common.mk file a little bit adding new argument, for instance image-hex , to locate the generated hex file into the core, the path:

That's a great idea! Basically, we could just use this variable to pass another parameter (like setting a generic according to the file path) to GHDL:

https://github.com/stnolting/neorv32/blob/457f902d117e07a895f0fdfe0000ad1fc978d3b3/sw/common/common.mk#L46

But this will only work for the build-in GHDL simulation. When using a different simulator, the path configuration (or more precisely: the way of passing this to the simulator) might be a little bit different. So this is a portability problem again... 😅

> is going to be constant, right? As similar way that the argument image does.

Well, it is constant but without an actual path the simulator will assume this file is located inside the simulation (build) folder.

> In addition to that, in the same way that you provide by default the boot-image in .vhd, you could provide it in .hex.

That's right - at least half of it I think. The VHDL image file is just another HDL **source** file that is included together with all the other rtl files. Yes, you'll need to specify the path to that specific folder when creating a completely new setup. But then, you would have to specify _another_ path for the HEX file as these need to be imported separately (as they are no HDL source files).

I mean this is no big deal and lot of other soft-processor do it this way. But it is yet another pitfall (especially for newcomers) that I would like to avoid.

However, maybe it would be nice to have this "load from hex file" feature implemented in the default testbench. This file is no (critical) rtl core file so it could also be used to illustrate some more "advanced" concepts - just like loading a HEX file.

Unike267, 2024-11-22:
Hi Stephan!

> But this will only work for the build-in GHDL simulation. When using a different simulator, the path configuration (or more precisely: the way of passing this to the simulator) might be a little bit different. So this is a portability problem again... 😅

Oh my bad, you are absolutely right! Sometimes I forgot that there are others simulators jaja. 

> That's right - at least half of it I think. The VHDL image file is just another HDL source file that is included together with all the other rtl files. Yes, you'll need to specify the path to that [...]

> I mean this is no big deal and lot of other soft-processor do it this way. But it is yet another pitfall (especially for newcomers) that I would like to avoid.

Yeah, I understand that.

> However, maybe it would be nice to have this "load from hex file" feature implemented in the default testbench. This file is no (critical) rtl core file so it could also be used to illustrate some more "advanced" concepts - just like loading a HEX file.

Ohh! This is a great idea! ❤️

Cheers!

stnolting, 2024-11-25:
> Oh my bad, you are absolutely right! Sometimes I forgot that there are others simulators **jaja**.

I really had to google that "jaja"... I thought that was a German pun. 😅 You always learn something new.

> Ohh! This is a great idea!

I'll have a look at this. I'm not sure yet if there is a nice way to implement this in a setup-independent way. 🤔

Unike267, 2024-11-26:
> I really had to google that "jaja"... I thought that was a German pun. 😅 You always learn something new.

Jajajajaja! That’s how we laugh through text in Spanish (I thought it was understood internationally 😅).

I knew that different languages express laughter differently. For instance, in Basque, it's "kar kar kar." Out of curiosity, how do people laugh in Deutschland?

> I'll have a look at this. I'm not sure yet if there is a nice way to implement this in a setup-independent way. 🤔

Oh, don't worry! (Here, I could use the Spanish expression "VVCC" (Vamos Viendo Con Calma) which means "let's take it slow and see how it goes.")  :smile:

stnolting, 2024-11-26:
> how do people laugh in Deutschland?

Not at all.

Just kidding. 😅 We are also using "haha" and I always through this would be "international". Oh what a naive view I had.

> "let's take it slow and see how it goes.

Right. I'll try to come up with some ideas.