stnolting/neorv32 #1088: Zephyr RTOS  URL needs update

cryptobharathi, 2024-11-04:
The URL pointing to Zephyr RTOS in README (https://github.com/stnolting/neorv32) is `https://docs.zephyrproject.org/latest/boards/riscv/neorv32/doc/index.html ` is modified in Zephyr RTOS documentation page, which needs to be updated to `https://docs.zephyrproject.org/latest/boards/others/neorv32/doc/index.html`

[image]

stnolting, 2024-11-04:
Thanks for the hint!
I'll fix that (or do you want to make a PR? :wink:).

stnolting, 2024-11-06:
Fixed.

closed 2024-11-06 by stnolting

cryptobharathi, 2024-11-08:
> Thanks for the hint! I'll fix that (or do you want to make a PR? 😉).

Its a small observation, so i feel direct fix is better than PR :) 

Happy to see, its fixed and closed 👍