stnolting/neorv32 #1097: Multiple unexpected nullpointer behaviour

kendegemaro, 2024-11-19:
Hello Stephan,

Sadly, I require your assistance with my previous issue #1089 , because I still do not understand the real cause of the problem.
Therefore, I made my best effort to document my used code and software so I hope you can somehow replicate and compare the problem. As I have exhausted my ideas and debug options, your help would be really appreciated, so I can continue my academy project.

Please see the details below.
----------------------------------------------------------------------------------------------------------------
**Describe the bug**
I am using NEORV32 Release 1.9.3 with your prebuilt compiler toolchain [rv32i-131023](https://github.com/stnolting/riscv-gcc-prebuilt/releases/tag/rv32i-131023). During the execution of my application, multiple variables, for example the global function pointer k2c_linear, contain nullpointer unexpectedly, although they are assigned a value in a global context. If you compare the same application execution on a Windows architecture, everything is working as expected.

**To Reproduce**
Please compile and execute the source code yourself from: [My repo](https://github.com/kendegemaro/NEORV32_MNIST_Demo/tree/main/target_platforms/NEORV32). Use MEM_INT_IMEM_SIZE=64\*1024 and MEM_INT_DMEM_SIZE=32\*1024 for your Quartus project.

**Expected behavior**
The application should not crash. At the end of main(), the while loop should execute a simple binary 1Hz counter on the GPIOs.

**Observed behaviour**
The application crashed mid-execution. The binary counter at the end of main() is not being executed. As an effect of the crash, the controller either freezes or reboots.

**Screenshots**

**Environment:**
 - Windows 11 for Quartus Prime Lite & Linux Mint for compiling the application.
 - Hardware version 1.9.3
 - Precompiled compiler toolchain rv32i-131023
 - FPGA Target: DE0-Nano_SoC
 - NEORV32 ISA Extensions: CPU_EXTENSION_RISCV_C, CPU_EXTENSION_RISCV_M, CPU_EXTENSION_RISCV_Zicntr, CPU_EXTENSION_RISCV_Zfinx
 - Used compiler MARCH = rv32i_zicsr_zifencei_zfinx
 - Top level entity: neorv32_test_setup_approm.vhd 
 - Application image transfer method: neorv32_application_image.vhd

--------------------
I thank you a lot in advance!
kendegemaro

stnolting, 2024-11-20:
Hey @kendegemaro.

> During the execution of my application, multiple variables, for example the global function pointer k2c_linear, contain nullpointer unexpectedly,

How did you find that out?

First of all, I recommend to enable the default [:books: NEORV32 runtime environment](https://stnolting.github.io/neorv32/#_neorv32_runtime_environment):

```c
neorv32_rte_setup();
```

This will output a message if there is any unhandled exception (if UART0 is enabled and configured).

Additionally, you could setup a [:books: PMP](https://stnolting.github.io/neorv32/#_smpmp_isa_extension) rule so that null-pointer accesses raise an exception.

I did a quick test of your code on my testsetup. The code compiles without errors so any null-pointer issue must be caused by a runtime situation. I have used the makefile/configuration from your project. The code starts normally but after executing `MNIST_NeuralNet(&test1_flatten_input_input,&c_dense_1_test1);` _something_ happens and the the code restarts.

After doing some investigations with the on-chip debugger I found out that there was a stack overflow. The stack grows (downwards) and at some point it collides with the `data` and `bss` sections overriding some critical variables/pointers (see [:books: RAM Layout](https://stnolting.github.io/neorv32/#_ram_layout)).

I've increased the RAM size to 64kB and (at least in simulation) the code seems to run fine now. Could you try that on your side as well?

stnolting, 2024-11-25:
> I really did not expect a simple stack overflow [...]

These can be really hard to find sometimes as this is runtime-specific thing... 🙈

> Because my only debug option were the LEDs, I printed the binary content of pointer variables (4 times 8 bit) onto the LEDs. Some variables seemed to have the value 0x0, which must be an effect of the stack overflow.

Oh, that's... _basic_... 😅 Can't you use the UART?

> As the issue is solved

🚀 🎉

kendegemaro, 2024-11-28:
>Can't you use the UART?

Until now, I have been hindered by not having a USB-UART adapter for my PC (and because I was afraid of the amount of work I'd have to put into the project to make it work, but let's not talk about that 🫠).

In the end though, implementing a working UART interface was surprisingly easy and your **_getting started_** tutorial, together with the example code and the C library, were very handy. I have even dared to try out the GPTMR module with success :)

stnolting, 2024-12-01:
> Until now, I have been hindered by not having a USB-UART adapter for my PC (and because I was afraid of the amount of work I'd have to put into the project to make it work, but let's not talk about that 🫠).

It should be plug & play on both sides (PC and NEORV32) - at least the FTDI-based adapters work out of the box (on Windows and Linux).

> In the end though, implementing a working UART interface was surprisingly easy and your getting started tutorial, together with the example code and the C library, were very handy. I have even dared to try out the GPTMR module with success :)

Thant's great to hear! Thank you very much for the feedback! 😄