stnolting/neorv32 #1109: Zephyr RTOS Fails to Boot on NEORV32 SoC (Arty A7-100T)

mohamedElbouazzati, 2024-11-29:
**Describe the bug**

I am running Zephyr RTOS on the NEORV32 SoC implemented on a **Digilent Arty A7-100T FPGA board**. However, the terminal fails to boot and is stuck at :  `Booting from 0x00000000`
I Have Tried :
1. Verified the memory mapping between the **NEORV32** implementation and the **Zephyr device tree**; they appear to match.
2. Increased the instruction memory to **64 KB** and data memory to **32 KB**, ensuring they align with Zephyr's linker configuration.

Despite these adjustments, the issue persists. Assistance in diagnosing and resolving this boot failure would be greatly appreciated.

**To Reproduce**
Steps to reproduce the behavior

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
- Hello world test : 
 ```

<< NEORV32 Bootloader >>

BLDV: Jan 21 2023
HWV:  0x01080000
CID:  0x00000000
CLK:  0x05f5e100
ISA:  0x40801104 + 0x00000081
SOC:  0x0007000d
IMEM: 0x00010000 bytes @0x00000000
DMEM: 0x00008000 bytes @0x80000000

Autoboot in 8s. Press any key to abort.
Aborted.

Available CMDs:
 h: Help
 r: Restart
 u: Upload
 s: Store to flash
 l: Load from flash
 x: Boot from flash (XIP)
 e: Execute
CMD:> u
Awaiting neorv32_exe.bin... OK
CMD:> e
Booting from 0x00000000...

                                                                                      ##        ##   ##   ##    
 ##     ##   #########   ########    ########   ##      ##   ########    ########     ##      ################  
####    ##  ##          ##      ##  ##      ##  ##      ##  ##      ##  ##      ##    ##    ####            ####
## ##   ##  ##          ##      ##  ##      ##  ##      ##          ##         ##     ##      ##   ######   ##  
##  ##  ##  #########   ##      ##  #########   ##      ##      #####        ##       ##    ####   ######   ####
##   ## ##  ##          ##      ##  ##    ##     ##    ##           ##     ##         ##      ##   ######   ##  
##    ####  ##          ##      ##  ##     ##     ##  ##    ##      ##   ##           ##    ####            ####
##     ##    #########   ########   ##      ##      ##       ########   ##########    ##      ################  
                                                                                      ##        ##   ##   ##    
Hello world! :)
```

- Last version of zephyr and neorv32

```
BLDV: Oct 31 2024
HWV:  0x01100605
CLK:  0x05f5e100

MISA: 0x40801104
XISA: 0x00000083
SOC:  0x0003800d
IMEM: 0x00010000
DMEM: 0x00008000

Autoboot in 10s. Press any key to abort.
Aborted.

Available CMDs:
 h: Help
 r: Restart
 u: Upload
 s: Store to flash
 l: Load from flash
 x: Boot from flash (XIP)
 e: Execute
CMD:> e
No executable.
CMD:> u
Awaiting neorv32_exe.bin... OK
CMD:> e
Booting from 0x00000000...
```
 - Zephyr SDk 0.15.2 and neorv32 v1.8.0
```
BLDV: Jan 21 2023
HWV:  0x01080000
CID:  0x00000000
CLK:  0x05f5e100
ISA:  0x40801104 + 0x00000081
SOC:  0x0007000d
IMEM: 0x00010000 bytes @0x00000000
DMEM: 0x00008000 bytes @0x80000000

Autoboot in 8s. Press any key to abort.
Aborted.

Available CMDs:
 h: Help
 r: Restart
 u: Upload
 s: Store to flash
 l: Load from flash
 x: Boot from flash (XIP)
 e: Execute
CMD:> u
Awaiting neorv32_exe.bin... OK
CMD:> e
Booting from 0x00000000..
```

**Environment:**
 - Zephyr OS
 - zephyr-sdk-0.15.2
 - riscv32-unknown-elf-gcc  12.2.0
 - tried also with **last zephyr version** 

**Hardware:**
 - SoC on vivado folder from neorv32-setups : `commit ; 62083bd`
 - neorv32 v1.8.0
 - increased memory for d and i on` /neorv32/rtl/test_setups/neorv32_test_setup_bootloader.vhd`
 - updated provided xdc file from 35 to 100t
 - Tried also with **current version of neorv32**

**Additional context**
I followed also [github discussion]( https://github.com/stnolting/neorv32/discussions/172) and tested the installation from [ReconOS_Zephyr_RISCV](https://github.com/Lien182/ReconOS_Zephyr_RISCV/tree/master )
Does NEORV32 still support Zephyr RTOS?

stnolting, 2024-12-01:
Hey @mohamedElbouazzati!

> Hello world test :

This looks good! So the processor seems to operate just fine! :+1:

> However, the terminal fails to boot and is stuck at : Booting from 0x00000000

That it gets stuck at this point could indicate that there is a problem with the memory configuration (stack too small, insufficient memory size, etc.). Since you've already checked that, I'm a bit helpless now too.

> Does NEORV32 still support Zephyr RTOS?

That is a really good question. I still haven't found the time to get to grips with the build system yet, so the current Zephyr "port" still fully depends on repository-external resources.

There we some Zephyr experts here, maybe they can help out?

stnolting, 2025-03-04:
That is so cool! Thank you so much!

I still want to get into Zephyr too, but I just haven't really gotten around to it yet. Can you recommend a good tutorial for a "hardware perspective" (apart from your great porting of course)? 😉

I was wondering if it would be possible to automate the creation of device trees / drivers / header files / etc. by (only) using the processor's [SVD file](https://github.com/stnolting/neorv32/tree/main/sw/svd)?

robhancocksed, 2025-03-04:
The Zephyr board documentation for NEORV32 has a pretty good tutorial for how to get things up and running with a sample configuration: https://docs.zephyrproject.org/latest/boards/others/neorv32/doc/index.html

henrikbrixandersen, 2025-03-04:
> I still want to get into Zephyr too, but I just haven't really gotten around to it yet. Can you recommend a good tutorial for a "hardware perspective" (apart from your great porting of course)? 😉

The most "low-level"/hardware-near documentation is probably the various porting guides available here: https://docs.zephyrproject.org/latest/hardware/porting/index.html

> I was wondering if it would be possible to automate the creation of device trees / drivers / header files / etc. by (only) using the processor's [SVD file](https://github.com/stnolting/neorv32/tree/main/sw/svd)?

Some vendors do that with various degree of success. The SVDs allow for easy generation of e.g. addresses and interrupt mapping, but mapping to the properties of the individual node bindings is where it gets tricky. I honestly do not think it's worth it for the NEORV32.

henrikbrixandersen, 2025-03-04:
> The Zephyr board documentation for NEORV32 has a pretty good tutorial for how to get things up and running with a sample configuration: https://docs.zephyrproject.org/latest/boards/others/neorv32/doc/index.html

My plan going forward is to provide board variants for this matching the various (currently Minimal, MinimalBoot, and UP5kDemo) NEORV32 processor templates available in the upstream repository - with instructions on how to use Zephyr board extensions for supporting custom NEORV32 processor configurations out-of-tree.

henrikbrixandersen, 2025-04-02:
> My plan going forward is to provide board variants for this matching the various (currently Minimal, MinimalBoot, and UP5kDemo) NEORV32 processor templates available in the upstream repository - [...]

This has now landed in upstream Zephyr: https://docs.zephyrproject.org/latest/boards/others/neorv32/doc/index.html

I am currently working on adding support for more NEORV32 peripherals in Zephyr.

stnolting, 2025-04-04:
@henrikbrixandersen 

Thanks for the update and all your work! I really appreciate it.

closed 2025-09-06 by stnolting