stnolting/neorv32 #1115: A suggestion about IMEM in simulation

mahdi259, 2024-12-05:
**Is your feature request related to a problem? Please describe.**
To instantiate external instruction and data memories in neorv32_tb.vhd a unified file is instantiated twice and IMEM is not initialized.

**Describe the solution you'd like**
Is it possible to add initialization option to xbus_memory.vhd when it's instantiated as IMEM?

In that case, we can assign 1 in `BOOT_MODE_SELECT` and use external IMEM.

stnolting, 2024-12-10:
That's a great idea I think! So maybe something like the initialization mechanims in the RISCOF testbench?! (-> https://github.com/stnolting/neorv32-riscof/blob/b285dc0811ee7d028b4b02c9553ef7f3f84ccc65/sim/neorv32_riscof_tb.vhd#L64)

Just pass a HEX file as (another) testbench generic to initialize any of the `xbus_memory` components?

mahdi259, 2024-12-11:
Yes that's right. Btw you have application_image.vhd file, why not using that?

stnolting, 2024-12-11:
Good point!

But then we could only use one image (for one or more memory blocks) since we need to "hardcode" the image constant (i.e. `application_init_image`) and the the image package (i.e. `neorv32_application_image`). But it would also simplify simulation setups as you do not need to add further files (like some HEX files) to the setup... 🤔

stnolting, 2024-12-11:
> Sorry for late reply.

Don't worry! I'm also way too slow at the moment when it comes to answers (I've just moved to another town)...

> Do you intend to pack multiple compiled programs into one hex file?

I thought about providing three new generics for the testbench. Each one could be used to pass an individual HEX file path to one of the three testbench memory modules. 🤔

mahdi259, 2024-12-12:
That way you should make separate hex files for dmem, imem and the third memory. Right?

stnolting, 2024-12-12:
> That way you should make separate hex files for dmem, imem and the third memory. Right?

That is up to the user.

I am preparing a proposal right now where the testbench is equipped with to additional generics like this:

```vhdl
MEM_FILE_1 : string := "",
MEM_FILE_2 : string := "",
MEM_FILE_3 : string := "",
```

If `MEM_FILE` is set to `""` the according memory is not initialized at all (or maybe just set to all-zero). If the user provides a HEX file (path) then that image is used to initialize the according memory.

closed 2024-12-14 by stnolting

mahdi259, 2024-12-14:
Thanks a lot