stnolting/neorv32 #1116: Merge neorv32-vunit with neorv32

csantosb, 2024-12-05:
**Is your feature request related to a problem? Please describe.**

Commit a0f43490 succeeds on running sim tests, see [here](https://builds.sr.ht/~csantosb/job/1381186#task-sim)

But it fails to run tests in neorv32-vunit, see [here](https://builds.sr.ht/~csantosb/job/1381186#task-vunit), due to a [test runner timeout](https://builds.sr.ht/~csantosb/job/1381186#task-vunit-1104)

I manage to reproduce the issue locally.

Now, I think that the point on unit testing with vunit is to check this kind of errors for every new commit, being notified of any such issue. Having two separate repositories goes against this logic.

**Describe the solution you'd like**

Merge neorv32-vunit with neorv32.

**Describe alternatives you've considered**

What I’m doing now is to mirror two repositories, combining them and pushing somewhere else where I run the tests. But this is far from ideal.

**Additional context**

I’m running here the whole tests using a fully reproducible pipeline under guix, but I need to merge different repositories, doing some scripting.

stnolting, 2024-12-10:
> Now, I think that the point on unit testing with vunit is to check this kind of errors for every new commit, being notified of any such issue. Having two separate repositories goes against this logic.

I fully agree with you.

The only problem is all the dependencies. Some commits ago, vunit was part of this repository. It was included as a GitHub package (from a another repository) that is based on a Docker file that gets pulled-in from another source. If someone changes something in that chain the entire workflow is broken. We already have this problem with GHDL, RISC-V GCC, FreeRTOS, the RISC-V architecture tests, ... And I fail completely in trying to manage them all. 😅

That's why certain parts of this project got "sourced-out" into seperate repositories. I also know that this is not ideal, but unfortunately I can't manage it any other way. 🙈

csantosb, 2024-12-11:
I see the problem. This is annoying.

This is precisely the kind of troubles I’m trying to fix once forever with [guix](https://infosec.press/csantosb/ci-sourcehut). 

Having a couple of manifest files to create a declarative and reproducible env for ci is easy; pushing manually to github/lab is not a problem. Achieving the same automatically is more complicated, I’m afraid (and so I use https://sr.ht by now).

csantosb, 2024-12-12:
I've been setting up a declarative docker image to be used with ci github actions, [here](https://github.com/csantosb/hdl-image#test). Repo tags match image tags.

It is fully reproducible, and based on a [set of packages](https://github.com/csantosb/hdl-image/blob/master/manifest.scm). The image is deterministic in the sense that if may be produced again bit per bit as for its [channels file](https://github.com/csantosb/hdl-image/blob/master/channels.scm), see [readme](https://github.com/csantosb/hdl-image/blob/master/readme.org), so no upstream breakages.

Tested with neorv32 and its vunit suite.

Hope it helps !

stnolting, 2025-01-06:
> I've been setting up a declarative docker image to be used with ci github actions, [here](https://github.com/csantosb/hdl-image#test).

That looks cool! I really need to work more with Docker again... 🙈 Anyway, thanks for sharing! :+1:

closed 2025-01-07 by csantosb