stnolting/neorv32 #1127: Examples compiled with xpack toolchain crash on neorv32_test_setup_bootloader.vhd

jpf91, 2024-12-23:
**Describe the bug**
Hi,

I've tried porting neorv32 to a new target (Tan Nano 20K with OSS Toolchain). Because of this, I started with a new setup with NeoRV32 `v1.10.7` and the ` xpack-riscv-none-elf-gcc-14.2.0-3-linux-x64` toolchain.

The NeoRV32 boots successfully into the precompiled bootloader. But if I recompile the bootloader using the xpack toolchain or use the original bootloader to upload and execute an example, the application crashes.

**To Reproduce**
Using this configuration based on the `neorv32_test_setup_bootloader.vhd` example:
```vhdl
        generic map (
            -- Clocking --
            CLOCK_FREQUENCY   => 100000000,         -- clock frequency of clk_i in Hz
            -- Boot Configuration --
            BOOT_MODE_SELECT  => 0,                 -- boot via internal bootloader
            -- Enable JTAG / Debugger --
            OCD_EN            => true,
            -- RISC-V CPU Extensions --
            RISCV_ISA_C       => true,              -- implement compressed extension?
            RISCV_ISA_M       => true,              -- implement mul/div extension?
            RISCV_ISA_Zicntr  => true,              -- implement base counters?
            -- Internal Instruction memory --
            MEM_INT_IMEM_EN   => true,              -- implement processor-internal instruction memory
            MEM_INT_IMEM_SIZE => 8096,              -- size of processor-internal instruction memory in bytes
            -- Internal Data memory --
            MEM_INT_DMEM_EN   => true,              -- implement processor-internal data memory
            MEM_INT_DMEM_SIZE => 1024,              -- size of processor-internal data memory in bytes
            -- Processor peripherals --
            IO_GPIO_NUM       => 4,                 -- number of GPIO input/output pairs (0..64)
            IO_MTIME_EN       => true,              -- implement machine system timer (MTIME)?
            IO_UART0_EN       => true,              -- implement primary universal asynchronous receiver/transmitter (UART0)?
            IO_SPI_EN         => true,              -- Used for flash bootloader
            IO_GPTMR_EN       => true,
            IO_DMA_EN         => true
       )
```

In `sw/example/hello_world`:
```bash
make clean_all exe

Memory utilization:
   text    data     bss     dec     hex filename
   5516       0     116    5632    1600 main.elf
Compiling image generator...
Generating neorv32_exe.bin
Executable size in bytes:
5528
```

And uploading:
```
<< NEORV32 Bootloader >>

BLDV: Oct 31 2024
HWV:  0x01100700
CLK:  0x05f5e100
MISA: 0x40801104
XISA: 0x00000c83
SOC:  0x1007c01d
IMEM: 0x00002000
DMEM: 0x00000400

Autoboot in 10s. Press any key to abort.
Aborted.

Available CMDs:
 h: Help
 r: Restart
 u: Upload
 s: Store to flash
 l: Load from flash
 x: Boot from flash (XIP)
 e: Execute
CMD:> u
Awaiting neorv32_exe.bin... OK
CMD:> e
Booting from 0x00000000...

<NEORV32-RTE> [M] Store access fault @ PC=0x0000020C, MTINST=0x02C12823, MTVAL=0x80000400 </NEORV32-RTE>
<NEORV32-RTE> [M] Store access fault @ PC=0x00000210, MTINST=0x02D12A23, MTVAL=0x80000404 </NEORV32-RTE>
```

I think I followed the documentation closely. Do I need additional ISA extensions in the CPU? Or is there any additional flag I should pass to the compiler?

Also it would be nice to know how the bootloader image was created (toolchain version, special flags?). When I run `make bootloader` with the xpack toolchain, the size increases from 4072 bytes to 5420 bytes. So even if that compiled code actually worked, this would still be a drawback.

**Expected behavior**
Hello world example should run.

**Environment:**
 - Linux, Fedora 40
 - xpack-riscv-none-elf-gcc-14.2.0-3-linux-x64

**Hardware:**
 - v1.10.7 tag
 - See above for HW configuration

stnolting, 2024-12-23:
Hey @jpf91!

> I've tried porting neorv32 to a new target (Tang Nano 20K with OSS Toolchain). [...] The NeoRV32 boots successfully into the precompiled bootloader.

That sounds great!

> Using this configuration based on the neorv32_test_setup_bootloader.vhd example:

This looks fine. I am not really sure how much RAM ("DMEM) is required by the hello_world program, but 1k should be sufficient.

> Do I need additional ISA extensions in the CPU? Or is there any additional flag I should pass to the compiler?

You did not specify any additional compiler flags so `rv32i` is used as default. So your ISA HW configuration is just fine.

The problem is here:

```
<NEORV32-RTE> [M] Store access fault @ PC=0x0000020C, MTINST=0x02C12823, MTVAL=0x80000400 </NEORV32-RTE>
```

The core is trying to access data memory (DMEM) at `0x80000400`, but this is beyond the 1k limit that you have configured (`MEM_INT_DMEM_SIZE => 1024,`). I guess the executable was compiler using an incorrect memory layout.

You need to adjust the memory sizes according to your generic sconfiguration in the [application makefile](https://stnolting.github.io/neorv32/#_application_makefile). Alternatively, you can do the configuration when invoking `make`:

```bash
neorv32/sw/example/hello_world$ make USER_FLAGS+="-Wl,--defsym,__neorv32_rom_size=8k -Wl,--defsym,__neorv32_ram_size=1k" clean_all exe
```

> Also it would be nice to know how the bootloader image was created (toolchain version, special flags?). When I run make bootloader with the xpack toolchain, the size increases from 4072 bytes to 5420 bytes. So even if that compiled code actually worked, this would still be a drawback.

The toolchain version should not be (too) relevant here. Any recent version should work. All the relevant flags are defined in the bootloader makefile: https://github.com/stnolting/neorv32/blob/main/sw/bootloader/makefile

To (re-) compile it:

```bash
neorv32/sw/bootloader$ make clean_all bootloader
```

jpf91, 2024-12-24:
> That sounds great!

Yeah, it's quite cool :-) It seems the internal PLL is not working properly with the OSS toolchain and I think the BRAM inference is probably broken when using more then 16 kBit. (I guess I can fix the  BRAM inference issue by manually splitting memories in VHDL). But compared to Vivado, just running `make` once and having a bitstream ready in 7 minutes is awesome :-)

The board itself is quite nice. Small form-factor, JTAG+UART+SPI combined on one USB-C connector, lots of peripherals, SDRAM, etc. So far I have NEORV32 booting from the FPGA SPI flash (it has 64 MBit but needs only 8 for FPGA) and using the UART. I wonder if I can repurpose the FPGA JTAG for the NEORV32 OCD... Then it could be a nice and cheap development platform for NEORV32 in an Arduino Nano form-factor.

For the proprietary toolchain, there are even NES/SNES/Gameboy clones, so the board is quite capable.

> The core is trying to access data memory (DMEM) at 0x80000400

Oh, I'm such an idiot. I should have seen that :see_no_evil: Thanks, that was indeed the only issue. hello_world still didn't work with `__neorv32_ram_size=1k`, but I just increased the dmem to 8k now. That doesn't use more hardware BRAMs anyway...

> The toolchain version should not be (too) relevant here

I figured out what caused the size increase of the bootloader, it was indeed not the toolchain. I initially used `-DUART_BAUD=115200`, which causes the size increase. I probably didn't properly run `make clean` after that test then.

Thanks for all the help!
Maybe once I have this setup ready we can think about whether to include it in neorv32-setups. (I didn't know NEORV32 is already tested with the OSS flow... That explains why things worked so smoothly :smile: )

closed 2024-12-24 by jpf91

stnolting, 2025-01-06:
> The board itself is quite nice.

I'm also thinking about getting one of these boards. 🤔

> Maybe once I have this setup ready we can think about whether to include it in neorv32-setups.

That would be really cool! :+1: